espressif/vscode-esp-idf-extension #1242: VSCODE ESP-IDF (v5.2.2) "Monitor Device" button not working as expected (IDFGH-13199)

TtijasS, 2024-07-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

VS Code express installed IDF v5.2.2

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-S3-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

ESP-IDF: Monitor Device button in VS Code should open the serial monitor with the output similar to:
```
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM6 instead...
--- esp-idf-monitor 1.4.0 on \\.\COM6 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x29 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378a06
0x40378a06: esp_cpu_wait_for_intr at C:/Users/tugos/esp/v5.2.2/esp-idf/components/esp_hw_support/cpu.c:145

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x178c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xcbc
load:0x403cc700,len:0x2d9c
entry 0x403c9914
```

### What is the actual behavior?

ESP-IDF: Monitor Device button used to work normally with the IDF v5.2.1, but in the the current version, v5.2.2, i'm unable to normally start the serial monitor with a simple click. 

cmd output upon clicking the button:
```
Microsoft Windows [Version 10.0.22631.3737]
(c) Microsoft Corporation. All rights reserved.

C:\Users\tugos\EspProjects\hello_world>set IDF_PATH='C:\Users\tugos\esp\v5.2.2\esp-idf'

C:\Users\tugos\EspProjects\hello_world>'C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe' 'C:\Users\tugos\esp\v5.2.2\esp-idf\tools\idf_monitor.py' -p COM6 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 'c:\Users\tugos\EspProjects\hello_world\build\hello_world.elf'
The filename, directory name, or volume label syntax is incorrect.
```
I can make it work again, but only temporarry, if i call the next commands. As can be seen, i removed single ticks ' ' around the idf path, and replaced the single ticks with double quotes in the the next call:
```
C:\Users\tugos\EspProjects\hello_world>set IDF_PATH=C:\Users\tugos\esp\v5.2.2\esp-idf                                                                                                                                   

C:\Users\tugos\EspProjects\hello_world>"C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe" "C:\Users\tugos\esp\v5.2.2\esp-idf\tools\idf_monitor.py" -p COM6 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 "c:\Users\tugos\EspProjects\hello_world\build\hello_world.elf"
```

After that the terminal starts normally with:
```
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM6 instead...
--- esp-idf-monitor 1.4.0 on \\.\COM6 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x29 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378a06
0x40378a06: esp_cpu_wait_for_intr at C:/Users/tugos/esp/v5.2.2/esp-idf/components/esp_hw_support/cpu.c:145

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x178c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xcbc
load:0x403cc700,len:0x2d9c
entry 0x403c9914
```

### Steps to reproduce.

1. Install VS Code
2. Install ESP IDF extension
3. Configure ESP IDF with express configuration, selecting the v5.2.2 stable version.
4. Open any example or existing project that you know should work
5. build and flash the project
6. open serial monitor by clicking on the "ESP-IDF: Monitor Device" ESP-IDF extension button

### Debug Logs.

```plain
GNU gdb (esp-gdb) 14.2_20240403
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-w64-mingw32 --target=xtensa-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
warning: could not convert 'main' from the host encoding (CP1252) to UTF-32.
This normally should not happen, please file a bug report.
set remotetimeout 20
0x40378a06 in esp_cpu_wait_for_intr () at C:/Users/tugos/esp/v5.2.2/esp-idf/components/esp_hw_support/cpu.c:145
145	}
connected to target using provided connectCommands
set remote hardware-watchpoint-limit 2
mon reset halt
JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
[esp32s3.cpu0] requesting target halt and executing a soft reset
[esp32s3.cpu0] Debug controller was reset.
[esp32s3.cpu0] Core was reset.
[esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
[esp32s3.cpu0] Reset cause (3) - (Software core reset)
[esp32s3.cpu1] requesting target halt and executing a soft reset
[esp32s3.cpu0] Core was reset.
[esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
[esp32s3.cpu1] Debug controller was reset.
[esp32s3.cpu1] Core was reset.
[esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
[esp32s3.cpu1] Reset cause (3) - (Software core reset)
[esp32s3.cpu0] Reset cause (3) - (Software core reset)
maintenance flush register-cache
Register cache flushed.
thb app_main
Hardware assisted breakpoint 1 at 0x420081ec: file C:/Users/tugos/EspProjects/hello_world/main/hello_world_main.c, line 17.
GDB unhandled notify: breakpoint-created: {"bkpt":{"number":"1","type":"hw breakpoint","disp":"del","enabled":"y","addr":"0x420081ec","func":"app_main","file":"C:/Users/tugos/EspProjects/hello_world/main/hello_world_main.c","fullname":"C:\\Users\\tugos\\EspProjects\\hello_world\\main\\hello_world_main.c","line":"17","thread-groups":["i1"],"times":"0","original-location":"app_main"}}

In the Debug Console view you can interact directly with GDB.
To display the value of an expression, type that expression which can reference
variables that are in scope. For example type '2 + 3' or the name of a variable.
Arbitrary commands can be sent to GDB by prefixing the input with a '>',
for example type '>show version' or '>help'.
[esp32s3.cpu0] Target halted, PC=0x420081EC, debug_reason=00000001
Set GDB target to 'esp32s3.cpu0'
[esp32s3.cpu1] Target halted, PC=0x40378A06, debug_reason=00000000
[New Thread 1070176128]
[Remote target exited]
[New Thread 1070179896]
[New Thread 1070178012]
[New Thread 1070167176]
[New Thread 1070171620]
[New Thread 1070165548]
[Switching to Thread 1070176128]

Thread 2 "main" hit Temporary breakpoint 1, app_main () at C:/Users/tugos/EspProjects/hello_world/main/hello_world_main.c:17
17	{
gdbserver stopped[Inferior 1 (Remote target) detached]
```

### More Information.

PROBLEMS tab output:
c_cpp_properties.json
[{
	"resource": "/c:/Users/tugos/EspProjects/hello_world/.vscode/c_cpp_properties.json",
	"owner": "c:\\Users\\tugos\\EspProjects\\hello_world",
	"severity": 4,
	"message": "Cannot find \"c:/Users/tugos/EspProjects/hello_world/build/compile_commands.json\"",
	"startLineNumber": 6,
	"startColumn": 32,
	"endLineNumber": 6,
	"endColumn": 79
},{
	"resource": "/c:/Users/tugos/EspProjects/hello_world/.vscode/c_cpp_properties.json",
	"owner": "c:\\Users\\tugos\\EspProjects\\hello_world",
	"severity": 4,
	"message": "Cannot find \"${env:IDF_PATH}/components/\"",
	"startLineNumber": 8,
	"startColumn": 17,
	"endLineNumber": 8,
	"endColumn": 57
},{
	"resource": "/c:/Users/tugos/EspProjects/hello_world/.vscode/c_cpp_properties.json",
	"owner": "c:\\Users\\tugos\\EspProjects\\hello_world",
	"severity": 4,
	"message": "Cannot find \"${env:IDF_PATH}/components\"",
	"startLineNumber": 14,
	"startColumn": 21,
	"endLineNumber": 14,
	"endColumn": 58
}]

NgoHungCuong, 2024-07-07:
Monitor:
- V1.8.0 is buggy.
- V1.7.1 is OK.

NgoHungCuong, 2024-07-07:
Because Espressif use ' to surround the path, they need to use " the path

NgoHungCuong, 2024-07-07:
File: src/espIdf/monitor/index.ts
`const quotePath = (path) => {
      if (shellType === "PowerShell") {
        return `'${path.replace(/'/g, "''")}'`;
      } else {
        return `'${path}'`;
      }
    };`
Change to
`const quotePath = (path) => {
      if (shellType === "PowerShell") {
        return `"${path.replace(/'/g, "''")}"`;
      } else {
        return `"${path}"`;
      }
    };`

closed 2024-07-07 by TtijasS

brianignacio5, 2024-07-07:
Did you check the bug fix PR before ?

kobylin, 2024-09-17:
I fixed this problem by updating outdated module `esp-idf-panic-decoder` from `1.2.0` to `1.2.1`. There was a bug in that library.

```
cd ~/.espressif/python_env/idf5.3_py3.11_env/bin
source activate #enable correct python venv
pip install --upgrade esp-idf-panic-decoder
```

PR #1234 (2024-07-18): Fix monitoring for Command Prompt
## Description

- Fix monitoring for Command Prompt
- Fix "Git Bash" white space support
- Fix edge-case for users that have default terminal settings from older setups. We now correctly identify the shell that's been used by the user

Fix https://github.com/espressif/vscode-esp-idf-extension/issues/1230
Fix https://github.com/espressif/vscode-esp-idf-extension/issues/1233
Fix #1239 
Fix #1242

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Monitor a flashed project while having "Command Prompt" as  [... cut]