espressif/vscode-esp-idf-extension #1244: idf.postBuildTask not working in v1.8.0

policeman0077, 2024-07-07:
### OS

Windows

### Operating System version

win11

### Visual Studio Code version

1.91.0

### ESP-IDF version

4.4.5

### Python version

3.11.6

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22631 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Python311\Scripts\;C:\Python311\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Program Files\PuTTY\;C:\ProgramData\chocolatey\bin;C:\Program Files\CMake\bin;C:\Program Files\Git\cmd;C:\Program Files\qemu;C:\mingw64\bin;C:\Program Files\Meld\;C:\Program Files\ffmpeg-2024-02-04-git-7375a6ca7b-essentials_build\bin;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:<HOMEPATH>\.cargo\bin;C:\Program Files\SEGGER\JLink;C:<HOMEPATH>\AppData\Local\Programs\Python\Python39\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Roaming\Python\Python311\Scripts;C:<HOMEPATH>\.dotnet\tools;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.91.0 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.7.1 
Workspace folder c:<HOMEPATH>\Desktop\firmware\esp32_solar 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.23.1\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM4
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.23.1\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.10.2: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.3\ccache-4.3-windows-64: true
[... 158 lines cut ...]
[image]

Does the grammar for this function changed in v1.8.0? Is there any document for how to use this function? I could not find any examples of this function.

### Debug Message

```plain
here is the output of the error message: 

*  Executing task: c:\Users\Sam\Desktop\firmware\esp32_solar\build\python ota_zip_gen.py  

 *  The terminal process failed to launch: Path to shell executable "c:\Users\Sam\Desktop\firmware\esp32_solar\build\python ota_zip_gen.py" does not exist.
```

### Other Steps to Reproduce

1. create a python script
2. add "idf.postBuildTask":"python script.py", to setting.jason
3. build project
4. you will see the error message
[image]

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-07-07:
This is a duplicate of #1232 

Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1664025402.zip) ?

policeman0077, 2024-07-08:
> This is a duplicate of #1232
> 
> Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1664025402.zip) ?

Thank you. The installer you provided is working.

PR #1238 (2024-08-05): Fix custom tasks
## Description

This is a fix for the custom tasks

Fixes [#1232](https://github.com/espressif/vscode-esp-idf-extension/issues/1232)
Fix #1244 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

1. Open a project
2. Add a custom task to 
    - go to .vscode folder inside the project and open settings.json
    - add the following line: `"idf.preBuildTask": "echo 'Pre-build task executed!'"`
3. Build the project

In the Pre-build tab, you should see the following:
[image]

## How has this been tested?

As described above [... cut]
closed 2024-08-05 by radurentea