espressif/vscode-esp-idf-extension #1246: ESP_LOGI, uintXX_t undefined or syntax error in intelliSense

louisjonathan, 2024-07-08:
### OS

Windows

### Operating System version

Ubuntu 24.04 LTS

### Visual Studio Code version

1.90.1

### ESP-IDF version

5.5.2

### Python version

3.12

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 6.8.0-36-generic 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 <HOMEPATH>/.local/bin:<HOMEPATH>/.cargo/bin:<HOMEPATH>/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.90.1 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/bash 
ESP-IDF Extension version 1.8.0 
Workspace folder <HOMEPATH>/Desktop/Research/jl-research-summer-2024/Pasteables_Software/ESP32C3/CUI_V2 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320/
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/.espressif/python_env/idf5.2_py3.12_env/bin/python
Serial port (idf.port) /dev/ttyS4
OpenOCD Configs (idf.openOcdConfigs) board/esp32-wrover-kit-3.3v.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) git
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin: true
Access to <HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
[... 147 lines cut ...]
**Best Guess:**

I believe the unrecognized types are due to some error in intelliSense mistakenly thinking that the types are already defined. most of the `uintXX_t` are "undefined" to intelliSense, but defined to idf.py

[image]
[image]
[image]

There is definitely something fishy going on here. 

The ESP_LOGX commands have `expected ')'` as the reason for the "error", but again, idf.py compiles them without even a warning. 

[image]

(above code is shamelessly lifted off of [espressif's ble gap server](https://github.com/espressif/esp-idf/tree/v5.2.2/examples/bluetooth/bluedroid/ble/gatt_security_server) so it is correct, and to reiterate: `idf.py build` is successful for this project)

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-07-09:
As you might have seen, this Language configuration is not done by our extension but we provide template for the Microsoft C/C++ extension which is the one extension that provide language support.

I'm trying to reproduce your issue without success. I'm using IDF v5.2.2 with the linked example you provided and the intellisense works well for me on macOS.
[image]
[image]

louisjonathan, 2024-07-10:
sorry for the late response. I will try to move the source files to a new template, if that doesn't work, I will post here and then likely reinstall everything.

louisjonathan, 2024-07-10:
here's what I did that (I think) fixed it

ESP-IDF extension -> create new project
ESP-IDF terminal -> . ~/esp/esp-idf/export.sh
ESP-IDF terminal -> idf.py set-target esp32c3
(at this point headers like esp_bt.h are not found)
settings.json -> change 
              ``` 
              "idf.openOcdConfigs": [
                      "board/esp32-wrover-kit-3.3v.cfg"
                  ],
              ```
      to 
              ```
              "idf.openOcdConfigs": [
                      "board/esp32c3-builtin.cfg"
                  ],
              ```
ESP-IDF terminal -> idf.py menuconfig 
(re-enable all of my previous settings manually, just to be sure)
ESP-IDF terminal -> idf.py clean

(at this point all the errors are gone, all headers are found, and all types and macros are properly identified by intelliSense)

ESP-IDF terminal -> idf.py build

I will reopen this if somehow the issues come back and update it does

closed 2024-07-10 by louisjonathan