espressif/vscode-esp-idf-extension #1248: [Feature Request]: Support testing of C++ files

11CaptainKirk, 2024-07-09:
**Is your feature request related to a problem? Please describe.**
Cannot use unit testing feature with C++ source files as only c test files are detected/supported when using the "unity-app" testing from the test menu in the side bar. Per the documentation on unit testing using the extension, only files following the pattern `**/test/test_*.c` are detected.

**Describe the solution you'd like**
Some way to allow c++ test files. Examples of testing c++ files can be found [here](https://github.com/espressif/esp-idf-cxx/blob/main/test_apps/cxx_exception/main/cxx_exception.cpp)
It would be nice if it automatically detects if any c++ test files are present, and if so makes test_app_main a cpp file, otherwise just does the existing functionality.

**Describe alternatives you've considered**
It's possible to test C++ files already but not with the simple vscode tooling that exists for c files. This would probably be a very easy feature to implement.

**Additional context**

fenderle, 2025-06-16:
I also need this - plus the possibility to select qemu instead of esp. I patched the extension to

1. use the glob "**/test/test_*.cpp" instead of "**/test/test_*.c"
2. run "pytest --junitxml test.xml --skip-autoflash y --embedded-services idf,qemu -s --test-name '${t}'" instead of "esp,idf"

With these changes in place my unit tests work as I want them to.

It would be super helpful to make thos two line configurable through settings or some other kind of override.

brianignacio5, 2025-06-25:
Hi @fenderle I have added new 2 commands for both Build unit test app and flash unit test app in the PR.

Please take a look when you can

fenderle, 2025-07-09:
@brianignacio5 it took a little while, sorry for that. I created a c++23 version of the unit-test example: 

https://github.com/fenderle/unit_test_cpp

Feel free to do whatever you wish with it.

PR #1593 (2025-07-29): [VSC-1670] Pytest unit test customize services and glob pattern
## Description

Add `idf.pyTestEmbeddedServices` and `idf.unitTestFilePattern` to allow customize the pytest embedded services and the unit test file GLOB pattern to find test files.

Add `Unit Test: Build Unit Test App` and `Unit Test: Flash Unit Test App` to just build or just flash the unity-app.

This should allow cpp unit testing according to our user @fenderle. Please take a look.

This PR also fix these issue with current Run test implementation.

Check that the `${workspaceFolder}/unity-app`  exists on `Testing` tab -> `Ru [... cut]
closed 2025-07-29 by brianignacio5