espressif/vscode-esp-idf-extension #1249: idf.py coredump-debug fails to use correct cmake directory

gadget-man, 2024-07-12:
### OS

Windows

### Operating System version

macOS 14.5 (23F79

### Visual Studio Code version

1.91.0

### ESP-IDF version

5.2.2

### Python version

3.12.3

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin x64 23.5.0 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 <HOMEPATH>/.docker/bin:<HOMEPATH>/.composer/vendor/bin:/Library/Frameworks/Python.framework/Versions/3.7/bin:/Library/Frameworks/Python.framework/Versions/3.6/bin:<HOMEPATH>/Library/Python/2.7/bin:<HOMEPATH>/.nvm/versions/node/v20.11.1/bin:<HOMEPATH>/.docker/bin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin:/Library/Apple/usr/bin:/Applications/Wireshark.app/Contents/MacOS:/usr/local/sbin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.91.0 
Visual Studio Code language en 
Visual Studio Code shell /bin/zsh 
ESP-IDF Extension version 1.8.0 
Workspace folder <HOMEPATH>/Documents/iParcelBox/3-FIRMWARE/ESP-IDF_Sandbox/iParcelBox_v3 

### Description

I have set a custom build directory in the VSCode Extension settings IDF Build Path as `${workspaceFolder}/build.nosync`, which works correctly when I build the firmware.

However if I call idf.py coredump-debug, this parameter gets ignored and a new 'build' directory is created and the core dump fails. I've tried also adding Idf: Custom Extra Vars as `CMAKE_BUILD_DIRECTORY=./build.nosync`, but that makes no difference.

### Debug Message

```plain
No error message, I just see `Running cmake in directory /[path to folder]/build` and a new build directory is created, the coredump then eventually fails with `Failed to load core dump: Invalid application image for coredump: coredump SHA256(13012e635) != app SHA256().`
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-07-15:
Since you are using the command line to start a core dump debug, have you tried passing the build directory as argument `idf.py -B build.nosync coredump-debug`.

You can also just use the  `ESP-IDF: Launch IDF Monitor for CoreDump / GDB-Stub Mode` command to launch a monitor session which will start a debug on the panic event if you configured it to use Core dump.

brianignacio5, 2024-07-29:
Closing this issue for lack of user response. Comment here if you still need assistance.

closed 2024-07-29 by brianignacio5