espressif/vscode-esp-idf-extension #1250: [Feature Request]: Use prefix_map_gdbinit for debuger and monitor when `CONFIG_APP_REPRODUCIBLE_BUILD=y`

yyjdelete, 2024-07-15:
**Is your feature request related to a problem? Please describe.**
For better use the plugin when reproducible build is enabled(`CONFIG_APP_REPRODUCIBLE_BUILD=y`)
https://github.com/espressif/esp-idf/blob/master/docs/en/api-guides/reproducible-builds.rst#reproducible-builds-and-debugging

**Describe the solution you'd like**
1. If it possible to add `-x build/prefix_map_gdbinit` to `launch.json` autoly if reproducible build is enable?
2. Is it also possible to use the same file to map the stacktrace in monitor, and make vscode jump to the right file when click on stacktrace autoly?

brianignacio5, 2024-07-15:
By default the CDT debug adapter in our extension will use is:

```json
"initCommands": [
        "set remote hardware-watchpoint-limit {IDF_TARGET_CPU_WATCHPOINT_NUM}",
        "mon reset halt",
        "maintenance flush register-cache",
        "thb app_main"
      ],
```

where initCommands does the same as gdbinitFile.

You can set also the gdbinit file in the launch.json like this:

```json
gdbinitFile: "${workspaceFolder}/build/prefix_map_gdbinit",
initCommands: []
```

Please give it a try.

jon-cubik, 2025-01-21:
I just ran into this same problem.

To respond to @brianignacio5  :
> By default [...] initCommands does the same as gdbinitFile.

But that doesn't do what prefix_map_gdbinit does.  With the default settings, the debugger can't find any of my source files, because I'm using `CONFIG_APP_REPRODUCIBLE_BUILD=y`, so all the paths in my executable have been rewritten.  This means the debugger does not work - it can't show the source files so I can't use it for debugging.  The prefix_map_gdbinit file has the correct GDB configuration to make the debugger work.

> You can set also the gdbinit file in the launch.json like this:

I don't think that works either.  VS Code autocomplete doesn't think I can use a `"gdbinit"` key in the launch.json.  Even then, prefix_map_gdbinit is not a complete set of GDB init commands, it's just the commands to set the prefix mappings.  So the extra commands are (presumably) needed.

I did get this to work, by combining your "initCommands" with the GDB documentation.  Here's the launch.json I used:

```
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "gdbtarget",
      "request": "attach",
      "name": "Eclipse CDT GDB Adapter",
      "initCommands": [
        "set remote hardware-watchpoint-limit {IDF_TARGET_CPU_WATCHPOINT_NUM}",
        "mon reset halt",
        "maintenance flush register-cache",
        "thb app_main",
        "source ${workspaceFolder}/build/prefix_map_gdbinit"
      ]
    },
    {
      "type": "espidf",
      "name": "Launch",
      "request": "launch"
    }
  ]
}
```

The extension should check if `CONFIG_APP_REPRODUCIBLE_BUILD` is set, and if so then it should add the extra line to initCommands automatically - `"source ${workspaceFolder}/build/prefix_map_gdbinit"`.

(I also don't think this is a feature request.  The debugger not working is a bug!)

bryghtlabs-richard, 2025-01-30:
If ESP-IDF VSCode cannot reproducible projects automatically, could we add the fix to the ESP-IDF VSCode documentation?

>You can set also the gdbinit file in the launch.json like this:

Is there a schema for this file? Do you know where the suggested fix goes?

bryghtlabs-richard, 2025-01-30:
Thanks @jon-cubik , was able to adapt your launch.json to work for my project. Agreed that ESP-IDF VSCode should handle this.

brianignacio5, 2025-02-17:
Could you please try the fix [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/2584140586.zip) ?

You can test these changes by installing this VSIX by click menu View -> Command Palette..., type Install from VSIX and then select downloaded esp-idf-extension.vsix file to install the extension.

PR #1426 (2025-03-11): [VSC-1434] add prefix_map_gdbinit initCommands if reproducible build
## Description

Add source `${buildPath}/prefix_map_gdbinit` in CDT debug configuration provider initCommands if `CONFIG_APP_REPRODUCIBLE_BUILD` is enabled.

Fixes #1250

## Type of change

- New feature (non-breaking change which adds functionality)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Enable `CONFIG_APP_REPRODUCIBLE_BUILD` in `SDK Configuration Editor`.
2. Build flash and start debugging.
3. Observe results. Check in the debug console that `source ${ [... cut]
closed 2025-03-11 by github-actions[bot]