espressif/vscode-esp-idf-extension #1251: Error during installation instructions: cmd_ command not found

voidzero, 2024-07-16:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.91.1

### ESP-IDF version

5.2.2

### Python version

I don't know

### Doctor command output

I don't know

### Description

Hello, total newbie here. I'm just following the instructions from:
https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md

Step 1 t/m step 9: done
Step 10 says:
> Pick an ESP-IDF version to download (or find ESP-IDF in your system) and the python executable to create the virtual environment.

Ok well there is no python executable but later that's said to be ok. Then the installation falis with the following message:

```
Downloaded v5.2.2 (release version). Extracting...

Extracted C:\Users\skype\esp\esp-idf-v5.2.2.zip in C:\Users\skype\esp.

C:/Users/skype/.espressif/tools/idf-git/2.39.2/mingw64/libexec/git-core\git-submodule: line 610: cmd_: command not found
Command failed: C:\Users\skype\.espressif\tools\idf-git\2.39.2\cmd\git.exe submodule foreach --recursive git config --local core.fileMode false
C:/Users/skype/.espressif/tools/idf-git/2.39.2/mingw64/libexec/git-core\git-submodule: line 610: cmd_: command not found
HEAD is now at 3b8741b172 change(version): Update version to 5.2.2
``` 

The rest of the installation seems to run just fine, but this does pop up as an error notification, so I wanted to report it regardless.

### Debug Message

```plain
C:/Users/skype/.espressif/tools/idf-git/2.39.2/mingw64/libexec/git-core\git-submodule: line 610: cmd_: command not found
Command failed: C:\Users\skype\.espressif\tools\idf-git\2.39.2\cmd\git.exe submodule foreach --recursive git config --local core.fileMode false
C:/Users/skype/.espressif/tools/idf-git/2.39.2/mingw64/libexec/git-core\git-submodule: line 610: cmd_: command not found
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-07-19:
Sorry @voidzero I'm not able to reproduce your issue on Windows 11 here. Do you perhaps have the situation described [here](https://github.com/desktop/desktop/issues/9089#issuecomment-584652898) ?

brianignacio5, 2024-07-29:
Closing this issue for lack of user response. Comment here if you still need assistance.

closed 2024-07-29 by brianignacio5

voidzero, 2024-08-09:
My sincere apologies for not responding to this ticket in a timely fashion.

Yes Brian, it looks like that issue does describe the situation I ran into, so we can keep this issue closed - I will look into the ASLR related possible solutions. Thank you for referring me, I appreciate it. With kind regards.