espressif/vscode-esp-idf-extension #1254: The extension does not see python or pip binary in the venv while trying to install ESP-Matter

Vybo, 2024-07-19:
### OS

Windows

### Operating System version

macOS 14.5

### Visual Studio Code version

Version: 1.91.1 (Universal) Commit: f1e16e1e6214d7c44d078b1f0607b2388f29d729 Date: 2024-07-09T22:07:54.982Z Electron: 29.4.0 ElectronBuildId: 9728852 Chromium: 122.0.6261.156 Node.js: 20.9.0 V8: 12.2.281.27-electron.0 OS: Darwin arm64 23.5.0

### ESP-IDF version

5.3 or 5.22 (tried both)

### Python version

3.9.6 in venv; 3.9.6 @ /usr/bin/python3; 3.12.4 @ /opt/homebrew/bin/python3

### Doctor command output

Submitting as a file, due to the output being too large for Github.
[doctor.txt](https://github.com/user-attachments/files/16313094/doctor.txt)

### Extension

The file is empty.

### Description

When trying to install ESP-Matter extension through CSCode in a Workspace where ESP-IDF is set up, I get an error in the ouput about missing pip (and python itself, if I edit the activate.sh script to print `which python` and `which pip`. 

If I replace the `pip install` command with `pip3`, it will use the homebrew installed python3. That won't let me install the requirements, because it's user space (global?) python. That points to the fact that that activate script does not use the venv for some reason.

If I just use  `which pip` in VSCode terminal, I get the correct venv path.

### Debug Message

```plain
WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
 ▀▀    █   █    ▀▀

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (2.3s)
  Setting up Project actions........skipped (0.1s)
  Setting up Python environment.....[|]
ninja: Entering directory `/Users/dan.vybiral/esp/idf-tools/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out'
[1/589] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../.. --default-toolchain=//build/toolchain/host:mac_arm64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --capture-output --module piptools --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files python/gen/matter_build_venv._compile_requirements_metadata_path_list.txt -- compile --resolver=backtracking --allow-unsafe --output-file python/gen/matter_build_venv/compiled_requirements.txt python/gen/matter_build_venv/generated_requirements.txt ../../scripts/setup/requirements.build.txt
FAILED: python/gen/matter_build_venv/compiled_requirements.txt 
python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../.. --default-toolchain=//build/toolchain/host:mac_arm64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --capture-output --module piptools --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files python/gen/matter_build_venv._compile_requirements_metadata_path_list.txt -- compile --resolver=backtracking --allow-unsafe --output-file python/gen/matter_build_venv/compiled_requirements.txt python/gen/matter_build_venv/generated_requirements.txt ../../scripts/setup/requirements.build.txt
    error: subprocess-exited-with-error
    
    × Getting requirements to build wheel did not run successfully.
    │ exit code: 1
    ╰─> [54 lines of output]
        running egg_info
        writing lib/PyYAML.egg-info/PKG-INFO
        writing dependency_links to lib/PyYAML.egg-info/dependency_links.txt
        writing top-level names to lib/PyYAML.egg-info/top_level.txt
        Traceback (most recent call last):
          File "/Users/dan.vybiral/esp/idf-tools/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out/python-venv/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
            main()
          File "/Users/dan.vybiral/esp/idf-tools/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out/python-venv/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
            json_out['return_val'] = hook(**hook_input['kwargs'])
                                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
          File "/Users/dan.vybiral/esp/idf-tools/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out/python-venv/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
            return hook(config_settings)
                   ^^^^^^^^^^^^^^^^^^^^^
          File "/private/var/folders/pk/612ynjh177d1db0hpgyrxvym0000gn/T/pip-build-env-0b_trewk/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 327, in get_requires_for_build_wheel
            return self._get_build_requires(config_settings, requirements=[])
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

[shortened]

['ninja', '-C', '/Users/dan.vybiral/esp/idf-tools/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']

Traceback (most recent call last):
  File "/Users/dan.vybiral/esp/idf-tools/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 421, in install_packages
    subprocess.check_call(ninja_cmd, stdout=outs, stderr=outs)
  File "/opt/homebrew/Cellar/python@3.12/3.12.4/Frameworks/Python.framework/Versions/3.12/lib/python3.12/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['ninja', '-C', '/Users/dan.vybiral/esp/idf-tools/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/dan.vybiral/esp/idf-tools/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 995, in <module>
    sys.exit(main())
             ^^^^^^
  File "/Users/dan.vybiral/esp/idf-tools/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 987, in main
    return EnvSetup(**vars(parse())).setup()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/dan.vybiral/esp/idf-tools/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 577, in setup
    result = step(spin)
             ^^^^^^^^^^
  File "/Users/dan.vybiral/esp/idf-tools/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 766, in virtualenv
    if not virtualenv_setup.install(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/dan.vybiral/esp/idf-tools/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 443, in install
    install_packages(gn_target)
  File "/Users/dan.vybiral/esp/idf-tools/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 424, in install_packages
    raise subprocess.CalledProcessError(
subprocess.CalledProcessError: Command '['ninja', '-C', '/Users/dan.vybiral/esp/idf-tools/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.
Pip version:
python not found
Installing pip requirements for all...
_install_additional_pip_requirements:31: command not found: pip
```

### Other Steps to Reproduce

I'm trying to get the Swift Embeded Matter example repository to build, however I have issues with building without the Matter extension due to "swift embeded error: unknown argument: '-import-bridging-header'", which will probably be an issue of missing the required headers from the C++ library.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-07-23:
Thanks for your issue report. 

I've prepared a fix in PR #1258 Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1728641280.zip) to see if the issue is fixed?

Vybo, 2024-07-23:
@brianignacio5 Hi, thank you for looking into this. Unfortunately, I haven't been able to make it work with the extension from your PR either, although the error is now different.

Steps I took:
1. Uninstalled distributed ESP-IDF extension from VSCode
2. Installed the extension from your vsix (command pallete > install extension from vsix)
3. Restarted VSCode
4. Ran ESP-IDF configuration for the workspace

First, I tried running the 'ESP-IDF: Install ESP-Matter' command with ESP-IDF Python being configured to Homebrew version (3.12), which didn't work.

Then, I reconfigured the extension to use the systems or venv's 3.9, which didn't work, but produced a different error from my understanding.

I'm attaching both error logs.
[log.venv-system-3.9.txt](https://github.com/user-attachments/files/16347378/log.venv-system-3.9.txt)
[log_homebrew_3.12.txt](https://github.com/user-attachments/files/16347381/log_homebrew_3.12.txt)

brianignacio5, 2024-07-25:
Hi @Vybo I think is probably an issue with the system Python itself rather than vscode extension or something.

I've tested on MacOS with ESP-IDF v5.2.2 and current master from scratch and both work on my end. Did you make sure to have Matter prerequisites before ?

https://github.com/espressif/connectedhomeip/blob/v1.1-branch/docs/guides/BUILDING.md#prerequisites

Vybo, 2024-07-25:
It is possible. I was trying it on a temporary machine, so I'll try the complete from 0 setup on a fresh one, thank you. I believe I had the pre-requisities, but I'll make sure to run the install commans anyway.

Vybo, 2024-07-25:
@brianignacio5 The Esp-matter installation passed fine on the new machine. I only have the `/usr/bin/python3` available there (3.9.6) and using that one worked. Installing other stuff through pip3 also does not throw the new behaviour error like it did on the old machine. For some reason, even-though brew is installed, it does not offer its /opt/ python installation this time, which seems to have helped.

brianignacio5, 2024-08-06:
Closing this issue then. Let me know if you need more help.

closed 2024-08-06 by brianignacio5