espressif/vscode-esp-idf-extension #1257: Compiler undefined with default install

Brandon-Ballard25, 2024-07-21:
### OS

Windows

### Operating System version

OS win32 x64 10.0.22631 

### Visual Studio Code version

1.91.1 

### ESP-IDF version

 5.2.2

### Python version

3.11.2

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22631 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\jdk-21.0.2\bin;C:\Program Files\Azure Data Studio\bin;C:\Program Files (x86)\NodistSenquip\bin;C:\Program Files\Git\cmd;C:\Program Files (x86)\Nodist\bin;C:\Program Files\jdk-21.0.2\bin;C:\Program Files\Azure Data Studio\bin;C:\Program Files (x86)\Nodist\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin;C:<HOMEPATH>\.expressif\tools\idf-python\3.11.2\Scripts;C:<HOMEPATH>\.expressif\tools\idf-python\3.11.2; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.91.1 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.8.0 
Workspace folder c:<HOMEPATH>\OneDrive\Desktop\Projects\.vscode\blink 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) ${env:IDF_PATH}
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) 
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) ${env:IDF_TOOLS_PATH}
Git Path (idf.gitPath) ${env:programfiles}\Git\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) false
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to Virtual env Python Path (idf.pythonBinPath) false
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) false
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path true
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in Virtual env Python Path (idf.pythonBinPath) false
Spaces in ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version undefined
ESP-IDF version undefined
Python version undefined
Python's pip version undefined
-------------------------------------------------- Project configuration settings ----------------------------------------------------------
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages undefined
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
{
    "errno": -4058,
    "code": "ENOENT",
    "syscall": "spawn ${env:programfiles}\\Git\\cmd\\git.exe",
    "path": "${env:programfiles}\\Git\\cmd\\git.exe",
    "spawnargs": [
        "--version"
    ],
    "cmd": "${env:programfiles}\\Git\\cmd\\git.exe --version"
}
--------------------------------------------------------------------------------------------------------------------------------------------
----------------------------------------------------------- Logfile -----------------------------------------------------------------

### Description

When creating the sample 'blink' project or any other project using the ESP-IDF extension for VS Code, I encounter an issue with the c_cpp_properties.json file. Specifically, the compilerPath entry is set to ${config:idf.toolsPathWin}undefined, which causes the build process to fail with the error: "Something went wrong while trying to build the project.". I have been stuck on this for days so any help would be greatly appreciated. 

**Troubleshooting Steps Taken:**

- Reinstalled the ESP-IDF extension multiple times.
- Attempted various environment variable configurations.
- Tried uninstalling and reinstalling the ESP-IDF framework.

### Debug Message

```plain
Something went wrong while trying to build the project
```

### Other Steps to Reproduce

**Steps to Reproduce:**

- Install the ESP-IDF extension for VS Code.
- Select version 5.2.2 of the ESP-IDF.
- Choose all default settings during setup.
- Create an example project (e.g., 'blink').
- Attempt to build the project, leading to the error.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-07-22:
Your settings seems to be incorrect. In the Doctor command output, the IDF_PATH setting is pointing to ${env:IDF_PATH} which is not correct in your case. You can see the Executable versions section is all undefined because it cannot find ESP-IDF, git and Python. I don't think the setup was completed properly.

Brandon-Ballard25, 2024-07-22:
@brianignacio5 Many thanks for the reply. I will try and reinstall in an hour. How many environment variables should I have, and where should they point to if I have the default configuration?

brianignacio5, 2024-07-22:
You should make sure that `Executables Versions ` show all version number and `Check ESP-IDF python requirements.txt` show that all requirements are satisfied. Please check that the setup is correct and if any issue you can see the extension log or the View -> Output -> ESP-IDF for more information.

Brandon-Ballard25, 2024-07-22:
I don't know what's going on im at a complete loss its still undefined, I have removed all environment variables cleared all caches, removed the extension and redownloaded it and configured using the express mode with everything default i did not touch anything and it still will not work?

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22631 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\jdk-21.0.2\bin;C:\Program Files\Azure Data Studio\bin;C:\Program Files (x86)\NodistSenquip\bin;C:\Program Files\Git\cmd;C:\Program Files (x86)\Nodist\bin;C:\Program Files\jdk-21.0.2\bin;C:\Program Files\Azure Data Studio\bin;C:\Program Files (x86)\Nodist\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.91.1 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.8.0 
Workspace folder c:<HOMEPATH>\fuckyou\blink 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\v5.2.2\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.11.1;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM7
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.11.1: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64: true
Access to C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path true
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin: false
[... cut ...]
    }
  ],
  "version": 4
}
 
----------------------------------------------------------- Latest error -----------------------------------------------------------------
{
    "code": 1,
    "killed": false,
    "signal": null,
    "cmd": "C:<HOMEPATH>\\.espressif\\python_env\\idf5.2_py3.11_env\\Scripts\\python.exe C:<HOMEPATH>\\esp\\v5.2.2\\esp-idf\\tools\\check_python_dependencies.py -r c:<HOMEPATH>\\.vscode\\extensions\\espressif.esp-idf-extension-1.8.0\\esp_debug_adapter\\requirements.txt"
}

brianignacio5, 2024-07-22:
Now it seems that the extension is configured. Just delete the ` c_cpp_properties.json` file and run the `ESP-IDF: Add vscode configuration folder` to recreate the file or just delete `compilerPath` and run the `ESP-IDF: Run idf.py reconfigure task` to generate compile_commands.json

Brandon-Ballard25, 2024-07-22:
Unfortunately `ESP-IDF: Add vscode configuration folder` returned undefined for the compiler path as seen previously and i also tried to delete the `compilerPath `and run the `ESP-IDF: Run idf.py reconfigure task` this created the compile_commands.json but it still would not build

ill once again just send this for reference

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22631 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\jdk-21.0.2\bin;C:\Program Files\Azure Data Studio\bin;C:\Program Files (x86)\NodistSenquip\bin;C:\Program Files\Git\cmd;C:\Program Files (x86)\Nodist\bin;C:\Program Files\jdk-21.0.2\bin;C:\Program Files\Azure Data Studio\bin;C:\Program Files (x86)\Nodist\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.91.1 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.8.0 
Workspace folder c:<HOMEPATH>\projects\blink 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\v5.2.2\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.11.1;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM7
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.11.1: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64: true
Access to C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path true
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin: false
[... cut ...]
      "compileCommands": "${config:idf.buildPath}/compile_commands.json",
      "includePath": [
        "${config:idf.espIdfPath}/components/**",
        "${config:idf.espIdfPathWin}/components/**",
        "${workspaceFolder}/**"
      ],
      "browse": {
        "path": [
          "${config:idf.espIdfPath}/components",
          "${config:idf.espIdfPathWin}/components",
          "${workspaceFolder}"
        ],
        "limitSymbolsToIncludedHeaders": true
      }
    }
  ],
  "version": 4
}

brianignacio5, 2024-07-22:
Build ? Nothing we have done is related to build. Did you try to use the `ESP-IDF: Build your project` command ?

Brandon-Ballard25, 2024-07-22:
After doing these step i have been trying to build as well to see if anything has changed, there is no output from the view -> output  -> esp-idf. I just get

>  "Something went wrong while trying to build the project" Source: ESP-IDF

Pop up in the corner of VS code

brianignacio5, 2024-07-22:
Delete the build directory, and try again.

brianignacio5, 2024-07-22:
[image]
Build should be found and looks like this

Brandon-Ballard25, 2024-07-22:
It just says that error message but in regards to the previous message i have set the COM, and the device target and here is the log
[esp_idf_vsc_ext.log](https://github.com/user-attachments/files/16327910/esp_idf_vsc_ext.log)

brianignacio5, 2024-07-22:
So based on your extension the issue happens because there is a command `where cmake` and `where ninja` which doesn't work on your terminal. This usually means something wears it happening in your Powershell terminal and probably some commands will also not work.

Please review posts like https://learn.microsoft.com/en-us/answers/questions/218722/powershell-broke-basic-commands-not-working-not-lo

Brandon-Ballard25, 2024-07-22:
when i use `where cmake` or `where ninja` it just returns nothing e.g. 
```
PS C:\Users\25bra\projects\blink> where cmake
PS C:\Users\25bra\projects\blink> 
```
So this is likely the problem? the link suggests adding an env variable `PSModulePath` i have that already with the values they suggest which is confusing `C:\ProgramFiles\WindowsPowerShell\Modules;C:\WINDOWS\system32\WindowsPowerShell\v1.0\Modules`

brianignacio5, 2024-07-22:
Maybe some paths are missing from PATH environment variable which make these command not to work.

If you open an `ESP-IDF: Open ESP-IDF Terminal`

Brandon-Ballard25, 2024-07-22:
Nothing is returned when open the esp-idf terminal 
```
PS C:\Users\25bra\projects\blink> where cmake
PS C:\Users\25bra\projects\blink> 
```

brianignacio5, 2024-07-22:
Seems that is the issue.

Brandon-Ballard25, 2024-07-22:
Okay so I must configure env vars then

brianignacio5, 2024-07-22:
Make sure that `C:\Windows\System32` is in your environment PATH variable.

Brandon-Ballard25, 2024-07-22:
You are absolutely brilliant, thank you very much, that was it. I got it working!

brianignacio5, 2024-07-22:
If you try to delete the c_cpp_properties.json and recreate it with the `ESP-IDF: Add vscode configuration folder` should fix that too. Glad it works now !

If you find the extension useful, please leave us a review in the [Visual Studio Code Marketplace](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension) we haven't received any GitHub issue regarding setup problem as mentioned.

Brandon-Ballard25, 2024-07-22:
I will be sure to add a review. Thanks again

closed 2024-07-23 by brianignacio5