espressif/vscode-esp-idf-extension #1260: Can't build with extension 1.8. Error: "CMake Error: Could not create named generator =Ninja"

rtheil-growlink, 2024-07-23:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.9.1

### ESP-IDF version

4.3.7

### Python version

3.8.7

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Microsoft SDKs\Azure\CLI2\wbin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\Microsoft SQL Server\150\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files\dotnet\;C:<HOMEPATH>\AppData\Local\particle\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\Microsoft SQL Server\150\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files\dotnet\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\.dotnet\tools 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.91.1 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.8.0 
Workspace folder c:\repos\espfw\controller-firmware 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:<HOMEPATH>\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.16.4\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.1;C:<HOMEPATH>\.espressif\tools\ccache\3.7;C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM6
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) \Git\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.16.4\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.10.2: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.1: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\3.7: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
[... 89 lines cut ...]

### Description

Can no longer build from scratch after build directory is deleted. Error is "CMake Error: Could not create named generator =Ninja". This is when using the Build option from the vs code command palette.

This is because the plugin is now adding a = sign between the -G option and the "Ninja" value
[image]

If I go into an ESP-IDF terminal and run idf.py build, it will build and then I can use the command palette to build. This problem only started when the extension was updated to 1.8. I purposefully no longer update this tool because these kinds of minor problems are all too common, but in this case vs code auto updated it. What's strange is that I downgraded the extension to 1.7.x and the problem still exists.

Even if I try to run that command in an idf terminal and fix the -G option, it still doesn't work
[image]

This is because the -G, -B and -S options don't use the = sign and never have
[image]

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-07-24:
Hi @rtheil-growlink thank your for your bug report.

We've found out this is true from CMake < 3.20 but if you notice after ESP-IDF v4.4 we started using CMake higher than 3.20.
We have tested this and it does work on later ESP-IDF versions, so please give it a try if you can.

We usually test with ESP-IDF version currently maintained and 4.3.7 we already drop out the support. I'll provide a fix but consider that we may provide less and less support for older ESP-IDF versions in the future.

rtheil-growlink, 2024-07-24:
Thanks @brianignacio5. Unfortunately, the last time I attempted to test 4.4.x on our current setup, we encountered a lot of issues. 4.3.7 is rock solid stable, so being forced to move to a less stable version of the IDF just isn't in the cards for us at this time. We have literally thousands of production devices using firmware built on 4.3.x. It may be worth noting that v4.3.7 was released just 6 months ago.

Should I update CMake? Is that specific version installed as part of the IDF? I don't have any other projects that have specific CMake version requirements, so I'm open to other suggestions.

Thanks for the fix, as we are forced to continue LTS for firmware built on 4.3.x. Hoping that with the easy switch between IDF versions, I can take the time to go back and attempt an upgrade to 4.4.x in the future.

brianignacio5, 2024-07-25:
Completely understand @rtheil-growlink . We will work on fixing this. We used these = in order to work the issue of whitespaces in project path but as you fix something you break something. Apologies for this.

If you don't mind you could try using a newer version of CMake by updating the `idf.customExtraPaths` setting replace the old CMake container directory with the new CMake path in the meantime we prepare a fix for your use case.

radurentea, 2024-07-31:
Hi@rtheil-growlink,

I've created a fix for this issue, can you try installing the extension using this [esp-idf-extension.vsix.zip](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1759715813.zip)?

radurentea, 2024-08-01:
Hi @rtheil-growlink,

When you have a moment, please test the latest version of the extension available at: https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1765446112.zip
I've implemented a fix for the CMake issue affecting ESP-IDF versions prior to 4.4. This update should resolve the problem you've been experiencing.

Your feedback on this new version would be greatly appreciated. If you encounter any issues or have any questions, please don't hesitate to let me know.

Thank you for your patience and cooperation in helping us improve the extension.

rtheil-growlink, 2024-08-02:
ok, just ran through some testing, and it took a few extra steps to get it back to normal (delete build dir, restart vs code), and now it's building fine with 4.3.7. Thank you!

A couple issues though. One may not be related to the plugin, but I'm unsure. Maybe it's a CMAKE thing.
1. Super minor, but out of the ordinary. Now, after just building, no flash, it switches away from the Terminal tab after showing ESP-IDF Size, to the output tab. Has never done that in the past. If I build and flash, it correctly stays in the Monitor.

2. When I built our project with IDF 4.4.8, it failed, being unable to find a header file that is within the project dir. vs code shows no errors finding that file, but CMAKE does. This file is in a custom IDF component that's well defined and detected as a dependency when dependencies.lock is generated. I ran cmake --version in an ESP-IDF terminal window, and it's 3.16.4, so maybe it's cmake version related? I figure that's something relevant since you guys were saying that it should be >3.20

brianignacio5, 2024-08-05:
About (2) IDF 4.4.8, it shouldn't be related to CMake itself. In this case seems you are using system CMake or different version, in IDF the make used is 3.23.1 as shown in https://github.com/espressif/esp-idf/blob/v4.4.8/tools/tools.json#L650

About (1) the switching, there is an output for users regarding flashing, but we will remove the auto switch to output.

rtheil-growlink, 2024-08-05:
The only version of cmake on the machine I'm testing on, is the one installed by the espressif vscode plugin. Seems like when I installed 4.4.8, it should have downloaded and installed a newer version of cmake as well.

rtheil-growlink, 2024-08-06:
cmake is clearly installed in the .espressif/tools/cmake folder, but in all cases when running cmake from an ESP-IDF command prompt, it runs 3.16.4.
[image]

rtheil-growlink, 2024-08-06:
Just wiped out the .espressif and esp-idf folders and started from scratch with v4.4.8, so that fixes the cmake version issue.

Still had the build issues, so started attacking from other angles. I have figured out a way to get past the build issues in our project with v4.4.8, but it definitely seems like the espressif plugin needs a way to handle updates to tools like cmake.

Thanks for all the help, guys!

radurentea, 2024-08-06:
Hmm... let's go through this step by step:

1) Make sure the version you want to use is selected in the bottom bar.
[image]
I don't remember exactly why and when it happens, but sometimes when configuring the extension using `ESP-IDF: Configure ESP-IDF Extension`, the changes don't take effect immediately and old settings persist. Or this might have been fixed. @brianignacio5 might have more insight on this. If not, I'll try to reproduce the behavior and investigate the code to understand the reasoning.

2) Either way, you can select de version either from that highlighted area in the screenshot, either by running the command "ESP-IDF: Select Current ESP-IDF Version".

3) Try using the `ESP-IDF: Doctor Command` to check the CMake version being used inside `Custom extra paths`.

4) The CMake version you're seeing in the integrated terminal (3.16.4) is likely the system-wide installation or another version in your system PATH, not the one used by the extension.

If you still encounter issues with changing the CMake versions after trying these steps, please let me know. I'll do further testing and possibly open a new issue for it. We want to keep this separate from the initial issue that has been resolved.

closed 2024-08-06 by radurentea

rtheil-growlink, 2024-08-06:
I do not have a system-wide version of cmake installed. The only version of cmake installed, was what was installed by the espressif plugin. There is no cmake in the system path. If I type cmake from a regular old command prompt, it's not found. 

cmake only works within an ESP-IDF command prompt from within vs code. I renamed the 3.16.4 folder and then it was not found. Seems like whatever the IDF prompt path has set, was always set to the old version, and the plugin, on installation of the new cmake, is not updating the path to the new cmake version.

I did swap back and forth between IDF versions seeing if that made any difference. I even closed IDF command prompts between changes, and at one point even restarted vs code afterwards to see if maybe the IDF command prompt just didn't have the newest path.

I redid this machine maybe 6 months ago, so when I installed the IDF, it installed cmake 3.16.4, likely because I only installed IDF 4.3.7. When I installed 4.4.8 first today, it installed cmake 3.23.1.

I agree that this is definitely a separate issue from the original one. I have no problem copying this stuff and creating a new issue to help get this one fixed for future stuff. But I think this problem is as simple as when installing a new version of cmake, the path to cmake is not being updated.

brianignacio5, 2024-08-15:
The issue here is that ESP-IDF Tools are set in a configuration setting called `idf.customExtraPaths`.  Probably when installed before the CMake being referred in ESP-IDF was indeed 3.16.4 and that path was saved in `idf.customExtraPaths`. You can also look in that ESP-IDF path `$IDF_PATH/tools/tools.json` and see which CMake version is listed there.

I imagine that the version you installed before indeed referred to 3.16.4 but a newer version release would have update the CMake itself.

PR #1266 (2024-08-06): Fix CMake args for esp-idf lower than 4.4
## Description

Please include a summary of the change and which issue is fixed.

Fixes #1260 

## Type of change
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request
- Download esp-idf version 4.3.6 from https://[dl.espressif.com/dl/esp-idf/?idf=4.3](https://dl.espressif.com/dl/esp-idf/?idf=4.3) 
- Configure extension to use existing esp-idf (select the folder where 4.3.6 was installed)
- Create a project from examples (ex: hello world)
- Build the project

The project should build with no issues.

## How has this b [... cut]