espressif/vscode-esp-idf-extension #1262: I need to export esp-idf environment variables in a new PowerShell environment.

dxzzxb, 2024-07-26:
### OS

Windows

### Operating System version

win10

### Visual Studio Code version

1.85.1

### ESP-IDF version

5.1.1

### Python version

3.9 and 3.11

### Doctor command output

none

### Description

I installed esp-idf through the VSCode extension, and everything works fine. However, I want to automate a series of compilation and packaging tasks through a script. When I restart a terminal and run the command .$HOME/esp/esp-idf/export.ps1, an error occurs, stating "ERROR: C:\Espressif\python_env\idf5.1_py3.9_env\Scripts\python.exe doesn't exist! Please run the install script or 'idf_tools.py install-python-env' in order to create it." But in my C:\Espressif\python_env directory, there is only idf5.1_py3.11_env. I want to know what the problem is.

### Debug Message

```plain
PS C:\Users\admin\esp\esp-idf> .$HOME/esp/esp-idf/export.ps1
Setting IDF_PATH: C:\Users\admin\esp\esp-idf
Checking Python compatibility
Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool cmake found in PATH: 3.23.1. To use it, run 'C:\msys64\mingw64\bin\python.exe C:\Users\admin\esp\esp-idf\tools\idf_tools.py export --prefer-system'

Name                           Value
----                           -----
OPENOCD_SCRIPTS                C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\share\openocd\s...
IDF_CCACHE_ENABLE              1
ESP_ROM_ELF_DIR                C:\Espressif\tools\esp-rom-elfs\20230320\
ESP_IDF_VERSION                5.1
IDF_DEACTIVATE_FILE_PATH       C:\Users\admin\AppData\Local\Temp\tmpzm509wmpidf_39248

Added to PATH
-------------
C:\Users\admin\esp\esp-idf\components\app_update
C:\Users\admin\esp\esp-idf\components\espcoredump
C:\Users\admin\esp\esp-idf\components\partition_table
C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin
C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin
C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin
C:\Espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin
C:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin
C:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin
C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin
C:\Espressif\tools\cmake\3.24.0\bin
C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin
C:\Espressif\tools\ninja\1.10.2\
C:\Espressif\tools\idf-exe\1.0.3\
C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64
C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64
C:\Users\admin\esp\esp-idf\tools
Checking if Python packages are up to date...
ERROR: C:\Espressif\python_env\idf5.1_py3.9_env\Scripts\python.exe doesn't exist! Please run the install script or "idf_tools.py install-python-env" in order to create it

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:
    idf.py build
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-07-29:
Could you share the doctor command output as requested in the issue template.

Is `C:\Espressif` the same IDF_TOOLS_PATH from your vscode configuration? Only way we could know is with the doctor command output.

brianignacio5, 2024-08-12:
Closing this for lack of response.

closed 2024-08-12 by brianignacio5