espressif/vscode-esp-idf-extension #1265: menuconfig button not relates on selected project configuration (multi_config project)

executer-uno, 2024-07-29:
### OS

Windows

### Operating System version

Windows 10 Pro

### Visual Studio Code version

1.91.1

### ESP-IDF version

5.1.4

### Python version

3.11.2

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\PuTTY\;C:\Program Files\IVI Foundation\VISA\Win64\bin;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\bin;C:\Program Files (x86)\IVI Foundation\IVI\bin;C:\Program Files\IVI Foundation\IVI\bin;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\msys64\mingw64\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:\CMake\bin;D:\Downloads\make-3.81-bin\bin; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.91.1 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.8.0 
Workspace folder d:\Documents\VSC_Workspace\IDF_multi_config2\multi_config 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) c:<HOMEPATH>\esp\v5.1\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) c:\esp-idf5.1\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin;c:\esp-idf5.1\tools\riscv32-esp-elf-gdb\12.1_20231023\riscv32-esp-elf-gdb\bin;c:\esp-idf5.1\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;c:\esp-idf5.1\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;c:\esp-idf5.1\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;c:\esp-idf5.1\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;c:\esp-idf5.1\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;c:\esp-idf5.1\tools\cmake\3.24.0\bin;c:\esp-idf5.1\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin;c:\esp-idf5.1\tools\ninja\1.10.2;c:\esp-idf5.1\tools\idf-exe\1.0.3;c:\esp-idf5.1\tools\ccache\4.8\ccache-4.8-windows-x86_64;c:\esp-idf5.1\tools\dfu-util\0.11\dfu-util-0.11-win64;c:\esp-idf5.1\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: c:\esp-idf5.1\tools\openocd-esp32\v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: c:\esp-idf5.1\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) c:\esp-idf5.1\python_env\idf5.1_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) c:\esp-idf5.1
Git Path (idf.gitPath) c:\esp-idf5.1\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to c:\esp-idf5.1\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin: true
Access to c:\esp-idf5.1\tools\riscv32-esp-elf-gdb\12.1_20231023\riscv32-esp-elf-gdb\bin: true
Access to c:\esp-idf5.1\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin: true
Access to c:\esp-idf5.1\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin: true
Access to c:\esp-idf5.1\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin: true
Access to c:\esp-idf5.1\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin: true
Access to c:\esp-idf5.1\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to c:\esp-idf5.1\tools\cmake\3.24.0\bin: true
Access to c:\esp-idf5.1\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin: true
Access to c:\esp-idf5.1\tools\ninja\1.10.2: true
Access to c:\esp-idf5.1\tools\idf-exe\1.0.3: true
[... 259 lines cut ...]
```plain
NA
```

### Other Steps to Reproduce

There is related issue [1202](https://github.com/espressif/vscode-esp-idf-extension/issues/1202) where problem behaviour described, actually, as one I would got. And only difference I noticed - I uses Windows and there is Linux used.
There is fresh workaround and pull request that supposed to fix related issue 1202  [PR1252](https://github.com/espressif/vscode-esp-idf-extension/pull/1252), so I expected it fixes mine. And I try it:
1. project full clean on both configurations, so no build folders left
2. uninstall ESP-IDF Extension and restart extensions after
3. via command "Install from VSIX" installs provided vsix from PR1252 https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1710895911.zip
4. One issue gone - no more default "build" folder created when menuconfig called.
5. No issue behaviour changed - sdkconfig file still sticks to configuration was chosen on VSC startup and not depends on actual Project Configuration selector. VSC command "ESP-IDF: Dispose Current SDK Configuration Editor Server Process" still helps after project configuration selector changed.  And sdkconfig filename still used instead of indexed files as sdkconfig1 and sdkconfig2.

There was a note about original root CMakeLists.txt line
set(SDKCONFIG "${CMAKE_BINARY_DIR}/sdkconfig")
to be commented out. So I try it with project full clean:
1. It fixes sdkconfig filename. Now proper indexed files sdkconfig1 and sdkconfig2 used as esp_idf_project_configuration.json requires.
2. But it not fixes dynamic change over the configurations. Still sticks to one selected on startup, and command "ESP-IDF: Dispose Current SDK Configuration Editor Server Process" requires after new configuration selection. All the rest looks fine.

So, PR1252 fixes almost all, but command "ESP-IDF: Dispose Current SDK Configuration Editor Server Process" required after configuration selection changed.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-08-06:
Could you try the latest [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1775943251.zip) ?

executer-uno, 2024-08-07:
Thanks a lot.
That fix helps. I made a test and confirms it fixes all my problems.

The problem with additional command "ESP-IDF: Dispose Current SDK Configuration Editor Server Process" required after configuration change - gone.

Proper filenames and folders from esp_idf_project_configuration.json used both for build and for menuconfig - done.

Also it fixes root's CMakeLists.txt line set(SDKCONFIG "${CMAKE_BINARY_DIR}/sdkconfig") to be commented out. Now it only forces sdkconfig filename but folder name it stored in is from esp_idf_project_configuration.json as expected. But I would not recomend to make such self-conflicted configuration, just set sdkconfig filename only in esp_idf_project_configuration.json, do not set it in CMakeLists.txt. (It still works but it sticks somewhere in temporaty folders (not in build one) and may be released only by cleaning project manually after you comment out SDKCONFIG variable in CMakeLists) - looks fine and acceptable.

PR #1252 (2024-10-11): fix support for multiple sdkconfig files
## Description

Fix support of multiple sdkconfig when using Project Configuration and multiple profiles.

Fixes #1202
Fixes #1265
Fixes #1283
Fixes #1299

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Using a project with multiple configurations like https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/multi_config run the `ESP-IDF: Open Project Configuration` and configure 2 profiles with different SDK Configuration Edito [... cut]
closed 2024-08-07 by executer-uno