espressif/vscode-esp-idf-extension #1268: Debugging ESP32S2 not working on MAC

arminbreiteneder, 2024-08-01:
### OS

Windows

### Operating System version

14.4.1

### Visual Studio Code version

1.91.1

### ESP-IDF version

4.4.6

### Python version

3.12.4

### Doctor command output

{"message":"OpenOCD server failed to start because of Error: Failed to get flash maps (-6)!\n❌ Error: Failed to get flash maps (-6)!\nWarn : Application image is invalid! Check configured binary flash offset 'appimage_offset'.\nWarn : Failed to get flash mappings (-4)!\n","stack":"Error: OpenOCD server failed to start because of Error: Failed to get flash maps (-6)!\n\tat Socket.<anonymous> (/dist/extension.js:2:1492804)\n\tat Socket.emit (node:events:514:28)\n\tat addChunk (node:internal/streams/readable:376:12)\n\tat readableAddChunk (node:internal/streams/readable:349:9)\n\tat Readable.push (node:internal/streams/readable:286:10)\n\tat Pipe.onStreamRead (node:internal/stream_base_commons:190:23)","level":"error","timestamp":"2024-08-01T05:42:17.946Z"}
{"message":"OpenOCD server failed to start because of Error: Failed to get flash maps (-6)!\n❌ Error: Failed to get flash maps (-6)!\nWarn : Application image is invalid! Check configured binary flash offset 'appimage_offset'.\nWarn : Failed to get flash mappings (-4)!\n","stack":"Error: OpenOCD server failed to start because of Error: Failed to get flash maps (-6)!\n\tat Socket.<anonymous> (/dist/extension.js:2:1492804)\n\tat Socket.emit (node:events:514:28)\n\tat addChunk (node:internal/streams/readable:376:12)\n\tat readableAddChunk (node:internal/streams/readable:349:9)\n\tat Readable.push (node:internal/streams/readable:286:10)\n\tat Pipe.onStreamRead (node:internal/stream_base_commons:190:23)","level":"error","timestamp":"2024-08-01T05:42:44.717Z"}
{"message":"OpenOCD server failed to start because of Error: Failed to get flash maps (-6)!\n❌ Error: Failed to get flash maps (-6)!\nWarn : Application image is invalid! Check configured binary flash offset 'appimage_offset'.\nWarn : Failed to get flash mappings (-4)!\n","stack":"Error: OpenOCD server failed to start because of Error: Failed to get flash maps (-6)!\n\tat Socket.<anonymous> (/dist/extension.js:2:1492804)\n\tat Socket.emit (node:events:514:28)\n\tat addChunk (node:internal/streams/readable:376:12)\n\tat readableAddChunk (node:internal/streams/readable:349:9)\n\tat Readable.push (node:internal/streams/readable:286:10)\n\tat Pipe.onStreamRead (node:internal/stream_base_commons:190:23)","level":"error","timestamp":"2024-08-01T05:42:44.900Z"}
{"message":"Cannot access filePath: ${env:ADF_PATH}","stack":"Error: ENOENT: no such file or directory, access '${env:ADF_PATH}'\n\tat Object.access (node:internal/fs/sync:42:11)\n\tat Object.accessSync (node:fs:248:10)\n\tat t.accessSync (node:electron/js2c/node_init:2:8888)\n\tat Se (/dist/extension.js:2:1461757)\n\tat /dist/extension.js:2:1685466\n\tat Fs (/dist/extension.js:2:1686534)\n\tat async /dist/extension.js:2:1902761","level":"error","timestamp":"2024-08-01T05:43:22.523Z"}
{"message":"Cannot access filePath: ${env:MDF_PATH}","stack":"Error: ENOENT: no such file or directory, access '${env:MDF_PATH}'\n\tat Object.access (node:internal/fs/sync:42:11)\n\tat Object.accessSync (node:fs:248:10)\n\tat t.accessSync (node:electron/js2c/node_init:2:8888)\n\tat Se (/dist/extension.js:2:1461757)\n\tat /dist/extension.js:2:1685644\n\tat Fs (/dist/extension.js:2:1686534)\n\tat async /dist/extension.js:2:1902761","level":"error","timestamp":"2024-08-01T05:43:22.523Z"}
{"message":"Cannot access filePath: ${env:ESP_MATTER_PATH}","stack":"Error: ENOENT: no such file or directory, access '${env:ESP_MATTER_PATH}'\n\tat Object.access (node:internal/fs/sync:42:11)\n\tat Object.accessSync (node:fs:248:10)\n\tat t.accessSync (node:electron/js2c/node_init:2:8888)\n\tat Se (/dist/extension.js:2:1461757)\n\tat /dist/extension.js:2:1685736\n\tat Fs (/dist/extension.js:2:1686534)\n\tat async /dist/extension.js:2:1902761","level":"error","timestamp":"2024-08-01T05:43:22.523Z"}
{"message":"Cannot access filePath: ${env:HOMEKIT_PATH}","stack":"Error: ENOENT: no such file or directory, access '${env:HOMEKIT_PATH}'\n\tat Object.access (node:internal/fs/sync:42:11)\n\tat Object.accessSync (node:fs:248:10)\n\tat t.accessSync (node:electron/js2c/node_init:2:8888)\n\tat Se (/dist/extension.js:2:1461757)\n\tat /dist/extension.js:2:1685832\n\tat Fs (/dist/extension.js:2:1686534)\n\tat async /dist/extension.js:2:1902761","level":"error","timestamp":"2024-08-01T05:43:22.524Z"}
{"message":"Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /dist/extension.js:2:1464939\n\tat ChildProcess.exithandler (node:child_process:415:7)\n\tat ChildProcess.emit (node:events:514:28)\n\tat maybeClose (node:internal/child_process:1105:16)\n\tat ChildProcess._handle.onexit (node:internal/child_process:305:5)","level":"error","timestamp":"2024-08-01T05:43:22.962Z"}
{"message":"Cannot access filePath: ${env:ADF_PATH}","stack":"Error: ENOENT: no such file or directory, access '${env:ADF_PATH}'\n\tat Object.access (node:internal/fs/sync:42:11)\n\tat Object.accessSync (node:fs:248:10)\n\tat t.accessSync (node:electron/js2c/node_init:2:8888)\n\tat Se (/dist/extension.js:2:1461757)\n\tat /dist/extension.js:2:1685466\n\tat Fs (/dist/extension.js:2:1686534)\n\tat async /dist/extension.js:2:1902761","level":"error","timestamp":"2024-08-01T05:43:56.967Z"}
{"message":"Cannot access filePath: ${env:MDF_PATH}","stack":"Error: ENOENT: no such file or directory, access '${env:MDF_PATH}'\n\tat Object.access (node:internal/fs/sync:42:11)\n\tat Object.accessSync (node:fs:248:10)\n\tat t.accessSync (node:electron/js2c/node_init:2:8888)\n\tat Se (/dist/extension.js:2:1461757)\n\tat /dist/extension.js:2:1685644\n\tat Fs (/dist/extension.js:2:1686534)\n\tat async /dist/extension.js:2:1902761","level":"error","timestamp":"2024-08-01T05:43:56.967Z"}
{"message":"Cannot access filePath: ${env:ESP_MATTER_PATH}","stack":"Error: ENOENT: no such file or directory, access '${env:ESP_MATTER_PATH}'\n\tat Object.access (node:internal/fs/sync:42:11)\n\tat Object.accessSync (node:fs:248:10)\n\tat t.accessSync (node:electron/js2c/node_init:2:8888)\n\tat Se (/dist/extension.js:2:1461757)\n\tat /dist/extension.js:2:1685736\n\tat Fs (/dist/extension.js:2:1686534)\n\tat async /dist/extension.js:2:1902761","level":"error","timestamp":"2024-08-01T05:43:56.968Z"}
{"message":"Cannot access filePath: ${env:HOMEKIT_PATH}","stack":"Error: ENOENT: no such file or directory, access '${env:HOMEKIT_PATH}'\n\tat Object.access (node:internal/fs/sync:42:11)\n\tat Object.accessSync (node:fs:248:10)\n\tat t.accessSync (node:electron/js2c/node_init:2:8888)\n\tat Se (/dist/extension.js:2:1461757)\n\tat /dist/extension.js:2:1685832\n\tat Fs (/dist/extension.js:2:1686534)\n\tat async /dist/extension.js:2:1902761","level":"error","timestamp":"2024-08-01T05:43:56.968Z"}
{"message":"Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /dist/extension.js:2:1464939\n\tat ChildProcess.exithandler (node:child_process:415:7)\n\tat ChildProcess.emit (node:events:514:28)\n\tat maybeClose (node:internal/child_process:1105:16)\n\tat ChildProcess._handle.onexit (node:internal/child_process:305:5)","level":"error","timestamp":"2024-08-01T05:43:57.343Z"}
{"message":"OpenOCD server failed to start because of Error: Failed to get flash maps (-6)!\n❌ Error: Failed to get flash maps (-6)!\nWarn : Application image is invalid! Check configured binary flash offset 'appimage_offset'.\nWarn : Failed to get flash mappings (-4)!\n","stack":"Error: OpenOCD server failed to start because of Error: Failed to get flash maps (-6)!\n\tat Socket.<anonymous> (/dist/extension.js:2:1492804)\n\tat Socket.emit (node:events:514:28)\n\tat addChunk (node:internal/streams/readable:376:12)\n\tat readableAddChunk (node:internal/streams/readable:349:9)\n\tat Readable.push (node:internal/streams/readable:286:10)\n\tat Pipe.onStreamRead (node:internal/stream_base_commons:190:23)","level":"error", [... cut ...]
The problem I have is that it is just a solution for my PC and not a solution that ca be easily be adopted when the project is cloned from someone else. Is there a possibility to feed this command into the openocd using the extension or the launch.json somehow?

Where is the correct place of the esp appimage_offset 0x20000 command?

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-08-05:
You can try set `"idf.openOcdLaunchArgs"; ["-c", "appimage_offset 0x20000"]` to add for openOCD launch arguments.

brianignacio5, 2024-08-14:
You can also try 

```json
"target": {
        "connectCommands": [
          "set remotetimeout 20",
          "-target-select extended-remote localhost:3333",
          "esp appimage_offset 0x20000"
        ]
      }
```

in your project launch.json

arminbreiteneder, 2024-08-19:
> You can try set `"idf.openOcdLaunchArgs": [ "-c", "init", "halt", "esp appimage_offset 0x20000" ]` to add for openOCD launch arguments.

When I try this I get the error: Unexpected command line argument: halt

> You can try this instead
> 
> ```json
> "target": {
>         "connectCommands": [
>           "set remotetimeout 20",
>           "-target-select extended-remote localhost:3333",
>           "esp appimage_offset 0x20000"
>         ]
>       }
> ```
> 
> in your project launch.json @arminbreiteneder

This leads to: Undefined command: "esp".  Try "help".

> This is working for me. We need to send `init` and `reset halt` commands before setting the new app offset.
> 
> ```
>   "idf.openOcdLaunchArgs": [
>     "-c",
>     "init",
>     "-c",
>     "reset halt",
>     "-c",
>     "esp appimage_offset 0x20000"
>   ],
> ```

This seems to be the only working solution. Thanks! 👍

closed 2024-09-09 by github-actions[bot]