espressif/vscode-esp-idf-extension #1274: Opening the port list is always slow

Tinyu-Zhao, 2024-08-08:
### OS

Windows

### Operating System version

macOS 12.7.6

### Visual Studio Code version

1.90.0

### ESP-IDF version

5.4.0

### Python version

3.12.3

### Doctor command output

[report.txt](https://github.com/user-attachments/files/16537319/report.txt)

### Description

Just like `ls /dev` or platformio, it shows the serial ports on the computer in one second.

https://github.com/user-attachments/assets/64fd64d7-48a2-401d-9e22-538560c2d341

https://github.com/user-attachments/assets/58a8c605-b111-44e6-b824-e462634b55d5

### Debug Message

```plain
Every time you click the Select Serial Port command (especially the first time), loading the serial port list is slow, the ls /dev command and platformio are very fast, complete in 1 second, I think this is a problem.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-08-08:
The thing is that now we added the feature that shows the chip id on the selection list (you can see that the ESP32 chip name is shown in list) which has to run a esptool.py chip_id to get the chip name from device, which makes the command slower than just listing the serial devices.

We could make such chip_id call optional if you don't find it valuable for your use case, but it was a feature request we received.

Tinyu-Zhao, 2024-08-08:
> The thing is that now we added the feature that shows the chip id on the selection list (you can see that the ESP32 chip name is shown in list) which has to run a esptool.py chip_id to get the chip [...]

Thank you for your reply, I think it is wise to let users choose, the wait for switching devices can sometimes be very long.

brianignacio5, 2024-08-13:
Hi @Tinyu-Zhao Could you please try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1794649002.zip) ? 

We added 2 optional configuration settings: `idf.enableSerialPortChipIdRequest` and `idf.useSerialPortVendorProductFilter`. Both are `true` by default, if you set the first one to `false` it will disable that esptool chip request we discussed before. The second one uses a filter to show only Espressif chips based on usb vendorID and productID fields.

Tinyu-Zhao, 2024-08-13:
Thank you for your quick response, but there seems to be a problem. When both are checked or only "Enable Serial Port Chip ID Request" is checked, the external devices on the computer can be displayed.

https://github.com/user-attachments/assets/0b3429b3-7a4b-4e61-abdd-054fccd426e5

https://github.com/user-attachments/assets/718bf19c-a227-403f-a87e-e592126fe15a

But there seems to be an abnormality in the combination of these options:

1. Uncheck Enable Serial Port Chip ID Request, the display list will not pop up

https://github.com/user-attachments/assets/9f998ccf-f807-4fcb-97d9-df4b989e2c8b

2. Uncheck both at the same time, the display list will not pop up

https://github.com/user-attachments/assets/61e6ffbf-5f95-46e2-8edb-eae04806d0f1

brianignacio5, 2024-08-13:
Could you share the extension log ? maybe there is an error printed there.

I'll take a look. Seems to be working for me on the local setup. It does seems that the filter is kept regardless of setting value and some error in the chip_id being disabled.

Tinyu-Zhao, 2024-08-13:
> Could you share the extension log ? maybe there is an error printed there.
> 
> I'll take a look. Seems to be working for me on the local setup. It does seems that the filter is kept regardless of setting value and some error in the chip_id being disabled.

Is the expansion log viewed here? If so, there is no output when I click to select the serial port.
[image]

brianignacio5, 2024-08-13:
The extension log is here:

Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

Tinyu-Zhao, 2024-08-13:
> espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log

Thank you. I find it.
[esp_idf_vsc_ext.log](https://github.com/user-attachments/files/16596558/esp_idf_vsc_ext.log)

brianignacio5, 2024-08-14:
I've made some fixes to the PR.

Please try this new [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1805589306.zip) @Tinyu-Zhao

Tinyu-Zhao, 2024-08-14:
> I've made some fixes to the PR.
> 
> Please try this new [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1805589306.zip) @Tinyu-Zhao

It works very well, thank you.

I have another suggestion, I don't know if it is reasonable. 

The serial port chip CH9102 will have two serial ports in the MAC environment, namely wchusbserialxxxxx and usbmodemxxxx. 
In fact, only wchusbserialxxxxx can be used for burning and serial port monitor, usbmodemxxxx can only be used as a serial port monitor.
Some friends who don't know this problem may select usbmodemxxxx and cannot download. 
Can we also add an option to filter out usbmodemxxxx when usbmodemxxxx and wchusbserialxxxxx appear at the same time?
[image]

brianignacio5, 2024-08-14:
Maybe it makes more sense to move the filter list as vscode configuration setting.

This is the one list I've seen from ESP-Launchpad:
```json
[
    { vendorId: 0x10c4, productId: 0xea60 } /* CP2102/CP2102N */,
    { vendorId: 0x0403, productId: 0x6010 } /* FT2232H */,
    {
      vendorId: 0x303a,
      productId: 0x1001,
    } /* Espressif USB_SERIAL_JTAG */,
    {
      vendorId: 0x303a,
      productId: 0x1002,
    } /* Espressif esp-usb-bridge firmware */,
    { vendorId: 0x303a, productId: 0x0002 } /* ESP32-S2 USB_CDC */,
    { vendorId: 0x303a, productId: 0x0009 } /* ESP32-S3 USB_CDC */,
    { vendorId: 0x1a86, productId: 0x55d4 } /* CH9102F */,
    { vendorId: 0x1a86, productId: 0x7523 } /* CH340T */,
    { vendorId: 0x0403, productId: 0x6001 } /* FT232R */,
  ]
```
I don't know all serial ports products and their vendorId and Product ID used by Espressif devices. And there might be user custom ones. What do you think ? We could add this as a setting and allow to add filter by use case.

Tinyu-Zhao, 2024-08-14:
> Maybe it makes more sense to move the filter list as vscode configuration setting.
> 
> This is the one list I've seen from ESP-Launchpad:
> 
> ```
>  [
>     { vendorId: 0x10c4, productId: 0xea60 } /* CP2102/CP2102N */,
>     { vendorId: 0x0403, productId: 0x6010 } /* FT2232H */,
>     {
>       vendorId: 0x303a,
>       productId: 0x1001,
>     } /* Espressif USB_SERIAL_JTAG */,
>     {
>       vendorId: 0x303a,
>       productId: 0x1002,
>     } /* Espressif esp-usb-bridge firmware */,
>     { vendorId: 0x303a, productId: 0x0002 } /* ESP32-S2 USB_CDC */,
>     { vendorId: 0x303a, productId: 0x0009 } /* ESP32-S3 USB_CDC */,
>     { vendorId: 0x1a86, productId: 0x55d4 } /* CH9102F */,
>     { vendorId: 0x1a86, productId: 0x7523 } /* CH340T */,
>     { vendorId: 0x0403, productId: 0x6001 } /* FT232R */,
>   ]
> ```
> 
> I don't know all serial ports products and their vendorId and Product ID used by Espressif devices. And there might be user custom ones. What do you think ? We could add this as a setting and allow to add filter by use case.

I think your idea of ​​giving users more flexibility in custom configurations is great.

PR #1276 (2024-09-13): Filter serial devices by default and allow user to disable chip id in serial port list

## Description

Add `idf.enableSerialPortChipIdRequest` and `idf.useSerialPortVendorProductFilter` which are set true by default.

Now the serial ports listed with the **ESP-IDF: Select port to use** command will only show know USB serial ports by filtering them with productID and vendorID.  This can be disabled with `"idf.useSerialPortVendorProductFilter": false` in settings.json

esptool.py chip_id is executed when running the previous command to get the chip name and show it in the list. This can be disa [... cut]
closed 2024-09-13 by brianignacio5