espressif/vscode-esp-idf-extension #1275: ParserError when opening "ESP-IDF: Monitor Device"

gustavosousa2208, 2024-08-08:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.92.0

### ESP-IDF version

5.3

### Python version

3.11.2

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22631 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Alacritty\;C:\Program Files\Common Files\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\usbipd-win\;C:\Program Files\Git\cmd;C:\Program Files\MATLAB\R2023a\runtime\win64;C:\Program Files\MATLAB\R2023a\bin;D:\Program Files\MATLAB\R2023a\runtime\win64;C:\Program Files\playit_gg\bin\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python312;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps\;C:\Program Files\Neovim\bin;C:\Program Files\PowerShell\7\;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Program Files\Alacritty\;C:\Program Files\Common Files\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\usbipd-win\;C:\Program Files\Git\cmd;C:\Program Files\MATLAB\R2023a\runtime\win64;C:\Program Files\MATLAB\R2023a\bin;D:\Program Files\MATLAB\R2023a\runtime\win64;C:\Program Files\playit_gg\bin\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python312;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps\;C:\Program Files\Neovim\bin;C:\Program Files\PowerShell\7\;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Program Files\GNU Octave\Octave-9.2.0;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;D:\intelFPGA_lite\23.1std\questa_fse\win64 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.92.0 
Visual Studio Code language en 
Visual Studio Code shell C:\Program Files\PowerShell\7\pwsh.exe 
ESP-IDF Extension version 1.8.0 
Workspace folder c:<HOMEPATH>\OneDrive\Documentos\Projetos_ESP-IDF\hello_hello_53 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\v5.3\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.38_20240113\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.11.1;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20240305
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20240305/
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM24
OpenOCD Configs (idf.openOcdConfigs) board/esp32s3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.38_20240113\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.11.1: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64: true
[... 358 lines cut ...]
      "type": "shell",
      "command": "& ${config:idf.pythonBinPath} ${config:idf.espIdfPath}/tools/idf.py -p ${config:idf.port} monitor",
      "windows": {
        "command": "${config:idf.pythonBinPathWin} ${config:idf.espIdfPathWin}\\tools\\idf.py -p ${config:idf.portWin} monitor",
        "options": {
          "env": {
            "PATH": "${env:PATH};${config:idf.customExtraPaths}"
          }
        }
      },
```

### Debug Message

```plain
ParserError: 
Line |
   1 |  … python.exe' 'C:\Users\gusta\esp\v5.3\esp-idf\tools\idf_monitor.py' -p …
     |                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     | Unexpected token ''C:\Users\gusta\esp\v5.3\esp-idf\tools\idf_monitor.py'' in expression or statement.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-08-08:
This is an issue with your specific terminal (pwsh.exe), if you try regular Powershell it should work. 

Covering all kinds of terminal formatting is hard. We should make a fix for your case very soon.

FYI Tasks.json doesn't work because is not related to the extension code that run the IDF Monitor.

radurentea, 2024-08-12:
Hi @gustavosousa2208,

Could you try testing the following fix?
[esp-idf-extension.vsix.zip](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1801753187.zip)

Thanks!

gustavosousa2208, 2024-08-13:
sorry for the delay in replying, it worked great!

[image]

PR #1279 (2024-10-01): Fix for pwsh users
## Description

Fixes monitoring for users that have Powershell executable as pwsh.exe

Fixes #1275

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

1. Download the installer:
Go to the official PowerShell GitHub repository release page: https://github.com/PowerShell/PowerShell/releases
Look for the latest stable release and download the appropriate installer for your system (e.g., PowerShell-7.3.x-win-x64.msi for 64-bit Windows).
2. Run the installer:
Double-click the downloaded MSI file and follow the installatio [... cut]
closed 2024-08-13 by gustavosousa2208