espressif/vscode-esp-idf-extension #1281: QEMU emulated programm not updated when, changing, rebuilding the project and then restarting the QEMU Server

Voidler-cell, 2024-08-15:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.92.1

### ESP-IDF version

5.3.0

### Python version

3.12.1

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22631 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Eclipse Adoptium\jdk-17.0.11.9-hotspot\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\AMD\AMDuProf\bin;C:\Program Files\dotnet\;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:<HOMEPATH>\AppData\Local\Programs\Python\Python312\Scripts\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python312\;C:<HOMEPATH>\AppData\Local\Programs\Python\Launcher\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:\msys64\usr\bin;C:\msys64\mingw64\bin;C:\Program Files (x86)\Dr. Memory\bin\;C:\msys64\clang64\bin;C:<HOMEPATH>\.dotnet\tools;C:\raylib\raylib\src;C:<HOMEPATH>\apache-maven-3.9.6-bin\apache-maven-3.9.6\bin;C:\Program Files\zig\;C:<HOMEPATH>\.espressif\tools\qemu-xtensa\esp_develop_8.2.0_20240122\qemu\bin;C:<HOMEPATH>\.espressif\tools\qemu-riscv32\esp_develop_8.2.0_20240122\qemu\bin; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.92.1 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.8.0 
Workspace folder e:\codegit\bed_cooler 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\v5.3\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.38_20240113\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.11.1;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20240305
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20240305/
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.38_20240113\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.11.1: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64: true
[... 158 lines cut ...]
- Delete the "build/merged_qemu.bin" file
- Restart the QEMU Server
- View the updated correct output

Expected:
It would be expected, that the QEMU emulation, would update with the rebuild/restart, without needing to delete a build file.

### Debug Message

```plain
None
```

### Other Steps to Reproduce

I am not running inside a remote container.
I just added the path to the qemu-system-xtensa to my PATH variables.

When starting the server without any build/merged_qemu.bin file already pressent, a "Merge binaries task is done (thunder emojy)" message is displayed, that is missing when starting with rebuild changes, but an already existing build/merged_qemu.bin file.

I expect, that the Launch QEMU Server only creates a new merged_qemu.bin file, if no other exists. Setting this to always overwrite it , could be a quick fix.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-08-15:
Thank you for your issue report. We will fix this behavior as soon as possible.

brianignacio5, 2024-08-22:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1819614993.zip)?

Voidler-cell, 2024-08-22:
Yes, so I tried it and it did not work.

But then I tried the version at [the linked pull request](https://github.com/espressif/vscode-esp-idf-extension/pull/1282) and it worked.

I tried the installer here a few more times, in case I installed it wrong, but it seems to not work.

But thank you for the fix.

PR #1282 (2024-10-28): [VSC-1473/1495] update dockerfile and qemu options
## Description

Update Dockerfile and QEMU options.:

- Remove adding QEMU in Dockerfile and using the one from ESP-IDF tools. 
- Adding QEMU options to support both esp32 and esp32c3 as target.
- Add `${idf.buildPath}/merged-qemu.bin` delete when `${idf.buildPath}/.bin_timestamp` changes to trigger rebuilding the merged QEMU bin image.

Fixes #1281
Fixes #1320

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to tes [... cut]
closed 2024-09-30 by github-actions[bot]