espressif/vscode-esp-idf-extension #1283: Multi config and IntelliSense configuration issues

executer-uno, 2024-08-18:
### OS

Windows

### Operating System version

Windows 10 Pro

### Visual Studio Code version

1.92.2

### ESP-IDF version

5.1.4

### Python version

3.11.7

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\PuTTY\;C:\Program Files\IVI Foundation\VISA\Win64\bin;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\bin;C:\Program Files (x86)\IVI Foundation\IVI\bin;C:\Program Files\IVI Foundation\IVI\bin;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\msys64\mingw64\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:\CMake\bin;D:\Downloads\make-3.81-bin\bin; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.92.2 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.8.0 
Workspace folder d:\Documents\VSC_Workspace\IDF_multi_config2\multi_config 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) c:<HOMEPATH>\esp\v5.1\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) c:\esp-idf5.1\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin;c:\esp-idf5.1\tools\riscv32-esp-elf-gdb\12.1_20231023\riscv32-esp-elf-gdb\bin;c:\esp-idf5.1\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;c:\esp-idf5.1\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;c:\esp-idf5.1\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;c:\esp-idf5.1\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;c:\esp-idf5.1\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;c:\esp-idf5.1\tools\cmake\3.24.0\bin;c:\esp-idf5.1\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin;c:\esp-idf5.1\tools\ninja\1.10.2;c:\esp-idf5.1\tools\idf-exe\1.0.3;c:\esp-idf5.1\tools\ccache\4.8\ccache-4.8-windows-x86_64;c:\esp-idf5.1\tools\dfu-util\0.11\dfu-util-0.11-win64;c:\esp-idf5.1\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: c:\esp-idf5.1\tools\openocd-esp32\v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: c:\esp-idf5.1\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) c:\esp-idf5.1\python_env\idf5.1_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) c:\esp-idf5.1
Git Path (idf.gitPath) c:\esp-idf5.1\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to c:\esp-idf5.1\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin: true
Access to c:\esp-idf5.1\tools\riscv32-esp-elf-gdb\12.1_20231023\riscv32-esp-elf-gdb\bin: true
Access to c:\esp-idf5.1\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin: true
Access to c:\esp-idf5.1\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin: true
Access to c:\esp-idf5.1\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin: true
Access to c:\esp-idf5.1\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin: true
Access to c:\esp-idf5.1\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to c:\esp-idf5.1\tools\cmake\3.24.0\bin: true
Access to c:\esp-idf5.1\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin: true
Access to c:\esp-idf5.1\tools\ninja\1.10.2: true
Access to c:\esp-idf5.1\tools\idf-exe\1.0.3: true
[... 195 lines cut ...]
Selected configuration: prod1

Configuration name: currentProjectConfKey
---- Build section ----
     Compile Arguments: 
     Ninja Arguments: 
     Build directory path: ${workspaceFolder}\build_prod1
```

```
-------------------------------------------------- Project configuration settings ----------------------------------------------------------
Selected configuration: prod2

Configuration name: currentProjectConfKey
---- Build section ----
     Compile Arguments: 
     Ninja Arguments: 
     Build directory path: ${workspaceFolder}\build_prod2
```

Did it reffers to some variable or config:idf.buildPath works fine for it somehow?

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-08-18:
The issue is that c_cpp_properties.json and Intellisense is not handle by our extension. Best we can do is to update the c_cpp_properties.json file when you change from one profile to another by update compile_commands.json path.

executer-uno, 2024-08-19:
So, there is no option to feed c_cpp_properties.json with some env path variable to point current compile_commands.json? I asuume that it will not work becouse Intellisense will not follow env path variable updates, sounds logical.

Ok, Thanks for that answer. It would be awesome if you provide some mechanics to make Intellisense work as it desired while configuration changed. Or just add a note about required manual operations to do after switch, not to make it hidden under lid.

So, I will check how it works if correct compile_commands.json provided in c_cpp_properties.json at least manually.

brianignacio5, 2024-08-19:
I've added the changes in #1252

Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1826718510.zip) ?

executer-uno, 2024-08-19:
Works great.
:white_check_mark: ifdef codeblocks dimming
:white_check_mark: unknown identifiers red underline
:white_check_mark: Ctrl+Click navigation between files to exactly right destination
Thank you wery much!

PR #1252 (2024-10-11): fix support for multiple sdkconfig files
## Description

Fix support of multiple sdkconfig when using Project Configuration and multiple profiles.

Fixes #1202
Fixes #1265
Fixes #1283
Fixes #1299

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Using a project with multiple configurations like https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/multi_config run the `ESP-IDF: Open Project Configuration` and configure 2 profiles with different SDK Configuration Edito [... cut]
closed 2024-08-19 by executer-uno