espressif/vscode-esp-idf-extension #1288: Can't recognize the global variance and idf.py can't be imported in the vscode

b55585wy, 2024-08-25:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.92.2

### ESP-IDF version

5.3

### Python version

3.12.4

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22631 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v12.4\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v12.4\libnvvp;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Python312\Scripts\;C:\Python312\;C:\Program Files (x86)\VMware\VMware Workstation\bin\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\dotnet\;C:\Program Files\Git\cmd;C:\ProgramData\chocolatey\bin;C:\Program Files\MATLAB\R2024a\bin;C:<HOMEPATH>\AppData\Roaming\nvm;C:\Program Files\nodejs;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Program Files\Docker\Docker\resources\bin;C:\Program Files (x86)\Pulse Secure\VC142.CRT\X64\;C:\Program Files (x86)\Pulse Secure\VC142.CRT\X86\;C:\Program Files (x86)\Common Files\Pulse Secure\TNC Client Plugin\;;C:\Program Files (x86)\pcsuite\;C:\Program Files\NVIDIA Corporation\Nsight Compute 2024.1.0\;C:<HOMEPATH>\AppData\Local\Apps\Eclipse Adoptium\jdk-21.0.4.7-hotspot\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:\Program Files\JetBrains\WebStorm 2024.1.5\bin;C:<HOMEPATH>\AppData\Roaming\npm;C:\Program Files\JetBrains\PyCharm 2024.1.4\bin;C:\Program Files\JetBrains\IntelliJ IDEA 2024.1.5\bin;C:<HOMEPATH>\anaconda3;C:<HOMEPATH>\anaconda3\Library\bin;C:<HOMEPATH>\anaconda3\Scripts;C:<HOMEPATH>\AppData\Roaming\Python\Python312\Scripts;C:<HOMEPATH>\Development kit\vcpkg;C:\Program Files\JetBrains\CLion 2024.2\bin;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin;C:\Program Files\Huawei\DevEco Studio\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.92.2 
Visual Studio Code language zh-cn 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.8.1 
Workspace folder c:<HOMEPATH>\CodeSources\SYNC\sync-control-layer 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:/Espressif/frameworks/esp-idf-v5.3/
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) C:\Espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.38_20240113\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.24.0\bin;C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin;C:\Espressif\tools\ninja\1.11.1;C:\Espressif\tools\idf-exe\1.0.3;C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Espressif\tools\esp-rom-elfs\20240305
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:\Espressif\tools\esp-rom-elfs\20240305/
Virtual env Python Path (idf.pythonBinPath) C:/Espressif/python_env/idf5.3_py3.11_env/Scripts/python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) board/esp32c3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Espressif
Git Path (idf.gitPath) C:/Espressif/tools/idf-git/2.44.0/cmd/git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin: true
Access to C:\Espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin: true
Access to C:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin: true
Access to C:\Espressif\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin: true
Access to C:\Espressif\tools\esp32ulp-elf\2.38_20240113\esp32ulp-elf\bin: true
Access to C:\Espressif\tools\cmake\3.24.0\bin: true
Access to C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin: true
Access to C:\Espressif\tools\ninja\1.11.1: true
Access to C:\Espressif\tools\idf-exe\1.0.3: true
Access to C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64: true
[... 101 lines cut ...]
{"message":"Cannot access filePath: ${env:MDF_PATH}","stack":"Error: ENOENT: no such file or directory, access 'C:\\Users\\a1396\\AppData\\Local\\Programs\\Microsoft VS Code\\${env:MDF_PATH}'\n\tat Object.accessSync (node:fs:246:11)\n\tat t.accessSync (node:electron/js2c/node_init:2:9017)\n\tat Se (\\dist\\extension.js:2:1421806)\n\tat \\dist\\extension.js:2:1651002\n\tat ua (\\dist\\extension.js:2:1651892)\n\tat async \\dist\\extension.js:2:1868405","level":"error","timestamp":"2024-08-25T11:04:49.912Z"}
{"message":"Cannot access filePath: ${env:ESP_MATTER_PATH}","stack":"Error: ENOENT: no such file or directory, access 'C:\\Users\\a1396\\AppData\\Local\\Programs\\Microsoft VS Code\\${env:ESP_MATTER_PATH}'\n\tat Object.accessSync (node:fs:246:11)\n\tat t.accessSync (node:electron/js2c/node_init:2:9017)\n\tat Se (\\dist\\extension.js:2:1421806)\n\tat \\dist\\extension.js:2:1651094\n\tat ua (\\dist\\extension.js:2:1651892)\n\tat async \\dist\\extension.js:2:1868405","level":"error","timestamp":"2024-08-25T11:04:49.912Z"}
{"message":"Cannot access filePath: ${env:HOMEKIT_PATH}","stack":"Error: ENOENT: no such file or directory, access 'C:\\Users\\a1396\\AppData\\Local\\Programs\\Microsoft VS Code\\${env:HOMEKIT_PATH}'\n\tat Object.accessSync (node:fs:246:11)\n\tat t.accessSync (node:electron/js2c/node_init:2:9017)\n\tat Se (\\dist\\extension.js:2:1421806)\n\tat \\dist\\extension.js:2:1651190\n\tat ua (\\dist\\extension.js:2:1651892)\n\tat async \\dist\\extension.js:2:1868405","level":"error","timestamp":"2024-08-25T11:04:49.913Z"}
{"message":"Cannot access filePath: C:/Espressif/python_env/idf5.3_py3.11_env/Scripts/python.exe","stack":"Error: ENOENT: no such file or directory, access 'C:\\Espressif\\python_env\\idf5.3_py3.11_env\\Scripts\\python.exe'\n\tat Object.accessSync (node:fs:246:11)\n\tat t.accessSync (node:electron/js2c/node_init:2:9017)\n\tat Se (\\dist\\extension.js:2:1421806)\n\tat \\dist\\extension.js:2:1651286\n\tat ua (\\dist\\extension.js:2:1651892)\n\tat async \\dist\\extension.js:2:1868405","level":"error","timestamp":"2024-08-25T11:04:49.913Z"}
{"message":"spawn C:/Espressif/python_env/idf5.3_py3.11_env/Scripts/python.exe ENOENT","stack":"Error: spawn C:/Espressif/python_env/idf5.3_py3.11_env/Scripts/python.exe ENOENT\n\tat ChildProcess._handle.onexit (node:internal/child_process:286:19)\n\tat onErrorNT (node:internal/child_process:484:16)\n\tat process.processTicksAndRejections (node:internal/process/task_queues:82:21)","level":"error","timestamp":"2024-08-25T11:04:50.166Z"}
{"user":true,"level":"warn","message":"扩展配置报告已复制到剪贴板，但有错误","timestamp":"2024-08-25T11:04:50.167Z"}

--------------------------------------------------------------------------------------------------------------------------------------------

### Description

idf.py can't be used in the vscode terminal. Otherwise, I checked the global environments and found that it has IDF_TOOLS_PATH. But whenever I open ESP-IDF-5.3 CMD, it prompts that "IDF_TOOLS_PATH not set. Se [... cut ...]
Cannot import module "esp_idf_monitor". This usually means that "idf.py" was not spawned within an ESP-IDF shell environment or the Python virtual environment used by "idf.py" is corrupted.
Please use idf.py only in an ESP-IDF shell environment. If the problem persists, please try to install ESP-IDF tools again as described in the Get Started guide.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-08-26:
Launch a  terminal using the `ESP-IDF: Open ESP-IDF Terminal` instead of just opening any regular terminal.

b55585wy, 2024-08-29:
Actually I did use this command

b55585wy, 2024-08-29:
> Launch a terminal using the `ESP-IDF: Open ESP-IDF Terminal` instead of just opening any regular terminal.

Actually I did use this command

closed 2024-08-29 by b55585wy