espressif/vscode-esp-idf-extension #1299: Multi config dependent components

executer-uno, 2024-08-30:
### OS

Windows

### Operating System version

Windows 10 Pro

### Visual Studio Code version

1.92.2

### ESP-IDF version

5.1.4

### Python version

3.11.7

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\PuTTY\;C:\Program Files\IVI Foundation\VISA\Win64\bin;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\bin;C:\Program Files (x86)\IVI Foundation\IVI\bin;C:\Program Files\IVI Foundation\IVI\bin;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\msys64\mingw64\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:\CMake\bin;D:\Downloads\make-3.81-bin\bin; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.92.2 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.8.0 
Workspace folder -=MY_PATH=-\IDF-MultiConfig\multi_config 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) c:<HOMEPATH>\esp\v5.1\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) c:\esp-idf5.1\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin;c:\esp-idf5.1\tools\riscv32-esp-elf-gdb\12.1_20231023\riscv32-esp-elf-gdb\bin;c:\esp-idf5.1\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;c:\esp-idf5.1\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;c:\esp-idf5.1\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;c:\esp-idf5.1\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;c:\esp-idf5.1\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;c:\esp-idf5.1\tools\cmake\3.24.0\bin;c:\esp-idf5.1\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin;c:\esp-idf5.1\tools\ninja\1.10.2;c:\esp-idf5.1\tools\idf-exe\1.0.3;c:\esp-idf5.1\tools\ccache\4.8\ccache-4.8-windows-x86_64;c:\esp-idf5.1\tools\dfu-util\0.11\dfu-util-0.11-win64;c:\esp-idf5.1\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: c:\esp-idf5.1\tools\openocd-esp32\v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: c:\esp-idf5.1\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) c:\esp-idf5.1\python_env\idf5.1_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) board/esp32s3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) c:\esp-idf5.1
Git Path (idf.gitPath) c:\esp-idf5.1\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to c:\esp-idf5.1\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin: true
Access to c:\esp-idf5.1\tools\riscv32-esp-elf-gdb\12.1_20231023\riscv32-esp-elf-gdb\bin: true
Access to c:\esp-idf5.1\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin: true
Access to c:\esp-idf5.1\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin: true
Access to c:\esp-idf5.1\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin: true
Access to c:\esp-idf5.1\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin: true
Access to c:\esp-idf5.1\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to c:\esp-idf5.1\tools\cmake\3.24.0\bin: true
Access to c:\esp-idf5.1\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin: true
Access to c:\esp-idf5.1\tools\ninja\1.10.2: true
Access to c:\esp-idf5.1\tools\idf-exe\1.0.3: true
[... 130 lines cut ...]
3. I dont want all found componetns to compile, so need to add minimal list in root CMakeLists.txt
```
set(COMPONENTS main bootloader esptool_py)
```
4. Now we need to add components dependency from selected configuration. Only way that works fine for me is
```
if(CONFIG_EXAMPLE_FUNC_IMPL_DEV)

    target_sources(${COMPONENT_LIB} PRIVATE "func_dev.c")
    target_link_libraries(${COMPONENT_LIB} PRIVATE Component_A)
    
elseif(CONFIG_EXAMPLE_FUNC_IMPL_PROD)

    target_sources(${COMPONENT_LIB} PRIVATE "func_prod.c")
    target_link_libraries(${COMPONENT_LIB} PRIVATE Component_B)

endif()
```
We can not use REQUIRES field of idf_component_register() becouse IDF Build system not accepts variables there, and, now it becomes clear, that there is a reson for it - on clear build we need to make sdkconfig file (one will contain CONFIG_EXAMPLE_FUNC_IMPL_PROD definition), that depends on components paticipates in project, and I want to make self-dependent link of dependencies: required components depends on sdkconfig definitions, sdkconfig defenitions depends on used components list. Lock.
6.  Now I got an error No such file or directory #include "Component_B.h" becouse Component_B is not in constant REQUIRES list of any components and it was not added to entire scope due to set(COMPONENTS main bootloader esptool_py) limited scope. target_link_libraries(${COMPONENT_LIB} PRIVATE Component_B) not helps, becouse IDF Build system scans required components onlu on first CMake run, that quits on idf_component_register() found.
7.  So, only way I have to go on - is to keep all my components compilable in all possible configurations and leave all the important bricks picking for linker. Am I right to proceed only that way?

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

executer-uno, 2024-08-31:
So, rafinated question is:
Are there any way to make component's REQUIRES list dependend on PROJECT_CONFIGURATION control element active selection down the VSC window? But not via indirect way using sdkconfig templates and CONFIG_ definitions?

executer-uno, 2024-08-31:
I start to think I miss-used MultiConfig concept provided by ESP-IDF.
That [VSCode provided way](https://randomnerdtutorials.com/vs-code-workspaces-esp32-esp8266/) of Workspace with several folders looks promising for complicated project.

So I tryed it with following reorganised project:
```
.
│   .gitignore
│   Project.code-workspace
│   README.md
│
├───components
│   ├───Component_A
│   │   │   CMakeLists.txt
│   │   │   Component_A.c
│   │   │
│   │   └───include
│   │           Component_A.h
│   │
│   └───Component_B
│       │   CMakeLists.txt
│       │   Component_B.c
│       │
│       └───include
│               Component_B.h
│
├───ProjectX
│   │   CMakeLists.txt
│   │   esp_idf_project_configuration.json
│   │
│   └───main
│           CMakeLists.txt
│           func.h
│           func_dev.c
│           func_prod.c
│           Kconfig.projbuild
│           multi_config_example_main.c
│
├───ProjectY
│   │   CMakeLists.txt
│   │   esp_idf_project_configuration.json
│   │
│   ├───build_prod2
│   │       .ninja_log
│   │
│   └───main
│           CMakeLists.txt
│           func.h
│           func_dev.c
│           func_prod.c
│           Kconfig.projbuild
│           multi_config_example_main.c
│
└───sdkconfig.defaults
        sdkconfig.prod1
        sdkconfig.prod2
        sdkconfig.prod_common
```

And it works fine. I can choose particular project on "folder" button on VSC status bar. It reffers to some common components in top folder by direct reference only required componen folder `list(APPEND EXTRA_COMPONENT_DIRS "../components/Component_B")` Then I can use static defined `REQUIRES Component_B` in mime main/CMakeLists and, optionally, `set(COMPONENTS main bootloader esptool_py)` in ProjectY/CMakeLists.txt. So, I just need to make individual main.c for each ProjectN in workspace that uses commonly available components in common folder, and i like it,

Minor issue that may be fixed: As you see in folders tree I try to mix IDF MultiConfig and VSC WorkSpace, and generally it works properly, but "Select Project Configuration" should be done manually each time I change active project in WorkSpace.

So, It feels for me like IDF MultiConfig is nice for DEV/PROD switchover and Projects in Workspace nice for "AirConditioner"/"MicrowaveOven"/"VacuumCleaner" switchover between projects used some common components in between.

brianignacio5, 2024-09-09:
Hi @executer-uno sorry for late reply.

As you described the multi config project use case is the latter use case. The issue with project configuration is that it doesn't override CMakeLists.txt just env variables and settings. I think this is a cleaner approach, if you have set some configuration within a CMakeLists.txt you probably want them to be dependent on sdkconfig values (i.e. `if(CONFIG_EXAMPLE_FUNC_IMPL_DEV)`) or have some hard defined behavior. In this way you could make a component be project (or project configuration with different sdkconfig) configurable.

With regards to the minor issue, I'll take a look I didn't consider the use case of having both project configuration with multi workspace folder. Since current  project configuration is saved as workspace state it could affect other project if using multiple workspace folders.

I think we can delete the selected project configuration when changing active workspace folder and either:

1) add selected project configuration to esp_idf_project_configuration.json so it can be picked automatically
2)show a dropdown to select project configuration on new workspace folder (if esp_idf_project_configuration.json exists)

executer-uno, 2024-09-09:
Hi, thanks for clarification.
I see your point, and its clear that my expectation was in conflict to CMake nature project.
Im happy with any way to organize multiproject and I like the way described in previous post, its very clear due to each project has its own CMakeLists and still used shared components.

Up to two options to solve selected project configuration when changing active workspace: I would prefer, as particular user, second option:
2. show a dropdown to select project configuration on new workspace folder (if esp_idf_project_configuration.json exists)

Thank you for support!

PR #1252 (2024-10-11): fix support for multiple sdkconfig files
## Description

Fix support of multiple sdkconfig when using Project Configuration and multiple profiles.

Fixes #1202
Fixes #1265
Fixes #1283
Fixes #1299

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Using a project with multiple configurations like https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/multi_config run the `ESP-IDF: Open Project Configuration` and configure 2 profiles with different SDK Configuration Edito [... cut]
closed 2024-09-30 by github-actions[bot]