espressif/vscode-esp-idf-extension #1301: "Configure ESP-IDF extension" fails to install GCC toolchains

badlogic, 2024-09-01:
### OS

MacOS

### Operating System version

macOS 14.5 (23F79)

### Visual Studio Code version

1.81.1

### ESP-IDF version

1.6.5

### Python version

3.12.5

### Doctor command output

N/A

### Extension

Installing package GDB for Xtensa
Installing tar.gz package GDB for Xtensa
Installing package GDB for RISC-V
Installing tar.gz package GDB for RISC-V
Installing package Toolchain for Xtensa (ESP32) based on GCC
Current system platform: macos-arm64
Installing tools: 

Installing package Toolchain for Xtensa (ESP32-S2) based on GCC
Current system platform: macos-arm64
Installing tools: 

Installing package Toolchain for Xtensa (ESP32-S3) based on GCC
Current system platform: macos-arm64
Installing tools: 

Installing package Toolchain for 32-bit RISC-V based on GCC
Current system platform: macos-arm64
Installing tools: 

Installing package Toolchain for ESP32 ULP coprocessor
Installing tar.gz package Toolchain for ESP32 ULP coprocessor
Installing package OpenOCD for ESP32
Installing tar.gz package OpenOCD for ESP32
Installing package ESP ROM ELFs
Installing tar.gz package ESP ROM ELFs

Creating a new Python environment in /Users/badlogic/.espressif/python_env/idf5.1_py3.12_env
Downloading https://dl.espressif.com/dl/esp-idf/espidf.constraints.v5.1.txt
Destination: /Users/badlogic/.espressif/espidf.constraints.v5.1.txt.tmp

0%
100%
Done
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Ignoring importlib_metadata: markers 'python_version < "3.8"' don't match your environment
Collecting setuptools (from -r /Users/badlogic/esp/esp-idf/tools/requirements/requirements.core.txt (line 3))
  Using cached https://dl.espressif.com/pypi/setuptools/setuptools-71.0.0-py3-none-any.whl (908 kB)
Collecting packaging (from -r /Users/badlogic/esp/esp-idf/tools/requirements/requirements.core.txt (line 4))
  Using cached https://dl.espressif.com/pypi/packaging/packaging-24.1-py3-none-any.whl (53 kB)
Collecting click (from -r /Users/badlogic/esp/esp-idf/tools/requirements/requirements.core.txt (line 7))
  Using cached https://dl.espressif.com/pypi/click/click-8.1.7-py3-none-any.whl (97 kB)
Collecting pyserial (from -r /Users/badlogic/esp/esp-idf/tools/requirements/requirements.core.txt (line 8))
  Using cached https://dl.espressif.com/pypi/pyserial/pyserial-3.5-py2.py3-none-any.whl (90 kB)
[... 188 lines cut ...]
When installing ESP-IDF through this extension, everything gets downloaded properly, including the various GCC toolchains in `.espressif/dist`, but the extension fails to unpack these GCC toolchains. See the end of the log output above, which probes for the existance of each GCC version and fails.

For some reason, not all tar.xz/gz files in `.espressif/dist` are not unpacked. The extension successfully unpacks the following tools into `.espressif/tools`

- esp-rom-elfs
- esp32ulp-elf
- openocd-esp32
- riscv32-esp-elf-gdb
- xtensa-esp-elf-gdb

It fails for all other GCC toolchains.

The problem can be fixed, by opening a terminal and executing `~/esp/esp-idf/install.sh`. The script successfully unpacks the remaining files in `.espressif/dist`.

I have not tried the installation process on another machine (as I lack a second machine). If this is indeed an issue on all Macs, it will confuse a lot of new comers :)

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-09-09:
You are using an old version of the extension. Could you try updating to newest version first ?

badlogic, 2024-09-09:
Ah, you're right! I'm stuck on an older VS Code build. The extension manager installed the the latest compatible ESP-IDF exfension, which is indeed quite old.

Newest VS Code + newest ESP-IDF extension works as intended! Sorry for adding to the work load unnecessarily!

closed 2024-09-09 by badlogic

brianignacio5, 2024-09-09:
No worries, that what we are here for !

If you find the extension useful, please consider leaving us a review in the [Visual Studio Code Marketplace](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension)