espressif/vscode-esp-idf-extension #1302: Automatically selected bad target

igparaul, 2024-09-02:
### OS

Windows

### Operating System version

Windows 10

### ESP-IDF version

5.2.1

### Python version

3.11

### Doctor command output

Nothing

### Description

Hello, this is only a bug report.

If you create a new project with "Ctrl+E N" and selects the target to use (in my case ESP32-S3), after the project is succesfully created, when you try to flash the code into the device, it can be noticed that the code is tried to be flashed on an ESP32 target.

### Debug Message

```plain
Nothing
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

radurentea, 2024-09-08:
Hi @igparaul,

Thank you for the report! I've managed to reproduce the issue.

However, we currently have three ways of creating a new project and we plan to stick to just one method. Will keep this ticket open for now and we will address it when we take a decision.

brianignacio5, 2025-02-17:
Could you test the latest [vsix installer from master branch](https://nightly.link/espressif/vscode-esp-idf-extension/workflows/ci/master/esp-idf-extension.vsix.zip) to see if the issue is fixed ?

PR #1498 (2025-04-21): [VSC-1640] profiles env should merge customExtraVars with profile env
## Description

When the user is working with multiple profiles, the profile env should merge with `idf.customExtraVars` instead of fully replace it. This is necessary to support IDF_TARGET when sdkconfig file doesn't exists.

Fixes #1497
Fixes #1302

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Create a project with multiple profile configurat [... cut]
closed 2025-04-21 by brianignacio5