espressif/vscode-esp-idf-extension #1303: ESP-IDF Setup got stuck in WSL2

liyafe1997, 2024-09-06:
### OS

Windows

### Operating System version

Win11 & WSL2 & Ubuntu 22.04

### Visual Studio Code version

1.93.0

### ESP-IDF version

No

### Python version

3.10.2

### Doctor command output

https://pastebin.com/rGccpbr0
It is too long so I put in pastebin.

### Description

1. Install an Ubuntu 22.04 in WSL2
2. Open VSCode in Windows, connect to the WSL2 Ubuntu, install ESP-IDF extension.
3. Click `Configure ESP-IDF Extension` , whatever use `EXPRESS` Setup or `ADVANCED`, just want to setup ESP-IDF in the VSCode Extension.
4. After finish `Installing ESP-IDF`, it will stuck on the first step of `Installing ESP-IDF Tools`, it is `Downloading GDB for Xtensa`.

[image]

I am 100000% sure it is not the networking problem. And I don't have any proxy settings.
I can directly `wget` that URL:
```
$ wget https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v14.2_20240403/xtensa-esp-elf-gdb-14.2_20240403-x86_64-linux-gnu.tar.gz
--2024-09-06 10:08:21--  https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v14.2_20240403/xtensa-esp-elf-gdb-14.2_20240403-x86_64-linux-gnu.tar.gz
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/140236854/73266772-b6cb-48ff-8b99-e9f22eb7630f?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20240906%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240906T080821Z&X-Amz-Expires=300&X-Amz-Signature=dfcfc0b578b6c48dbaf71a642230654041a8ab2af1bae6a92c9e5b8d9f8d853f&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=140236854&response-content-disposition=attachment%3B%20filename%3Dxtensa-esp-elf-gdb-14.2_20240403-x86_64-linux-gnu.tar.gz&response-content-type=application%2Foctet-stream [following]
--2024-09-06 10:08:21--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/140236854/73266772-b6cb-48ff-8b99-e9f22eb7630f?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20240906%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240906T080821Z&X-Amz-Expires=300&X-Amz-Signature=dfcfc0b578b6c48dbaf71a642230654041a8ab2af1bae6a92c9e5b8d9f8d853f&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=140236854&response-content-disposition=attachment%3B%20filename%3Dxtensa-esp-elf-gdb-14.2_20240403-x86_64-linux-gnu.tar.gz&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.108.133, 185.199.109.133, 185.199.110.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.108.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 31010320 (30M) [application/octet-stream]
Saving to: ‘xtensa-esp-elf-gdb-14.2_20240403-x86_64-linux-gnu.tar.gz’

xtensa-esp-elf-gdb-14.2_20240 100%[=================================================>]  29.57M  43.2MB/s    in 0.7s

2024-09-06 10:08:22 (43.2 MB/s) - ‘xtensa-esp-elf-gdb-14.2_20240403-x86_64-linux-gnu.tar.gz’ saved [31010320/31010320]
```

### Debug Message

```plain
No
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-09-09:
If you do `Preferences: Open Settings (UI)` and search for `http.proxy` or `http.proxyStrictSSL` do you see anything set ? Make sure to check User Workspace or WSL tabs.

liyafe1997, 2024-09-09:
> If you do `Preferences: Open Settings (UI)` and search for `http.proxy` or `http.proxyStrictSSL` do you see anything set ? Make sure to check User Workspace or WSL tabs.

No, I have double confirmed, also checked `export` there are no proxy things in my environment. I never use proxy on this machine.

liyafe1997, 2024-10-29:
I found the IDF extension even blocking the network for other extensions.

For example, once the IDF extension is loaded, the Github Copilot extension will no longer can connect to server.

[image]

If I disable the IDF extension, everything back to work!

liyafe1997, 2024-10-29:
@brianignacio5 Why this issue got closed? 
I replied your msg and no any other people reply from ESP32 then this got closed?

Should I make a new issue?

brianignacio5, 2024-10-30:
bot closed it due to lack of activity.

> I found the IDF extension even blocking the network for other extensions.

How ? We don't do any network modification as far as I can see (you can check the source code is fully open source) and not sure how we can alter other extensions.

I'm not able to reproduce your issue. I've followed the steps you mentioned but failed to see the network issue. Have you manage to reproduce the issue in a different machine ?

liyafe1997, 2024-10-30:
I also think it is incredible, but it really happens.
I only have that one Win11&WSL2 setup, I also can't reproduce on other kind of setup, like directly run VScode in a Linux machine with ESP extension.

Are you trying to use the WSL2, and vscode is running in the host Windows 11, then use the WSL extension connects to WSL2 Ubuntu, and install & run ESP extension?

brianignacio5, 2024-10-31:
Yes a separated machine using Windows 11 WSL 2 with Ubuntu 22.04. Installed esp-idf extension on WSL 2 with vscode, run the setup from start and no problems were shown.

liyafe1997, 2024-11-04:
Maybe I found the problem...

Because in my WSL there is a network interface with IPv6 local address (start with `fdxx`), and there is no internet access on this interface. But it is not a default route and other programs works fine (the internet traffic should goes to another interface), even in VSCode without IDF installed also works.  But after IDF installed, seems the whole VSCode route all internet traffics to that interface & IPv6...

After I disabled IPv6 on that interface everything works fine...

closed 2024-11-04 by liyafe1997