espressif/vscode-esp-idf-extension #1304: ERROR: MSys/Mingw is not supported.

dmaltsiniotis, 2024-09-08:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.92.2

### ESP-IDF version

1.8.1

### Python version

3.11.9

### Doctor command output

N/A

### Extension

```
{"message":"Cannot access filePath: C:\\Users\\Divine Shadow\\Desktop\\esp-idf\\tools\\idf.py","stack":"Error: ENOENT: no such file or directory, access 'C:\\Users\\Divine Shadow\\Desktop\\esp-idf\\tools\\idf.py'\n\tat Object.accessSync (node:fs:246:11)\n\tat t.accessSync (node:electron/js2c/node_init:2:9017)\n\tat Se (c:\\Users\\Divine Shadow\\.vscode\\extensions\\espressif.esp-idf-extension-1.8.1\\dist\\extension.js:2:1421806)\n\tat i.value (c:\\Users\\Divine Shadow\\.vscode\\extensions\\espressif.esp-idf-extension-1.8.1\\dist\\extension.js:2:1615589)\n\tat n.B (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:83:737)\n\tat n.fire (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:83:954)\n\tat w.$onMessage (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:146:19363)\n\tat E.S (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:151:5980)\n\tat E.Q (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:151:5746)\n\tat E.M (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:151:4778)\n\tat E.L (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:151:3830)\n\tat i.value (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:151:2297)\n\tat n.B (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:83:737)\n\tat n.fire (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:83:954)\n\tat s.fire (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:97:14453)\n\tat i.value (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:177:8655)\n\tat n.B (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:83:737)\n\tat n.fire (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:83:954)\n\tat s.fire (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:97:14453)\n\tat MessagePortMain.<anonymous> (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:177:6781)\n\tat MessagePortMain.emit (node:events:519:28)\n\tat MessagePortMain._internalPort.emit (node:electron/js2c/utility_init:2:2285)","level":"error","timestamp":"2024-09-08T00:10:29.951Z"}
{"message":"Cannot access filePath: C:\\dev\\esp-idf\\tools\\idf.py","stack":"Error: ENOENT: no such file or directory, access 'C:\\dev\\esp-idf\\tools\\idf.py'\n\tat Object.accessSync (node:fs:246:11)\n\tat t.accessSync (node:electron/js2c/node_init:2:9017)\n\tat Se (c:\\Users\\Divine Shadow\\.vscode\\extensions\\espressif.esp-idf-extension-1.8.1\\dist\\extension.js:2:1421806)\n\tat i.value (c:\\Users\\Divine Shadow\\.vscode\\extensions\\espressif.esp-idf-extension-1.8.1\\dist\\extension.js:2:1615589)\n\tat n.B (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:83:737)\n\tat n.fire (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:83:954)\n\tat w.$onMessage (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:146:19363)\n\tat E.S (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:151:5980)\n\tat E.Q (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:151:5746)\n\tat E.M (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:151:4778)\n\tat E.L (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:151:3830)\n\tat i.value (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:151:2297)\n\tat n.B (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:83:737)\n\tat n.fire (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:83:954)\n\tat s.fire (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:97:14453)\n\tat i.value (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:177:8655)\n\tat n.B (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:83:737)\n\tat n.fire (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:83:954)\n\tat s.fire (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:97:14453)\n\tat MessagePortMain.<anonymous> (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:177:6781)\n\tat MessagePortMain.emit (node:events:519:28)\n\tat MessagePortMain._internalPort.emit (node:electron/js2c/utility_init:2:2285)","level":"error","timestamp":"2024-09-08T00:10:59.290Z"}
{"message":"Cannot access filePath: C:\\dev\\esp-idf\\tools\\idf.py","stack":"Error: ENOENT: no such file or directory, access 'C:\\dev\\esp-idf\\tools\\idf.py'\n\tat Object.accessSync (node:fs:246:11)\n\tat t.accessSync (node:electron/js2c/node_init:2:9017)\n\tat Se (c:\\Users\\Divine Shadow\\.vscode\\extensions\\espressif.esp-idf-extension-1.8.1\\dist\\extension.js:2:1421806)\n\tat i.value (c:\\Users\\Divine Shadow\\.vscode\\extensions\\espressif.esp-idf-extension-1.8.1\\dist\\extension.js:2:1615589)\n\tat n.B (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:83:737)\n\tat n.fire (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:83:954)\n\tat w.$onMessage (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:146:19363)\n\tat E.S (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:151:5980)\n\tat E.Q (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:151:5746)\n\tat E.M (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:151:4778)\n\tat E.L (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:151:3830)\n\tat i.value (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:151:2297)\n\tat n.B (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:83:737)\n\tat n.fire (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:83:954)\n\tat s.fire (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:97:14453)\n\tat i.value (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:177:8655)\n\tat n.B (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:83:737)\n\tat n.fire (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:83:954)\n\tat s.fire (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:97:14453)\n\tat MessagePortMain.<anonymous> (c:\\Program Files\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\api\\node\\ext [... cut ...]
I am not using MSys/Mingw (at least purposefully), so I don't know why the extension installer is getting stuck on this.

The offline installer does install to completion, but the VS Code installer does not and I cannot "use an existing install" either, it fails with the same error.

Any help would be greatly appreciated.

### Debug Message

```plain
`ERROR: MSys/Mingw is not supported.`
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-09-09:
Could you share the doctor command output as requested in the issue template?

We have no means to understand your environment without it.

dmaltsiniotis, 2024-09-09:
> Could you share the doctor command output as requested in the issue template?
> 
> We have no means to understand your environment without it.

HI @brianignacio5, thank you for looking into this issues. From the issue template, it is not clear that' ESP-IDF: Doctor command is a VS command pallet option, and not some kind of command line analysis, after some poking around to figure out how to execture this this "doctor" command, I have amended the original issue with its output now.

brianignacio5, 2024-09-09:
This error is triggered in ESP-IDF idf.py here: https://github.com/espressif/esp-idf/blob/master/tools/idf_tools.py#L3240

It seems there is an environment variable `MSYSTEM` in your setup that is being detected as MSys/Mingw

dmaltsiniotis, 2024-09-09:
Thank you @brianignacio5 for this, this helped point me in the right direction. I think the issue stemmed from the fact that for my VS Code default terminal, I use GIT BASH which _is_ MSys based. However, I did not expect the VS Code extension to also use this default terminal when executing its installation commands. I had originally suspected as much, and flipped the default terminal to `cmd`, but I was still getting the same error. However, I have now learned that VS Code does not refresh environment variables between different terminal sessions, only complete exist of the application and restarting it.

Changing the default terminal, combined with being careful to not launch the GIT BASH terminal (even though it's not set as the default terminal) to not set the MSYSTEM environment variable, was what I needed to resolve the issue and allow the VS Code extension to install the ESP-IDF tools and environment.

Thank you from your help!

-Demetri

closed 2024-09-09 by dmaltsiniotis