espressif/vscode-esp-idf-extension #1307: [Feature Request]: Whether the user can choose whether or not certain features of the status bar are displayed

Tinyu-Zhao, 2024-09-10:
[image]

I almost never use these two features in my vscode, but the bottom status bar cannot be customized to enable certain specific features, and sometimes it looks too crowded on devices with small screens.

Can users customize the opening/closing of certain functions in the status bar at the bottom of the esp-idf plugin? I think this would be better.

On my 16-inch display, a lot of the bottom status bar is occupied by ESP-IDF. I think this is very convenient, but I still don't want to see them for functions that I don't use.

[image]

brianignacio5, 2024-09-10:
When working with https://github.com/espressif/vscode-esp-idf-extension/pull/1287 I wanted to propose this feature but it will require to make a UI wizard or add it to the welcome UI and have a flow to control them.

Thanks for the feature request, I'll share a link when we have an implementation for you to test.

Tinyu-Zhao, 2024-09-10:
> When working with #1287 I wanted to propose this feature but it will require to make a UI wizard or add it to the welcome UI and have a flow to control them.
> 
> Thanks for the feature request, I'll share a link when we have an implementation for you to test.

I think that's good, thank you.

brianignacio5, 2024-09-18:
@Tinyu-Zhao @GillesZunino 

I've made a PR to add an option for this. Checkboxes next to commands to show/hide the status bar items.

[image]

Here is the [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1947101057.zip). 

Changes are not persisted and if you close vscode and open again all icons are shown. I'm not sure if we should persist them. What do you think ?

Tinyu-Zhao, 2024-09-18:
> @Tinyu-Zhao @GillesZunino
> 
> I've made a PR to add an option for this. Checkboxes next to commands to show/hide the status bar items.
> 
> ![Screenshot 2024-09-18 at 18 00 55](https://private-user-images.githubusercontent.com/6033246/368520279-e7344c11-a7e6-482f-95e0-41645d8bf394.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY2NTUwMzksIm5iZiI6MTcyNjY1NDczOSwicGF0aCI6Ii82MDMzMjQ2LzM2ODUyMDI3OS1lNzM0NGMxMS1hN2U2LTQ4MmYtOTVlMC00MTY0NWQ4YmYzOTQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MThUMTAxODU5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NmY4MDJjMjhhNGExNDY1MzFiYTQyYTcxYTIyZTk3YWZkMjM2NWQ5NTk2NGQ3ZGY2YzQyOWU4MDNhOTg5OGNiMiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.vVmCNEHX61VIX96RFPC85shNZcG2ccQp8C4nIT1Q4_U)
> 
> Here is the [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1947101057.zip).
> 
> Changes are not persisted and if you close vscode and open again all icons are shown. I'm not sure if we should persist them. What do you think ?

Thank you very much for making this feature. I wonder if you can write this configuration into .vscode/settings.json so that you can maintain the previous settings?

[image]

brianignacio5, 2024-09-23:
Hi @Tinyu-Zhao so I've updated the PR to persist the user selection as global state instead.

I think this is better than saving them in settings.json to avoid increasing complexity (we just merged a PR to remove some settings #1262 )

Here is the [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1965362013.zip)

Tinyu-Zhao, 2024-09-23:
> Hi @Tinyu-Zhao so I've updated the PR to persist the user selection as global state instead.
> 
> I think this is better than saving them in settings.json to avoid increasing complexity (we just merged a PR to remove some settings #1262 ) of extension and settings.
> 
> Here is the [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1965362013.zip)

I will try this version, thank you☺️

Tinyu-Zhao, 2024-09-23:
I found a problem. Even if you choose to close [openOCD Server] and Start/Stop QEMU Server, it will still be displayed in the bottom status bar when you close vscode and reopen it again.

[image]

And I found that the Select Project Configuration option cannot match any function in the bottom status bar, and turning this function on and off has no effect.

brianignacio5, 2024-09-23:
I'll look into those QEMU/OpenOCD case, seems I forgot to check them.

About the Select Project Configuration, this is an option that is only shown when you are using the Project Configuration editor and already defined profiles in there. More information in https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/project_configuration.md

Tinyu-Zhao, 2024-09-24:
> I'll look into those QEMU/OpenOCD case, seems I forgot to check them.
> 
> About the Select Project Configuration, this is an option that is only shown when you are using the Project Configuration editor and already defined profiles in there. More information in https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/project_configuration.md
> 
> EDIT: I've fixed those qemu and openOCD and should work now. [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1965577547.zip)

There is no problem with the test, thank you.

But there is an aesthetic problem. If this function can appear in the bottom status bar, a check box will be added at the front of COMMANDS, but other COMMANDS will feel unevenly arranged. Can I add it in front of them? An uncheckable ☑️ box so that all function icons can be aligned

[image]

brianignacio5, 2024-09-24:
The problem with this latest request is that checkboxes can't be set as uncheckable. We can either enable or disable the checkbox (by setting an initial Checked/Unchecked state). If we add an unchecked state the user can change the state and for those commands that doesn't have icon in the status bar it will do nothing and will confuse users.

Tinyu-Zhao, 2024-09-24:
> The problem with this latest request is that checkboxes can't be set as uncheckable. We can either enable or disable the checkbox (by setting an initial Checked/Unchecked state). If we add an unchecked state the user can change the state and for those commands that doesn't have icon in the status bar it will do nothing and will confuse users.
> 
> This is a limitation of vscode api itself. We can't control this behavior from extensions.

Ok, I understand, thank you. I think this function is quite good so far. It has solved the problem of occupying too much status bar.

brianignacio5, 2024-09-26:
Hi @Tinyu-Zhao  so we decided to updated the list to show the items as shown in status bar and move other commands to Advanced section. Hope this makes more sense for you too.

[image]

Tinyu-Zhao, 2024-09-26:
> Hi @Tinyu-Zhao so we decided to updated the list to show the items as shown in status bar and move other commands to Advanced section. Hope this makes more sense for you too.
> 
> ![Screenshot 2024-09-26 at 16 15 35](https://private-user-images.githubusercontent.com/6033246/371046952-c29809d1-d8d5-41db-9265-d3b4b87e769c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjczMzkxNTUsIm5iZiI6MTcyNzMzODg1NSwicGF0aCI6Ii82MDMzMjQ2LzM3MTA0Njk1Mi1jMjk4MDlkMS1kOGQ1LTQxZGItOTI2NS1kM2I0Yjg3ZTc2OWMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MjZUMDgyMDU1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDUwNzhkMWJiMmM4ZTM0YjQwMzhiM2E1NTczZmE3ODk3YmExZjE5ZDdiNTMyYzdhMTgxZDE3MjRjNWM2Njk3OSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.d8W1gzsFD4sWghDQDDa1iMBLuG1jVn8GIOhSdb7rotg)

This looks great, thank you.

closed 2024-09-27 by brianignacio5

Tinyu-Zhao, 2024-09-30:
> I found a problem. Even if you choose to close [openOCD Server] and Start/Stop QEMU Server, it will still be displayed in the bottom status bar when you close vscode and reopen it again.
> 
> <img alt="image" width="1615" src="https://private-user-images.githubusercontent.com/44396641/369827126-5685e423-50ef-4367-a0ea-ab06f0f288d3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc2OTAzNjgsIm5iZiI6MTcyNzY5MDA2OCwicGF0aCI6Ii80NDM5NjY0MS8zNjk4MjcxMjYtNTY4NWU0MjMtNTBlZi00MzY3LWEwZWEtYWIwNmYwZjI4OGQzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTMwVDA5NTQyOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWEyZmMyYTMyYTBhOWZjYjJlYmU4ZWYzNDE3NzYzNDkwMDcxMGVlMDdiNWEyNjAyNTg4ZGQ0Y2I1MGQyZGUwZDcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.sOZMtC4ciEEe7KS41jErSeyOYgEy_paM5o2USk4K6vk">
> And I found that the Select Project Configuration option cannot match any function in the bottom status bar, and turning this function on and off has no effect.

@brianignacio5 Hello, I just tried to pull the latest code from GitHub and build the extension myself and found a similar problem in `Select Current Project workspace folder`. Even if I uncheck it, it will reappear after restarting VsCode.
[image]

PR #1314 (2024-09-27): Enable/Disable status bar icons in Commands Tree view
## Description

Add Checkbox in Espressif tab Commands to be enable/disable in status bar.

[image]

Fixes #1307

## Type of change

- New feature (non-breaking change which adds functionality)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "Espressif" icon tab. You can see a list of commands with checkbox. If you click the checkbox to disable it, the icon in the status bar will disappear. The same for the opposite.

- Expected behaviour:
Icons appear and disappear fr [... cut]