espressif/vscode-esp-idf-extension #1309: [Feature Request]: Specify cmake arguments from file like idf.py

mbastida123, 2024-09-13:
Hi, I don't know if this is already possible. I'm crosposting from https://github.com/espressif/esp-idf/issues/12257

I have created a file `config.txt`:
`-B builds/build1 -DSDKCONFIG=builds/build1 /sdkconfig -DSDKCONFIG_DEFAULTS=zconfig/sdkconfig.prod_common`

Following the multi-config example I can `idf.py @config.txt build`

However, what if I wanted to integrate this into the vscode extension? 
The extension doesn't use `idf.py` and instead invokes `cmake` directly. So adding `@config.txt` to `"idf.cmakeCompilerArgs"` is not an option.

brianignacio5, 2024-09-18:
We have extension settings to achieve this use case. There is `idf.buildPath` to define the build path, `idf.sdkconfigDefaults` to define SDKCONFIG_DEFAULTS and  `idf.cmakeCompilerArgs` for additional CMake arguments. Look at the settings documentation https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETTINGS.md

I believe you are trying to use the multi_config example. We have a tutorial for what we call project configuration which allows you to use multiple configuration in a single ESP-IDF project, please take a look https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/project_configuration.md

mbastida123, 2024-09-18:
> We have extension settings to achieve this use case. There is `idf.buildPath` to define the build path, `idf.sdkconfigDefaults` to define SDKCONFIG_DEFAULTS and `idf.cmakeCompilerArgs` for [...]

Hi @brianignacio5 

I was aware of all the examples. In fact, that's what we were using until now.

However, due to the complexity of our multiple sdkconfig.defaults we want to automate the assignation of each sdkconfig.defaults for each build. And to avoid touching the settings.json from a script I had seen the option to pass cmd arguments directly via file. But from your explanation I understand this is not possible with cmake. However, it is possible with idf.py

brianignacio5, 2024-09-18:
If you want to use idf.py and this config.txt file, you can just open an IDF terminal with `ESP-IDF: Open ESP-IDF Terminal` command and run the `idf.py @config.txt build`. You can also just put this command to the `idf.customTask; "idf.py @config.txt build"` and use the `ESP-IDF: Execute Custom Task` to run it (There is an icon in status bar for execute custom task too)

If you want to use extension commands, you can define multiple profiles (with ESP-IDF: Project Configuration Editor) and when you select a profile and build the effect is the same. It takes a bit of writing but it achieves the same use case.

You can create a profile in Project Configuration Editor UI, each profile can define its own SDKCONFIG_DEFAULTS value. When you select a profile and build, it will use the SDKCONFIG_DEFAULTS from the selected profile. So it is indeed possible with CMake to achieve multiple sdkconfig defaults, this is covered in the project_configuration tutorial I shared before.

Does this information help your use case ?

brianignacio5, 2024-09-18:
Yes it is saved as a json file in the project directory when you saved from the Project Configuration Editor UI. The file is also read when you open the former UI so it could be pushed into GitHub for example.

Each profile can define multiple settings as described in the shared documentation. Here is a schema of the JSON file:

```JSON
{
"profileName": {
    build: {
      compileArgs: string[];
      ninjaArgs: string[];
      buildDirectoryPath: string;
      sdkconfigDefaults: string[];
      sdkconfigFilePath: string;
    };
    env: { [key: string]: string };
    flashBaudRate: string;
    idfTarget: string;
    monitorBaudRate: string;
    openOCD: {
      debugLevel: number;
      configs: string[];
      args: string[];
    };
    tasks: {
      preBuild: string;
      preFlash: string;
      postBuild: string;
      postFlash: string;
    };
  }
}
```

The currently selected profile is not saved in the file and only saved as a workspace state.

> These profiles and each profile settings are going to be saved in the `/path/to/esp-project/esp_idf_project_configuration.json`.

brianignacio5, 2024-09-18:
I will suggest not using `sdkconfigFilePath ` until #1252 is merged since we made fixes to the `SDK Configuration Editor` when using multiple configuration for single project.

mbastida123, 2024-09-18:
THank you @brianignacio5 ! it's all clear now.

I will wait for the merge to use `sdkconfigFilePath`.

Final question: Is there any place on the lower bar where you can see change the configuration profile as well as see which configuration profile are you currently on? The later one is the most important. Something like we see with the IDF version would be really nice:
[image]

brianignacio5, 2024-09-18:
Yes there is an icon showing the currently selected profile and when you click you quickly change between profiles (a dropdown with all profiles option will be shown).

mbastida123, 2024-09-18:
> Yes there is an icon showing the currently selected profile and when you click you quickly change between profiles (a dropdown with all profiles option will be shown).
> 
> If you want to test it now, you can use the extension [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1826718510.zip) from the pull request

I have installed the vsix. I don't know if there's a way to tell that the vsix has been applied though.

Anyway, where's the icon?
[image]

BTW, the `sdkconfigFilePath` seems to work fine. I don't know what was wrong before though.

brianignacio5, 2024-09-18:
After you define the profiles, use `ESP-IDF: Select Project Configuration` command to select a profile.

See if the name is shown in the status command.

mbastida123, 2024-09-18:
> After you define the profiles with the UI, use `ESP-IDF: Select Project Configuration` command to select a profile.
> 
> See if the name is shown in the status command.

Don't know why but now it is showing even though before I did the same steps as now.

Related question: Is this system integrated with idf.py? 
I want to have a CI/CD pipeline that builds all the profiles. I would use idf.py for that. But does idf.py also accept this system or I have to parse the json manually if I want to use idf.py?

brianignacio5, 2024-09-18:
CI/CD is not intended for this feature, since it is a vscode extension feature. For that case I would suggest to go with the `idf.py @config.txt build` approach.

I believe that converting from the json to a .txt file is easy (and could be automated too). We could add a command to convert the currently select profile to such .txt if desired, but probably in CI/CD you might want to have these .txt as part as the repository.

mbastida123, 2024-09-18:
> CI/CD is not intended for this feature, since it is a vscode extension feature. For that case I would suggest to go with the `idf.py @config.txt build` approach.
>
> I believe that converting from the [...]

Makes sense, thank you!

Excelent work on the extension.

closed 2024-09-18 by mbastida123

brianignacio5, 2024-09-18:
If you find the extension useful. Please leave us a review in the [Visual Studio Code Marketplace](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension)

KestrelLuca, 2024-11-22:
hello guys,
My goal is to build the same code for different products conditionally and, as per  @mbastida123, everything works using `idf.py`.
I have created different profiles in Project Configuration Editor which differ just for the **Compile arguments** and **Build Directory path** (all the other fields are empty).

For a given product the **Compile arguments** has e.g. `-DPRODUCT=P1` and then the `CMakeLists.txt` adds the compile define 
`if(PRODUCT STREQUAL "P1")
    add_compile_definitions(P1)
elseif(PRODUCT STREQUAL "P2")
...
else()
    message(FATAL_ERROR "Unknown PRODUCT: ${PRODUCT}")
endif()
`
Now, after selecting the 

As suggested here I tried to add `idf.cmakeCompilerArgs` to the workspace `settings.json`  but the result is the same... I also tried to use a different approach like `"idf.cmakeCompilerArgs": ["-DP1=On"]` and change the CMakelists in 
`if(${P1})
    add_definitions(-DBootstrapMode)
endif()`