espressif/vscode-esp-idf-extension #1311: [Feature Request]: Allow setting hostname for the extension when running code-server in docker environment and / or with tailscale

csvke, 2024-09-14:
**Is your feature request related to a problem? Please describe.**
When running ESP-IDF VS Code extension installed in a Code-Server instance on a host machine, clinking `Configure ESP-IDF Extension` returns a blank page when visit by a browser in a client.
Seems to me the extension is trying to load the page in localhost, which makes sense if the extension is installed locally on a machine, but it will not work if VS Code is accessed through a browser remotely with Code Server. 
Using `code-server --host x.x.x.x` will not work in this case as that parameter aren't passed through to the code executed in individual VS Code extensions.

**Describe the solution you'd like**
I suspect if the ESP-IDF VS Code extension can define it's hostname instead of `localhost`, `Configure ESP-IDF Extension` can be loaded because Code-Server will automatically configure auto forwarding with proxy.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Tried to look into `package.json` but only `OpenOCD Server` can have a customised hostname.

brianignacio5, 2024-09-18:
When you open the `ESP-IDF: Configure ESP-IDF extension`, could you open the Command Palette (F1) and find the `Webview: Open Webview Developer Tools` and check if there is any error in the console output.

brianignacio5, 2024-10-08:
Closing this issue for lack of response. Comment here if you need more assistance.

closed 2024-10-08 by brianignacio5