espressif/vscode-esp-idf-extension #1312: Notification Mode woes

fink-at-trmc-dk, 2024-09-17:
### OS

Linux

### Operating System version

WSL

### Visual Studio Code version

1.93.1

### ESP-IDF version

5.3.1

### Python version

3.12.3

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 6.8.0-45-generic 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 <HOMEPATH>/opt/x-tools/aarch64-rpi3-linux-gnu/bin:<HOMEPATH>/.local/bin:<HOMEPATH>/opt/x-tools/aarch64-rpi3-linux-gnu/bin:<HOMEPATH>/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.93.1 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/bash 
ESP-IDF Extension version 1.8.1 
Workspace folder <HOMEPATH>/projects/fermentor 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/v5.3.1/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/cmake/3.24.0/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/ninja/1.11.1:<HOMEPATH>/.espressif/tools/esp-rom-elfs/20240305
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/.espressif/tools/esp-rom-elfs/20240305/
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/.espressif/python_env/idf5.3_py3.12_env/bin/python
Serial port (idf.port) /dev/ttyUSB0
OpenOCD Configs (idf.openOcdConfigs) board/esp32-wrover-kit-3.3v.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) git
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/cmake/3.24.0/bin: false
Access to <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin: true
Access to <HOMEPATH>/.espressif/tools/ninja/1.11.1: true
Access to <HOMEPATH>/.espressif/tools/esp-rom-elfs/20240305: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
[... 170 lines cut ...]
{"message":"Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:25)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:25)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /dist/extension.js:2:1424988\n\tat ChildProcess.exithandler (node:child_process:415:7)\n\tat ChildProcess.emit (node:events:519:28)\n\tat maybeClose (node:internal/child_process:1105:16)\n\tat Socket.<anonymous> (node:internal/child_process:457:11)\n\tat Socket.emit (node:events:519:28)\n\tat Pipe.<anonymous> (node:net:338:12)","level":"error","timestamp":"2024-09-17T12:31:57.005Z"}
{"message":"ENOENT: no such file or directory, open '/home/fink/projects/fermentor/.vscode/launch.json'","stack":"Error: ENOENT: no such file or directory, open '/home/fink/projects/fermentor/.vscode/launch.json'","level":"error","timestamp":"2024-09-17T12:31:57.484Z"}
{"user":true,"level":"warn","message":"Extension configuration report has been copied to clipboard with errors","timestamp":"2024-09-17T12:31:57.484Z"}

### Description

Regarding "Notification Mode" introduced in 1.7.0:
Since this introduction it has been impossible for me to show the terminal windows after build. It ALWAYS ends up in the output window regardless of the "Notification Mode" setting (applied to all profiles, BTW).
It seems that whatever settings I use, it does not change the behavior and I manually have to click "Terminal" and hereafter the instance of the terminal I need.

Before this was introduce, it simply stayed at the selected window...

### Debug Message

```plain
None
```

### Other Steps to Reproduce

Set "notification Mode" to 1 of the 4 options, build the project and observe that the focus changes to "output" window regardless of the settings...

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-09-18:
Hi @fink-at-trmc-dk Sorry for the issue.

The problem was that after build we print how the user would flash and arguments to do so. Unfortunately we made it so it will show automatically while it shouldn't. We will fix it soon and let you know.

fink-at-trmc-dk, 2024-09-26:
Hi,
Sorry for the late reply, but thanks for looking into it :-)
//Fink

closed 2024-10-17 by github-actions[bot]