espressif/vscode-esp-idf-extension #1316: Same icon for different functions

SergeyKharenko, 2024-09-20:
### OS

Linux

### Operating System version

Fedora Linux 40

### Visual Studio Code version

1.93.1

### ESP-IDF version

v5.2.2

### Python version

3.12.6

### Doctor command output

No need~

### Description

SDK Configuration Editor and Build Project have the same icon(gear icon) on status bar. This caused a lot of trouble for development～
This problem occurs on at least 3 machines with the fedora40 system, and the best way to reproduce it is to do nothing, just find a random fedora40 system and install this plugin,  quite simple!
It may not be difficult to solve, hope it can be solved soon ♥️♥️♥️
[image]

### Debug Message

```plain
No need~
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-09-20:
Which version of this extension are you using ? 

The latest marketplace version has updated the build icon in status bar.

SergeyKharenko, 2024-09-21:
It seems that the icon theme on vscode caused this issue. It could be closed. 
Thank you very much for your answer!!!

closed 2024-09-21 by SergeyKharenko