espressif/vscode-esp-idf-extension #1318: vscode使用uart+gdb调试

ysni, 2024-09-25:
### OS

Linux

### Operating System version

20.0.4

### ESP-IDF version

5.0.6

### Python version

3.8.10

### Doctor command output

idf无输出,还没进入到程序,
[image]

### Description

先要使用vscode进行断点调试,idf.py monitor里使用ctr+c可以进行断点调试,可是vscode里点击调试连不上/dev/ttyUSB0,终端直接执行gdb,也连不上
[image]
[image]
[image]

### Debug Message

```plain
Remote debugging using /dev/ttyUSB0
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
Remote replied unexpectedly to 'vMustReplyEmpty': timeout
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-09-26:
Are you using this extension ? Have you tried using our Eclipse CDT debug adapter configuration as shown here:

https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md

If you are not using our extension and only using the Microsoft C/C++ extension, check in the previous documentation a suggested configuration for `cppdbg`. Since we do not maintain this extension, that is the best we can do.

brianignacio5, 2024-10-08:
Closing this issue for lack of response. Comment here if you need more help.

closed 2024-10-08 by brianignacio5