espressif/vscode-esp-idf-extension #1320: devcontainer creates files as wrong user

tobiashagge, 2024-09-28:
### OS

Windows

### Operating System version

Windows 11 (but within the provided docker image for blink)

### Visual Studio Code version

1.93.1

### ESP-IDF version

1.8.1

### Python version

3.12.3

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.15.153.1-microsoft-standard-WSL2 
System environment variable IDF_PYTHON_ENV_PATH 
 /opt/esp/python_env/idf5.4_py3.12_env 
System environment variable PATH 
 /vscode/vscode-server/bin/linux-x64/38c31bc77e0dd6ae88a4e9cc93428cc27a56ba40/bin/remote-cli:/opt/esp/idf/components/espcoredump:/opt/esp/idf/components/partition_table:/opt/esp/idf/components/app_update:/opt/esp/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:/opt/esp/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin:/opt/esp/tools/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin:/opt/esp/tools/riscv32-esp-elf/esp-14.2.0_20240906/riscv32-esp-elf/bin:/opt/esp/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:/opt/esp/tools/cmake/3.30.2/bin:/opt/esp/tools/openocd-esp32/v0.12.0-esp32-20240821/openocd-esp32/bin:/opt/esp/tools/qemu-xtensa/esp_develop_9.0.0_20240606/qemu/bin:/opt/esp/tools/qemu-riscv32/esp_develop_9.0.0_20240606/qemu/bin:/opt/esp/python_env/idf5.4_py3.12_env/bin:/opt/esp/idf/tools:/opt/qemu/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.93.1 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/bash 
ESP-IDF Extension version 1.8.1 
Workspace folder /workspaces/blink 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) /opt/esp/idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) 
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: d:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: d:\Espressif\tools\esp-rom-elfs\20240305/
Virtual env Python Path (idf.pythonBinPath) /opt/esp/python_env/idf5.4_py3.12_env/bin/python
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) /opt/esp
Git Path (idf.gitPath) /usr/bin/git
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) false
[... 164 lines cut ...]
CMake Error at /opt/esp/idf/tools/cmake/third_party/GetGitRevisionDescription.cmake:78 (configure_file):
  Operation not permitted
Call Stack (most recent call first):
  /opt/esp/idf/tools/cmake/third_party/GetGitRevisionDescription.cmake:93 (get_git_head_revision)
  /opt/esp/idf/tools/cmake/build.cmake:99 (git_describe)
  /opt/esp/idf/tools/cmake/build.cmake:272 (__build_get_idf_git_revision)
  /opt/esp/idf/tools/cmake/idf.cmake:53 (__build_init)
  /opt/esp/idf/tools/cmake/project.cmake:29 (include)
  CMakeLists.txt:5 (include)

-- ccache will be used for faster recompilation
CMake Error at /opt/esp/tools/cmake/3.30.2/share/cmake-3.30/Modules/CMakeDetermineSystem.cmake:218 (configure_file):
  Operation not permitted
Call Stack (most recent call first):
  /opt/esp/idf/tools/cmake/project.cmake:564 (__project)
  CMakeLists.txt:6 (project)
```

### Other Steps to Reproduce

I have reproduced the error on four different machines, all running Windows 11, two with fresh installs of VS code, VS code C++ tools, WSL, and extensions.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

tobiashagge, 2024-09-28:
This seems like the sort of thing that would be user error in the configuration steps but I'm at a bit of a loss. I don't see the behavior if I create another project and add another dev container (python3) to it. I do see the behavior if I install the hello_world example project.

KammutierSpule, 2024-09-29:
I believe I'm facing similar problem on Linux (mind).
The issue is: it can't change ownership of pre-existent files on workspace, but it can create new own files.
This works fine if project is created after devcontainer is created, but won't work if eg: you are importing an existent project (eg: from git)

I post my question on forum:
https://www.esp32.com/viewtopic.php?f=40&t=42076

tobiashagge, 2024-09-30:
The Dockerfile invokes FROM espressif/idf, which in turn invokes FROM ubuntu:24.04. After some testing it appears that the ownership/permissions issue is inherited from ubuntu:24:04. Stripping down the Dockerfile and invoking FROM ubuntu:24:04 directly, the file owner for files written to host by user 'esp' is still incorrect (it is 'ubuntu'). However, downgrading to ubuntu:22.04 eliminates the issue.

The attached Dockerfile is the blink example project Dockerfile with espressif/idf's Dockerfile pasted at the top, but invoking FROM ubuntu:22:04 instead of 24:04. When the hello_world dockerfile is replaced with this one, and entrypoint.sh is downloaded as explained in the Dockerfile comments, the ownership/permissions for the container are correct (workspace files owned 
by 'esp'). Additionally the command 'idf.py set-target esp32' completes without error, as does the build.

[Dockerfile.txt](https://github.com/user-attachments/files/17184704/Dockerfile.txt)

KammutierSpule, 2024-09-30:
I believe solution posted by @tobiashagge is working for me...

tobiashagge, 2024-09-30:
The cause of the issue, and potential solutions, are discussed [at this link.](https://askubuntu.com/questions/1513927/ubuntu-24-04-docker-images-now-includes-user-ubuntu-with-uid-gid-1000). As this appears to be an issue in the espressif/idf docker file, not the VS code extension, I have opened an issue in that repository: https://github.com/espressif/esp-idf/issues/14652.

tobiashagge, 2024-10-03:
Discussion at https://github.com/espressif/esp-idf/issues/14652 has convinced me that the fix should be done here. I think that deleting the ubuntu user after the first line in templates/.devcontainer/Dockerfile will fix the issue. I'm happy to test the fix and submit a patch, but due to hardware issues it may take a week or so.

PR #1282 (2024-10-28): [VSC-1473/1495] update dockerfile and qemu options
## Description

Update Dockerfile and QEMU options.:

- Remove adding QEMU in Dockerfile and using the one from ESP-IDF tools. 
- Adding QEMU options to support both esp32 and esp32c3 as target.
- Add `${idf.buildPath}/merged-qemu.bin` delete when `${idf.buildPath}/.bin_timestamp` changes to trigger rebuilding the merged QEMU bin image.

Fixes #1281
Fixes #1320

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to tes [... cut]
closed 2024-10-24 by github-actions[bot]