espressif/vscode-esp-idf-extension #1323: Can't launch monitor within VSCode (M1 Mac, MacOS Sequoia)

brodendan, 2024-10-01:
### OS

Windows

### Operating System version

15.0

### Visual Studio Code version

1.93.1

### ESP-IDF version

5.3

### Python version

3.12.6

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin arm64 24.0.0 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin:/Library/TeX/texbin:/Applications/VMware Fusion Tech Preview.app/Contents/Public 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.93.1 
Visual Studio Code language en 
Visual Studio Code shell /usr/local/bin/bash 
ESP-IDF Extension version 1.8.1 
Workspace folder <HOMEPATH>/hello_world 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/esp-rom-elfs/20240305
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/.espressif/tools/esp-rom-elfs/20240305/
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/.espressif/python_env/idf5.3_py3.12_env/bin/python
Serial port (idf.port) /dev/tty.SLAB_USBtoUART
OpenOCD Configs (idf.openOcdConfigs) board/esp32s3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) git
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin: true
Access to <HOMEPATH>/.espressif/tools/esp-rom-elfs/20240305: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
[... 219 lines cut ...]
### Description

After upgrading to MacOS Sequoia, ESP-IDF extension having issues.

When trying to launch the monitor it appears to hang on the debug message.

I am able to run it outside of VSCode using 'idf.py monitor' with the standard '. ./export.sh' approach found here - [https://docs.espressif.com/projects/esp ... setup.html](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/linux-macos-setup.html)

Version Details:
OS: Mac OS Sequoia 15.0
VSCode: 1.93.1
ESP-IDF Framework: 5.3
Python: 3.12.6
ESP-IDF Extension: v1.8.1

### Debug Message

```plain
export IDF_PATH='/Users/XXX/esp/esp-idf'
'/Users/XXX/.espressif/python_env/idf5.3_py3.12_env/bin/python' '/Users/XXX/esp/esp-idf/tools/idf_monitor.py' -p /dev/tty.SLAB_USBtoUART -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 '/Users/XXX/hello_world/build/hello_world.elf'
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

radurentea, 2024-10-08:
Hi @brodendan,

I've had a look over the logs, they don't suggest that there is a problem with the monitor. I can see the following errors:

One related to flashing:
`Error: Task ESP-IDF Flash exited with code 2`

And the other one with the serial port which is either busy or doesn't exist
`Serial port /dev/tty.usbserial-21120\n\nA fatal error occurred: Could not open /dev/tty.usbserial-21120, the port is busy or doesn't exist`

Please make the sure to follow the uninstall steps I've told you about in the forum, I will paste them again here:

> I would suggest to make a clean install of the extension. Uninstall it first following these steps: [https://github.com/espressif/vscode-esp ... the-plugin](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/INSTALL.md#uninstalling-the-plugin). Sometimes there might be some esp-idf settings left overs in the settings.json both of your workspace and vs code user's settings. You can open both by using the command palette (CMD + shift + P) and just type "settings.json" (see screenshot) and remove any properties starting with "idf."

After you've reinstalled the extension, try to build, flash and monitor again. If the issue persists, run the "ESP-IDF: Doctor Command" once again and paste the output in your next reply.

closed 2024-10-23 by radurentea

brodendan, 2024-11-10:
Hi @radurentea,

Apologies for the late response as I was overseas. I've returned and followed your instructions: uninstalling ESP-IDF extension as per instructions, restarting & reinstalling and still I am unable to monitor from VSCode. I can build & flash though..

Appreciate any help you can provide as this greatly hampers productivity!

Latest doctor command output:

```

``` ---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin arm64 24.0.0 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /Library/Frameworks/Python.framework/Versions/3.13/bin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin:/Library/TeX/texbin:/Applications/VMware Fusion Tech Preview.app/Contents/Public 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.93.1 
Visual Studio Code language en 
Visual Studio Code shell /usr/local/bin/bash 
ESP-IDF Extension version 1.8.1 
Workspace folder <HOMEPATH>/hello_world 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/cmake/3.24.0/CMake.app/Contents/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/ninja/1.11.1:<HOMEPATH>/.espressif/tools/esp-rom-elfs/20240305
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/.espressif/tools/esp-rom-elfs/20240305/
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/.espressif/python_env/idf5.3_py3.12_env/bin/python
Serial port (idf.port) /dev/tty.SLAB_USBtoUART
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) git
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/cmake/3.24.0/CMake.app/Contents/bin: false
Access to <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin: true
Access to <HOMEPATH>/.espressif/tools/ninja/1.11.1: false
Access to <HOMEPATH>/.espressif/tools/esp-rom-elfs/20240305: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path true
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin: false
Spaces in <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin: false
Spaces in <HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin: false
[... 149 lines cut ...]
{"message":"non zero exit code null\n\n","stack":"Error: non zero exit code null\n\n\n\tat ChildProcess.<anonymous> (/dist/extension.js:2:1421636)\n\tat ChildProcess.emit (node:events:519:28)\n\tat ChildProcess._handle.onexit (node:internal/child_process:294:12)","leve [... cut ...] 8\n\tat ChildProcess.exithandler (node:child_process:415:7)\n\tat ChildProcess.emit (node:events:519:28)\n\tat maybeClose (node:internal/child_process:1105:16)\n\tat Socket.<anonymous> (node:internal/child_process:457:11)\n\tat Socket.emit (node:events:519:28)\n\tat Pipe.<anonymous> (node:net:338:12)","level":"error","timestamp":"2024-11-10T11:46:38.894Z"}

--------------------------------------------------------------------------------------------------------------------------------------------