espressif/vscode-esp-idf-extension #1324: ESP-IDF Extension does not work correctly with fresh install

yamaan93, 2024-10-01:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.93.1

### ESP-IDF version

5.2.3

### Python version

3.10.9

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 c:\program files\nvidia gpu computing toolkit\cuda\v11.8\bin;c:\program files\nvidia gpu computing toolkit\cuda\v11.8\libnvvp;c:\python39\scripts\;c:\python39\;c:\program files\common files\oracle\java\javapath;c:\program files (x86)\common files\oracle\java\javapath;c:\windows\system32;c:\windows;c:\windows\system32\wbem;c:\windows\system32\windowspowershell\v1.0\;c:\windows\system32\openssh\;c:\windows\system32;c:\windows;c:\windows\system32\wbem;c:\windows\system32\windowspowershell\v1.0\;c:\windows\system32\openssh\;c:\program files\microsoft sql server\client sdk\odbc\110\tools\binn\;c:\program files (x86)\microsoft sql server\120\tools\binn\;c:\program files\microsoft sql server\120\tools\binn\;c:\program files\microsoft sql server\120\dts\binn\;c:\program files (x86)\windows kits\8.1\windows performance toolkit\;c:\program files\dotnet\;c:\program files\nodejs\;c:\programdata\chocolatey\bin;c:\program files (x86)\yarn\bin\;c:\program files (x86)\dotnet\;c:\program files (x86)\pulse secure\vc142.crt\x64\;c:\program files (x86)\pulse secure\vc142.crt\x86\;c:\program files\nvidia corporation\nvidia nvdlisr;c:\program files\wooting-analog-sdk\;c:\program files\matlab\r2023b\runtime\win64;c:\program files\matlab\r2023b\bin;c:\program files\matlab\r2022a\runtime\win64;c:\program files\matlab\r2022a\bin;c:\program files\nvidia corporation\nsight compute 2022.3.0\;c:\program files\calibre2\;c:\program files\docker\docker\resources\bin;c:\program files (x86)\stmicroelectronics\stm32 st-link utility\st-link utility;c:\program files\github cli\;c:\program files\git\cmd;c:\mingw\bin;c:\users\ya\appdata\roaming\python\python310\scripts;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\PowerShell\7\;C:\Program Files\Microsoft VS Code\bin;C:<HOMEPATH>\.cargo\bin;C:<HOMEPATH>\AppData\Local\Programs\Python\Python310\Scripts\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python310\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python39\Scripts\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python39\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python38\Scripts\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python38\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:\src\flutter\bin;C:<HOMEPATH>\.dotnet\tools;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin;C:\Program Files\JetBrains\IntelliJ IDEA 2020.3.1\bin;;C:\Program Files\JetBrains\PyCharm 2020.3.2\bin;;C:<HOMEPATH>\AppData\Roaming\npm;C:<HOMEPATH>\AppData\Local\Yarn\bin;C:\Program Files\JetBrains\CLion 2023.3.3\bin;;C:<HOMEPATH>\AppData\Local\Programs\MiKTeX\miktex\bin\x64\;C:<HOMEPATH>\AppData\Local\Programs\oh-my-posh\bin;C:\MinGW\bin;C:\intelFPGA_lite\18.0\modelsim_ase\win32aloem;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Roaming\Python\Python310\Scripts; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.93.1 
Visual Studio Code language en 
Visual Studio Code shell C:\Program Files\PowerShell\7\pwsh.exe 
ESP-IDF Extension version 1.8.1 
Workspace folder c:<HOMEPATH>\repos\ESP32-IDF-Tests 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\v5.3.1\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.38_20240113\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.11.1;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20240305
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20240305/
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM6
OpenOCD Configs (idf.openOcdConfigs) board/esp32-wrover-kit-3.3v.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.38_20240113\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.11.1: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64: true
[... 197 lines cut ...]
```
CMake Error at C:/Users/Ya/esp/v5.3.1/esp-idf/tools/cmake/project.cmake:465 (message):
  Directory specified in EXTRA_COMPONENT_DIRS doesn't exist:
  C:/Users/Ya/repos/components
Call Stack (most recent call first):
  C:/Users/Ya/esp/v5.3.1/esp-idf/tools/cmake/project.cmake:549 (__project_init)
  CMakeLists.txt:13 (project)
```

so i realize I don't have a components dir so I add one and also change it's location in the CMakeLists.txt. Not sure why the default settings don't work here. Once I've done that, the project still doesn't build but now with a different error that I can't seem to figure out:
```
ninja: error: loading 'build.ninja': The system cannot find the file specified.
```

This is about as fresh an installation as it get's yet the extension doesn't seem to have configured itself correctly, which seems like a bug so I'm opening an issue here. 

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

radurentea, 2024-10-07:
Hi @yamaan93,

Unfortunately, the unity example is not being maintained.  To fix your issues, you should change the default value from CmakeLists.txt to `set(EXTRA_COMPONENT_DIRS "")`

The second error is because the components folder you've created is empty, so setting to the value I've recommended above, should solve it.

Currently, we have 3 ways of creating projects via templates and we plan to stick with just one in the future, which will solve the issue of having example projects that are not maintained.

Let me know if you have any other questions!

radurentea, 2024-10-23:
Hi @yamaan93,

I'm following up on this issue. Since there hasn't been any activity for 2 weeks and the solution was provided, I'll proceed to close this issue. 

If you've found the solution helpful and are using our ESP-IDF extension, we'd greatly appreciate if you could rate your experience on the [VS Code Marketplace](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension). Your feedback helps us improve the extension for the entire community.

If you run into any other issues, feel free to open a new issue!

closed 2024-10-23 by radurentea