espressif/vscode-esp-idf-extension #1325: Hide "Search Error Hints" Tab

shebbys, 2024-10-03:
### OS

MacOS

### Operating System version

13.4.1

### Visual Studio Code version

1.93.1

### ESP-IDF version

1.8.1

### Python version

3.9.16

### Doctor command output

n/a

### Description

How do you hide the "ESP-IDF: Search Error Hints" Tab?

[image]

### Debug Message

```plain
n/a
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

radurentea, 2024-10-07:
Hi @shebbys,

To hide the functionality you just need to right click the "..." and deselect it like in this screenshot:
[image]

Or right click directly the "ESP-IDF: Search Error Hint" and press the Hide option:
[image]

Let me know if you've managed to hide it!

shebbys, 2024-10-07:
For some reason I dont have the same options when I do that:

[image]

radurentea, 2024-10-07:
You are using left click, but you need to use right click

shebbys, 2024-10-07:
OMG THANK YOU!! 

I had to use an external mouse to get the right click to work!

closed 2024-10-07 by shebbys