espressif/vscode-esp-idf-extension #1326: Problem with command: "Select OpenOCD Board Configuration"

Gustice, 2024-10-05:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.92.0

### ESP-IDF version

v1.8.1

### Python version

3.11.2

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22631 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files (x86)\Common Files\Oracle\Java\java8path;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Tools\Python\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Microsoft SQL Server\150\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files\Git\cmd;C:\Program Files\dotnet\;C:\msys64\ucrt64\bin;C:\Tools\Python\Scripts;C:\Program Files\Meld\;C:\Program Files\usbipd-win\;;C:\Program Files\Docker\Docker\resources\bin;C:\Program Files\CMake\bin;C:\Program Files\PuTTY\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\.dotnet\tools;C:\msys64\usr\bin;C:<HOMEPATH>\AppData\Local\Programs\oh-my-posh\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.92.0 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.8.1 
Workspace folder d:\MeCorp\Studies\EspIdfDevicePlatform 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:\Tools\ESP-IDF\v5.3\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) C:\Tools\ESP\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin;C:\Tools\ESP\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin;C:\Tools\ESP\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin;C:\Tools\ESP\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin;C:\Tools\ESP\tools\esp32ulp-elf\2.38_20240113\esp32ulp-elf\bin;C:\Tools\ESP\tools\cmake\3.24.0\bin;C:\Tools\ESP\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin;C:\Tools\ESP\tools\ninja\1.11.1;C:\Tools\ESP\tools\idf-exe\1.0.3;C:\Tools\ESP\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Tools\ESP\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Tools\ESP\tools\esp-rom-elfs\20240305
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:\Tools\ESP\tools\openocd-esp32\v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:\Tools\ESP\tools\esp-rom-elfs\20240305/
Virtual env Python Path (idf.pythonBinPath) C:\Tools\ESP\python_env\idf5.3_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM8
OpenOCD Configs (idf.openOcdConfigs) board/esp32c6-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Tools\ESP
Git Path (idf.gitPath) C:\Tools\ESP\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Tools\ESP\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin: true
Access to C:\Tools\ESP\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin: true
Access to C:\Tools\ESP\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin: true
Access to C:\Tools\ESP\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin: true
Access to C:\Tools\ESP\tools\esp32ulp-elf\2.38_20240113\esp32ulp-elf\bin: true
Access to C:\Tools\ESP\tools\cmake\3.24.0\bin: true
Access to C:\Tools\ESP\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin: true
Access to C:\Tools\ESP\tools\ninja\1.11.1: true
Access to C:\Tools\ESP\tools\idf-exe\1.0.3: true
Access to C:\Tools\ESP\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to C:\Tools\ESP\tools\dfu-util\0.11\dfu-util-0.11-win64: true
[... 193 lines cut ...]
{"message":"OpenOCD server failed to start because of Error: missing data from bitq interface\n❌  failed jtag scan\r\nError: missing data from bitq interface\r\nError: failed jtag scan: -104\r\nError: [esp32c6] Failed DMI read at 0x11; status=2\r\nError: [esp32c6] polling failed!\r\n","stack":"Error: OpenOCD server failed to start because of Error: missing data from bitq interface\n\tat Socket.<anonymous> (\\dist\\extension.js:2:1452853)\n\tat Socket.emit (node:events:519:28)\n\tat addChunk (node:internal/streams/readable:559:12)\n\tat readableAddChunkPushByteMode (node:internal/streams/readable:510:3)\n\tat Readable.push (node:internal/streams/readable:390:5)\n\tat Pipe.onStreamRead (node:internal/stream_base_commons:191:23)","level":"error","timestamp":"2024-10-03T14:34:07.036Z"}
{"message":"Cannot access filePath: ${env:ADF_PATH}","stack":"Error: ENOENT: no such file or directory, access 'C:\\Tools\\VsCode4EspIdf\\${env:ADF_PATH}'\n\tat Object.accessSync (node:fs:246:11)\n\tat t.accessSync (node:electron/js2c/node_init:2:9017)\n\tat Se (\\dist\\extension.js:2:1421806)\n\tat \\dist\\extension.js:2:1650824\n\tat ua (\\dist\\extension.js:2:1651892)\n\tat async \\dist\\extension.js:2:1868405","level":"error","timestamp":"2024-10-05T16:03:23.573Z"}
{"message":"Cannot access filePath: ${env:MDF_PATH}","stack":"Error: ENOENT: no such file or directory, access 'C:\\Tools\\VsCode4EspIdf\\${env:MDF_PATH}'\n\tat Object.accessSync (node:fs:246:11)\n\tat t.accessSync (node:electron/js2c/node_init:2:9017)\n\tat Se (\\dist\\extension.js:2:1421806)\n\tat \\dist\\extension.js:2:1651002\n\tat ua (\\dist\\extension.js:2:1651892)\n\tat async \\dist\\extension.js:2:1868405","level":"error","timestamp":"2024-10-05T16:03:23.575Z"}
{"message":"Cannot access filePath: ${env:ESP_MATTER_PATH}","stack":"Error: ENOENT: no such file or directory, access 'C:\\Tools\\VsCode4EspIdf\\${env:ESP_MATTER_PATH}'\n\tat Object.accessSync (node:fs:246:11)\n\tat t.accessSync (node:electron/js2c/node_init:2:9017)\n\tat Se (\\dist\\extension.js:2:1421806)\n\tat \\dist\\extension.js:2:1651094\n\tat ua (\\dist\\extension.js:2:1651892)\n\tat async \\dist\\extension.js:2:1868405","level":"error","timestamp":"2024-10-05T16:03:23.577Z"}
{"message":"Cannot access filePath: ${env:HOMEKIT_PATH}","stack":"Error: ENOENT: no such file or directory, access 'C:\\Tools\\VsCode4EspIdf\\${env:HOMEKIT_PATH}'\n\tat Object.accessSync (node:fs:246:11)\n\tat t.accessSync (node:electron/js2c/node_init:2:9017)\n\tat Se (\\dist\\extension.js:2:1421806)\n\tat \\dist\\extension.js:2:1651190\n\tat ua (\\dist\\extension.js:2:1651892)\n\tat async \\dist\\extension.js:2:1868405","level":"error","timestamp":"2024-10-05T16:03:23.578Z"}
{"message":"Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat \\dist\\extension.js:2:1424988\n\tat ChildProcess.exithandler (node:child_process:415:7)\n\tat ChildProcess.emit (node:events:519:28)\n\tat maybeClose (node:internal/child_process:1105:16)\n\tat ChildProcess._handle.onexit (node:internal/child_process:305:5)","level":"error","timestamp":"2024-10-05T16:03:27.711Z"}

--------------------------------------------------------------------------------------------------------------------------------------------

### Description

It seems that the command Select OpenOCD Board Configuration does not actually do something useful. 
Had to manually change the selected cfg file in settings.json. 
After that i could finally start to debug my device 
ESP32-C6, with openOCD config "board/esp32c6-builtin.cfg"

### Debug Message

```plain
No message produced because this is an
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

radurentea, 2024-10-07:
Hi @Gustice,

I was not able to reproduce the issue you are reporting or maybe I did not understand the issue you are encountering.

Using the command "**ESP-IDF: Select OpenOCD Board Configuration**" from Command Palette, is changing the configuration inside `.vscode/settings.json` as expected.

Could you give me more details?

Gustice, 2024-10-08:
Hello @radurentea 
Thank you for your response! 

Actually i don't really have any further details.
I'm doing exactly that.
1. start command pallet by Ctrl+Shift+P
2. run command "ESP-IDF: Select OpenOCD Board Configuration"
3. three options are offered 
    - ESP32-C6 chip via buildin USB-JTAG
    - ESP32-C6 chip via ESP-PROG
    - ESP32-C6 chip via ESP-PROG-2
4. Selecting "ESP32-C6 chip via ESP-PROG-2" which should switch to esp32c6-bridge.cfg
5. Surprise ... Nothing happens idf.openOcdConfigs stays at board/esp32c6-builtin.cfg

i have noticed that i actually added some own configurations to the settings.json. 
After removing them, i still don't see any effect

Gustice, 2024-10-08:
Additional note: 

- Created a new project based off hello world.
- Started a new session and opened that new project

Changing Target chip has instant effect on idf.adapterTargetName
Changing OpenOCD Board Configuration has no effect but the configuration stays at "board/esp32-wrover-kit-3.3v.cfg"

brianignacio5, 2024-10-11:
It is possible that the settings.json are being saved in the Global Scope (VSCode User settings) and not in the workspace folder. We have a command to determine where to save settings which is default to global.

Open the project in vscode, run the `ESP-IDF: Select where to Save Configuration Settings`, Select workspace folder. Try the `ESP-IDF: Set Espressif Device Target` or `ESP-IDF: Select OpenOCD Board Configuration`.

I think is better to have openOCD config files always be saved in the current project settings.json. We will update this in the extension next release.

Gustice, 2024-10-11:
That's interesting ... it helped, thank you!

Global scope was the evil configuration. Choosing "Workspace" or "Workspace folder" did the trick.
Now both adapterTargetName and openOcdConfigs change according to selected target or openOcd-file respectively. 
(Selecting target adjusts them both, PERFECT, selecting OpenOcdConfig now also works as expected)

One side note: 
idf.saveScope appeared in User-settings.json (Global) after i switched to Workspace policy. Hence the policy was "Global scope" until now. 
However in the past days, switching the target changed the idf.adapterTargetName setting only in the Workspace-Settings.json (locally), which could be not consistent. Rhe User-settings.json (Global) stayed untouched and actually contains the configuration of the very first project (which i somehow missed until now).

Problem is solved for me, Thanks!

brianignacio5, 2024-10-12:
If you find the extension useful. Please leave us a review in the [Visual Studio Code Marketplace](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension)

closed 2024-10-12 by brianignacio5