espressif/vscode-esp-idf-extension #1329: Incorrect installation of extension for VSCode on Windows

kotyara12, 2024-10-21:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.94.2

### ESP-IDF version

5.3.1

### Python version

3.11.2

### Doctor command output

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files (x86)\Borland\Delphi7\Bin;C:\Program Files (x86)\Borland\Delphi7\Projects\Bpl\;C:\Program Files (x86)\Embarcadero\Studio\21.0\bin;C:\Users\Public\Documents\Embarcadero\Studio\21.0\Bpl;C:\Program Files (x86)\Embarcadero\Studio\21.0\bin64;C:\Users\Public\Documents\Embarcadero\Studio\21.0\Bpl\Win64;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Microsoft VS Code\bin;C:\Program Files\Git\cmd;C:\Program Files\PuTTY\;C:\Projects\Delphi7\RavLib7\Share\;C:\Program Files (x86)\Microsoft SQL Server\160\DTS\Binn\;C:\Program Files\Azure Data Studio\bin;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Program Files\Microsoft SQL Server\150\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files\dotnet\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\.platformio\penv\Scripts;C:\Program Files\Azure Data Studio\bin;C:\Projects\Delphi7\RavLib7\Share;C:<HOMEPATH>\.dotnet\tools;c:\Soft\Espressif\python_env\idf5.3_py3.11_env\Scripts; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.94.2 
Visual Studio Code language ru 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.8.1 
Workspace folder c:\Projects\Espressif\test-0005 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:\Projects\Espressif\esp-idf\v5.3.1\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) c:\Soft\Espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin;c:\Soft\Espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin;c:\Soft\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin;c:\Soft\Espressif\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin;c:\Soft\Espressif\tools\esp32ulp-elf\2.38_20240113\esp32ulp-elf\bin;c:\Soft\Espressif\tools\cmake\3.24.0\bin;c:\Soft\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin;c:\Soft\Espressif\tools\ninja\1.11.1;c:\Soft\Espressif\tools\idf-exe\1.0.3;c:\Soft\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;c:\Soft\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;c:\Soft\Espressif\tools\esp-rom-elfs\20240305
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: c:\Soft\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: c:\Soft\Espressif\tools\esp-rom-elfs\20240305/
Virtual env Python Path (idf.pythonBinPath) c:\Soft\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM8
OpenOCD Configs (idf.openOcdConfigs) board/esp32-wrover-kit-3.3v.cfg
ESP-IDF Tools Path (idf.toolsPath) c:\Soft\Espressif\
Git Path (idf.gitPath) c:\Soft\Espressif\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to c:\Soft\Espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin: true
Access to c:\Soft\Espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin: true
Access to c:\Soft\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin: true
Access to c:\Soft\Espressif\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin: true
Access to c:\Soft\Espressif\tools\esp32ulp-elf\2.38_20240113\esp32ulp-elf\bin: true
Access to c:\Soft\Espressif\tools\cmake\3.24.0\bin: true
Access to c:\Soft\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin: true
Access to c:\Soft\Espressif\tools\ninja\1.11.1: true
Access to c:\Soft\Espressif\tools\idf-exe\1.0.3: true
Access to c:\Soft\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
[... 212 lines cut ...]
And the doctor command gives out
`System environment variable IDF_PYTHON_ENV_PATH undefined`
Is this a normal installation?????

3. Thirdly, IDF_PATH and IDF_TOOL_PATH are only available inside VSCode. When called from the command line, nothing is configured and nothing works. How the hell is that possible? An attempt to execute exports.sh complains that there is no python in the system. But you installed it twice, damn it????

4. An attempt to manually register environment variables gave absolutely nothing.

I can try to fix something with my hands, but what??? Reading kilometers of forums for years in useless attempts to fix something....Yes, this is a real pleasure

### Debug Message

```plain
Nothing
```

### Other Steps to Reproduce

1. Just install VS Code, then the ESP-IDF extension.
2. Open the Hello world test project
3. Run idf.py menuconfig either in the vscode terminal or in the windows command line

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-10-22:
When you open a terminal in vscode, the environment variables from the extension are not loaded. For that, run the `ESP-IDF: Open ESP-IDF Terminal` and then `idf.py menuconfig` should work.

You can also use `ESP-IDF: SDK Configuration editor` to open a UI version of idf.py menuconfig with search functionality.

kotyara12, 2024-10-22:
> To achieve that, run the `ESP-IDF: Open ESP-IDF Terminal` command (menu `View` -> Command Palette -> Type `open esp-idf terminal`) which will create a new terminal with ESP-IDF settings loaded then `idf.py menuconfig` should work.

[image]
[image]

If this worked, would it be necessary to write this issue?

> You can also use `ESP-IDF: SDK Configuration editor` to open a UI version of idf.py menuconfig with search functionality.

A very inconvenient tool. All elements in a heap, without division into sections. After three scrolls down from the beginning, it is absolutely not clear to which section this or that setting element belongs.

kotyara12, 2024-10-22:
This is what I need (but it's running from pio)

[image]

[image]

brianignacio5, 2024-10-22:
You are just opening a regular vscode terminal. By running the command `ESP-IDF: Open ESP-IDF Terminal` the extension launch a modified vscode terminal with loaded env variables.

[image]

kotyara12, 2024-10-22:
Oh, God bless it - with menuconfig...
The most important thing is that without "external" idf.py, batch assembly of projects is impossible.
I have several dozen of them now. When something in the libraries has changed - recompiling them all via "manually moving" will be a "hassle"

brianignacio5, 2024-10-23:
1. _idf.py menuconfig keys doesn't work._ Which keys ? Please describe the issue and steps to reproduce it.
2. _The vscode terminal is unsuitable for bulky applications like menuconfig_ ? How ? What makes it unsuitable ? More detail about it would allow us to identify what needs to be fix.

> suggest a real way to run menuconfig from a regular command line without tricks
If you want to use ESP-IDF in any regular terminal outside vscode, please follow the ESP-IDF documentation here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html there is also steps.

The 2 python paths used by the extension are: `c:\Soft\Espressif\tools\idf-python\3.11.2\` is a portable python used to create a virtual env and `c:\Soft\Espressif\python_env\idf5.3_py3.11_env\Scripts\` is the created virtual env python. While the extension install them, they are not added to Windows env PATH globally, it does add it to tasks used by extension  like `ESP-IDF:  Build your project` command or the former `ESP-IDF: Open ESP-IDF Terminal` command.

So what is the issue you are facing? You don't want to use `ESP-IDF: Open ESP-IDF Terminal`  extension command to launch a terminal ? You find vscode terminal unsuitable ? Seems like personal preferences of tools rather than functional issues.

closed 2024-10-25 by kotyara12