espressif/vscode-esp-idf-extension #1334: Debugger problem with project configuration added

executer-uno, 2024-11-01:
### OS

Windows

### Operating System version

Windows 10 Pro

### Visual Studio Code version

1.8.0 (with hotfix from here https://github.com/espressif/vscode-esp-idf-extension/issues/1283)

### ESP-IDF version

5.1.4

### Python version

Python 3.11.2

### Doctor command output

[DOCTOR_output.txt](https://github.com/user-attachments/files/17598261/DOCTOR_output.txt)

### Extension

[extensionLOG.txt](https://github.com/user-attachments/files/17598207/extensionLOG.txt)

### Description

I have a project for ESP32S3 with Project Configurations added. And as soon as configurations were added - JTAG Debug failed to launch (debug via ESP32S3 built in USB JTAG).
So I made some workaround, based on clear [HelloWorld example](https://github.com/espressif/esp-idf/tree/master/examples/get-started/hello_world) - there are two projects in attached archive - hello_world and hello_world_no_config. Second one works great, so Im sure thar hardware and environment are Ok and able to debug via ESP32S3 built in USB JTAG, but project with configuration - not managed to start debug session.
[IDF_HelloWorld_JTAG.zip](https://github.com/user-attachments/files/17597793/IDF_HelloWorld_JTAG.zip)

**hello_world_no_config - works properly.**
compile, download (via UART), monitor - ok. debug - ok, connected and waiting on void app_main(void) entry point. Here is reference OUTPUT  log:
[proper_OUTPUT.txt](https://github.com/user-attachments/files/17598243/proper_OUTPUT.txt)

**hello_world - no debug session starts.**
As soon as project config added (at least one) via ESP-IDF Open Project configuration wizard debug stops working. 
Here is esp_idf_project_configuration.json (also in attached projects):
```
{
  "config_uno": {
    "build": {
      "compileArgs": [],
      "ninjaArgs": [],
      "buildDirectoryPath": "",
      "sdkconfigDefaults": [],
      "sdkconfigFilePath": ""
    },
    "env": {},
    "flashBaudRate": "",
    "idfTarget": "esp32s3",
    "monitorBaudRate": "",
    "openOCD": {
      "debugLevel": 4,
      "configs": [
        "board/esp32s3-builtin.cfg"
      ],
      "args": []
    },
    "tasks": {
      "preBuild": "",
      "preFlash": "",
      "postBuild": "",
      "postFlash": ""
    }
  }
}

```

So, after compile, download (via UART) and check HelloWorld in Monitoring Device console, calling DEBUG button makes following actions:
Error banner appears
[image]
Open OCD server launched and I see restarts of the ESP by the HelloWorld firmware:

```

C:\esp\esp_tools\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin\xtensa-esp32s3-elf-gdb.exe

C:\esp\esp_tools\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin\openocd.exe

[/Flash]
[OpenOCD]
Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html

debug_level: 2

Info : only
 one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and P
ID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000

Info : Listening 
on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections

Info : esp_usb_jtag:
 serial (DC:54:75:F1:7D:14)
Info : esp_usb_jtag: Device found. Base spe
ed 40000KHz, div range 1 to 255
Info : clock spee
d 40000 kHz

Info : JTAG tap
: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: e
sp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)

Info : starting gdb server for e
sp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb conn
ections

Info : [esp32s
3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.

Info : [esp32s3
.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.

Info : [esp32s3.cpu0] Target halted, P
C=0x40378FBA, debug_reason=00000000

Info : [esp32s3
.cpu0] Reset cause (3) - (Software core reset)

Info : [esp32s3.cpu1] Target hal
ted, PC=0x40378FBA, debug_reason=00000000

Info : [esp3
2s3.cpu1] Reset cause (3) - (Software core reset)

Info : [esp32s3.c
pu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.

Info : [esp32s3.
cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.

Info : [esp32s3.cpu0] 
Target halted, PC=0x4004883B, debug_reason=00000000

Info : [esp32s3.
cpu0] Reset cause (3) - (Software core reset)

Info : [esp32s3.cpu1] 
Target halted, PC=0x40041A76, debug_reason=00000000

Info : [esp32s3
.cpu1] Reset cause (3) - (Software core reset)

Info : [esp32s3.c
pu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.

Info : [
esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core wa
s reset.

Info : [esp32s3.c
pu0] Target halted, PC=0x40048C2B, debug_reason=00000000

Info : [esp32s3.cpu0] Reset cause (12) - (Software CPU0 reset)

Info : [esp32s3.cpu1] T
arget halted, PC=0x40041A76, debug_reason=00000000

Info : [esp32s3.cp
u1] Reset cause (12) - (Software CPU1 reset)

Info : [esp32s3.cpu
0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.

Info : [esp3
2s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core wa
s reset.

Info : [esp32s3.cpu0] 
Target halted, PC=0x4004883B, debug_reason=00000000

Info : [e
sp32s3.cpu0] Reset cause (12) - (Software CPU0 reset)

... and so on ...
```

But no debug session launched in VSC.

### Debug Message

```plain
See "Description" section
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

executer-uno, 2024-11-01:
Extra notices: Start/Stop OpenOCD Server command from ESP-IDF menu returns:
```
OPENOCD_SCRIPTS environment variable is missing. Please set it in idf.customExtraVars or in your system environment variables.
C:\esp\esp_tools\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin\openocd.exe
```
Ignoring the fact that OPENOCD_SCRIPTS alredy defined in Settings>Extensions>ESP-IDF>Custom extra vars to proper path
C:\esp\esp_tools\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\share\openocd\scripts

executer-uno, 2024-11-01:
Previous post help for HelloWorld project. But now I can not completly tansfer the solution to my master project. I have following inconsistence: 
In HelloWorld project both "Eclipse CDT GDB Adapter" and "Launch" debug profile starts and connects but in my master project only green play button ▶ works on "Launch" and from debug command bar:
[image]
[image]
But all other debugging buttons leads to error message:
[image]
[image]
[image]
[image]
Error:
[image]

So I feels standing on shaky floor

brianignacio5, 2024-11-04:
Is your environment using the network port `43476` ? This is the default network portused by debug adapter server and if it used it will not work.

You can try updating launch.json like this:

```json
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "gdbtarget",
      "request": "attach",
      "name": "Eclipse CDT GDB Adapter",
      "debugPort": ###
    }
  ]
} 
```

and replace ### for a different server port.

brianignacio5, 2024-11-05:
Hi @executer-uno 

Could you please try with the current master vsix installer [link here](https://nightly.link/espressif/vscode-esp-idf-extension/workflows/ci/master/esp-idf-extension.vsix.zip) ?

I've tried using esp32s3 devkitc-1 with the same project configuration you use in a blink ESP-IDF v5.4 example. I didn't observe the issue you posted (cannot read properties of null). We've might have fixed it in master already.

executer-uno, 2024-11-05:
You are right. VSIX from master works brilliant.
Thank you for quick support!

PR #1587 (2025-06-12): debugPort for debug adapter server in debugger settings
## Description

Fixes #1334

Add `debugPort` to Eclipse CDT GDB Debug configuration settings. Allow the user to customize the generated server for debug adapter and add validation if the server port doesn't exists.

Launch.json example. Change `47856` to whatever port to test. When is not set, 43476 will be used.
```JSON
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "gdbtarget",
      "request": "attach",
      "name": "Eclipse CDT GDB Adapter",
      "debugPort": 47856
    }
  ]
}
```

## Type of change

- Bug fix (no [... cut]
closed 2024-11-05 by executer-uno