espressif/vscode-esp-idf-extension #1339: Problems tab gets cleared after build is finished

fronders, 2024-11-08:
### OS

Windows

### Operating System version

Windows 11 24H2 26100.2161 running WSL2 Ubuntu 22

### Visual Studio Code version

1.95.1

### ESP-IDF version

5.3.1

### Python version

3.10.12 (included with ESP-IDF)

### Doctor command output

[doctor.txt](https://github.com/user-attachments/files/17683180/doctor.txt)

### Description

During the build process any errors & warnings emitted by compiler are shown on the Problems tab in VSCode
Here you can see the the build task log while it is still in progress (compilation finished, but linker is still running)

[image]

Then the build is completed successfully and size task is run. After that it switches to the Output tab displaying the "Build complete" message. Somewhere here (rather when it switches to Output, but it's hard to catch) - the Problems tab gets cleared and all warnings / errors are lost:

[image]
[image]

Same happens for "Build, Flash & Monitor" flow.

It does not happen if the build fails, i.e. Problems tab contains the errors emitted by compiler here:
[image]
[image]

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-11-11:
This is a duplicate of https://github.com/espressif/vscode-esp-idf-extension/issues/764

As described there, whenever you run multiple tasks, the last task clear the existing problems tasks. You can disable the IDF Size task from being executed after build task to maintain the errors from build.

closed 2024-11-11 by brianignacio5

fronders, 2024-11-11:
@brianignacio5 wouldn't it be better to make the size command part of the build task at the very end (if there are no errors)? This is what Eclipse CDT does for example. You don't even have the size task button on the toolbar, so I doubt it gets launched explicitly a lot. That way user would get both the size breakout output and keep the problems tab warnings

After disabling the size output, I also noticed that Monitor task does not erase problems, but the Flash does. Is it possible to modify Size / Flash tasks to behave in a similar way?

brianignacio5, 2024-11-11:
The issue is the way vscode handle task and task output to parse errors.

Each task executed is parsed to populate errors in the problems tabs. I have tested to reuse the same task terminal and have separated terminals without success. The vscode api just return exit code to know if task fail (warnings for example return exit code 0 which is same as successful build) so concatenation of build task + size task will result in same behavior of cleared problems. After a new task is executed, problems always get cleared and I haven't found a way to persist errors in problems tabs (problems are generated by vscode and the extension code can't persist them among tasks execution).

fronders, 2024-11-11:
But how come does monitor not erase Problems then?