espressif/vscode-esp-idf-extension #1340: DUPLICATE (VSC-1498)

brodendan, 2024-11-11:
Hi,

I've created this ticket to highlight that VSC-1498 (https://github.com/espressif/vscode-esp-idf-extension/issues/1323) should still be open. Apologies if this is not the correct workflow!

Regards
-------------------------------------------------------

Hi @radurentea,

Apologies for the late response as I was overseas. I've returned and followed your instructions: uninstalling ESP-IDF extension as per instructions, restarting & reinstalling and still I am unable to monitor from VSCode. I can build & flash though..

Appreciate any help you can provide as this greatly hampers productivity!

Latest doctor command output:

```

``` ---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin arm64 24.0.0 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /Library/Frameworks/Python.framework/Versions/3.13/bin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin:/Library/TeX/texbin:/Applications/VMware Fusion Tech Preview.app/Contents/Public 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.93.1 
Visual Studio Code language en 
Visual Studio Code shell /usr/local/bin/bash 
ESP-IDF Extension version 1.8.1 
Workspace folder <HOMEPATH>/hello_world 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/cmake/3.24.0/CMake.app/Contents/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/ninja/1.11.1:<HOMEPATH>/.espressif/tools/esp-rom-elfs/20240305
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/.espressif/tools/esp-rom-elfs/20240305/
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/.espressif/python_env/idf5.3_py3.12_env/bin/python
Serial port (idf.port) /dev/tty.SLAB_USBtoUART
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) git
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/cmake/3.24.0/CMake.app/Contents/bin: false
Access to <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin: true
Access to <HOMEPATH>/.espressif/tools/ninja/1.11.1: false
Access to <HOMEPATH>/.espressif/tools/esp-rom-elfs/20240305: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path true
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
[... 158 lines cut ...]
{"message":"non zero exit code null\n\nesptool.py v4.8.1\nSerial port /dev/tty.SLAB_USBtoUART\nConnecting...","stack":"Error: non zero exit code null\n\nesptool.py v4.8.1\nSerial port /dev/tty.SLAB_USBtoUART\nConnecting...\n\tat ChildProcess.<anonymous> (/dist/extension.js:2:1421636)\n\tat ChildProcess.emit (node:events:519:28)\n\tat ChildProcess._handle.onexit (node:internal/child_process:294:12)","level":"error","timestamp":"2024-11-10T11:45:40.973Z"}
{"message":"non zero exit code 2\n\nesptool.py v4.8.1\nSerial port /dev/tty.debug-console\n\nA fatal error occurred: Could not open /dev/tty.debug-console, the port is busy or doesn't exist.\n([Errno 35] Could not exclusively lock port /dev/tty.debug-console: [Errno 35] Resource temporarily unavailable)\n\n","stack":"Error: non zero exit code 2\n\nesptool.py v4.8.1\nSerial port /dev/tty.debug-console\n\nA fatal error occurred: Could not open /dev/tty.debug-console, the port is busy or doesn't exist.\n([Errno 35] Could not exclusively lock port /dev/tty.debug-console: [Errno 35] Resource temporarily unavailable)\n\n\n\tat ChildProcess.<anonymous> (/dist/extension.js:2:1421636)\n\tat ChildProcess.emit (node:events:519:28)\n\tat ChildProcess._handle.onexit (node:internal/child_process:294:12)","level":"error","timestamp":"2024-11-10T11:45:41.516Z"}
{"message":"non zero exit code 2\n\nesptool.py v4.8.1\nSerial port /dev/tty.usbserial-21130\n\nA fatal error occurred: Could not open /dev/tty.usbserial-21130, the port is busy or doesn't exist.\n([Errno 16] could not open port /dev/tty.usbserial-21130: [Errno 16] Resource busy: '/dev/tty.usbserial-21130')\n\n","stack":"Error: non zero exit code 2\n\nesptool.py v4.8.1\nSerial port /dev/tty.usbserial-21130\n\nA fatal error occurred: Could not open /dev/tty.usbserial-21130, the port is busy or doesn't exist.\n([Errno 16] could not open port /dev/tty.usbserial-21130: [Errno 16] Resource busy: '/dev/tty.usbserial-21130')\n\n\n\tat ChildProcess.<anonymous> (/dist/extension.js:2:1421636)\n\tat ChildProcess.emit (node:events:519:28)\n\tat ChildProcess._handle.onexit (node:internal/child_process:294:12)","level":"error","timestamp":"2024-11-10T11:45:41.517Z"}
{"message":"non zero exit code 2\n\nesptool.py v4.8.1\nSerial port /dev/tty.usbserial-21130\n\nA fatal error occurred: Could not open /dev/tty.usbserial-21130, the port is busy or doesn't exist.\n([Errno 16] could not open port /dev/tty.usbserial-21130: [Errno 16] Resource busy: '/dev/tty.usbserial-21130')\n\n","stack":"Error: non zero exit code 2\n\nesptool.py v4.8.1\nSerial port /dev/tty.usbserial-21130\n\nA fatal error occurred: Could not open /dev/tty.usbserial-21130, the port is busy or doesn't exist.\n([Errno 16] could not open port /dev/tty.usbserial-21130: [Errno 16] Resource busy: '/dev/tty.usbserial-21130')\n\n\n\tat ChildProcess.<anonymous> (/dist/extension.js:2:1421636)\n\tat ChildProcess.emit (node:events:519:28)\n\tat ChildProcess._handle.onexit (node:internal/child_process:294:12)","level":"error","timestamp":"2024-11-10T11:45:41.818Z"}
{"message":"non zero exit code 2\n\nesptool.py v4.8.1\nSerial port /dev/tty.debug-console\n\nA fatal error occurred: Could not open /dev/tty.debug-console, the port is busy or doesn't exist.\n([Errno 35] Could not exclusively lock port /dev/tty.debug-console: [Errno 35] Resource temporarily unavailable)\n\n","stack":"Error: non zero exit code 2\n\nesptool.py v4.8.1\nSerial port /dev/tty.debug-console\n\nA fatal error occurred: Could not open /dev/tty.debug-con [... cut ...] internal/child_process:1105:16)\n\tat Socket.<anonymous> (node:internal/child_process:457:11)\n\tat Socket.emit (node:events:519:28)\n\tat Pipe.<anonymous> (node:net:338:12)","level":"error","timestamp":"2024-11-10T11:46:38.894Z"}

--------------------------------------------------------------------------------------------------------------------------------------------

_Originally posted by @brodendan in https://github.com/espressif/vscode-esp-idf-extension/issues/1323#issuecomment-2466704613_

brianignacio5, 2024-11-11:
Could you try the current master [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/workflows/ci/master/esp-idf-extension.vsix.zip) and see if the issue is still there?

Could you share the Monitor command output ?

brodendan, 2024-11-11:
Hi @brianignacio5,

Thanks for quick response. I uninstalled previous ESP-IDF extension and installed the vsix installer you linked. Unfortunately now I cannot even build a 'hello world' project. I've attached the doctor command & highlight some error messages around 'pydantic' below.

Any ideas?

**Doctor Command**
```---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin arm64 24.0.0 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /Library/Frameworks/Python.framework/Versions/3.13/bin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin:/Library/TeX/texbin:/Applications/VMware Fusion Tech Preview.app/Contents/Public 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.93.1 
Visual Studio Code language en 
Visual Studio Code shell /usr/local/bin/bash 
ESP-IDF Extension version 1.8.1 
Workspace folder <HOMEPATH>/Hello World_11Nov/hello_world 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/cmake/3.24.0/CMake.app/Contents/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/ninja/1.11.1:<HOMEPATH>/.espressif/tools/esp-rom-elfs/20240305
ESP-IDF extra vars
    OPENOCD_SCRIPTS: <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/.espressif/tools/esp-rom-elfs/20240305/
User extra vars (idf.customExtraVars)
System python Path (idf.pythonInstallPath) /Library/Frameworks/Python.framework/Versions/3.13/bin/python3
Virtual environment Python path (computed) <HOMEPATH>/.espressif/python_env/idf5.3_py3.13_env/bin/python
Serial port (idf.port) /dev/tty.SLAB_USBtoUART
OpenOCD Configs (idf.openOcdConfigs) board/esp32s3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) git
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/cmake/3.24.0/CMake.app/Contents/bin: false
Access to <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin: true
Access to <HOMEPATH>/.espressif/tools/ninja/1.11.1: false
Access to <HOMEPATH>/.espressif/tools/esp-rom-elfs/20240305: true
Access to System python Path (idf.pythonInstallPath) true
Access to Virtual environment Python path (computed) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path true
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin: false
Spaces in <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin: false
Spaces in <HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin: false
[... 256 lines cut ...]
      from pydantic_core import PydanticUndefined
    File "/Users/user/.espressif/python_env/idf5.3_py3.13_env/lib/python3.13/site-packages/pydantic_core/__init__.py", line 6, in <module>
[... cut ...]
  'x86_64'))

Call Stack (most recent call first):
  /Users/user/esp/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!

 *  The terminal process "/usr/local/bin/cmake '-G=Ninja', '-DPYTHON_DEPS_CHECKED=1', '-DESP_PLATFORM=1', '-B=/Users/user/Hello World_11Nov/hello_world/build', '-S=/Users/user/Hello World_11Nov/hello_world', '-DSDKCONFIG=/Users/user/Hello World_11Nov/hello_world/sdkconfig'" terminated with exit code: 1. ```

brianignacio5, 2024-11-13:
This error seems to be related to pydantic using `arm64` build instead of `x86_64`. Are you running from a terminal using Rosetta ? Is this intended behavior ?

You can also try uninstalling and reinstalling with specific architecture like 

`arch -x86_64 <HOMEPATH>/.espressif/python_env/idf5.3_py3.13_env/bin/python -m pip install pydantic --no-cache`

closed 2024-12-04 by github-actions[bot]