espressif/vscode-esp-idf-extension #1346: Project configurations with complier arguments, wrong compiler

KestrelLuca, 2024-11-24:
hello guys,
My goal is to build the same code for different products conditionally and everything works using `idf.py`.
I have created different profiles in Project Configuration Editor which differ just for the **Compile arguments** and **Build Directory path** (all the other fields are empty).

For a given product the **Compile arguments** has e.g. `-DPRODUCT=P1` and then the `CMakeLists.txt` adds the compile define 
```
if(PRODUCT STREQUAL "P1")
    add_compile_definitions(P1)
elseif(PRODUCT STREQUAL "P2")
...
else()
    message(FATAL_ERROR "Unknown PRODUCT: ${PRODUCT}")
endif()
```
Now, after selecting the project configuration and build, it fails.
I have noticed that it is identifying the wrong compilers and I'm getting 
```
-- The C compiler identification is MSVC 19.37.32822.0
-- The CXX compiler identification is MSVC 19.37.32822.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - failed
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.37.32822/bin/Hostx64/x64/cl.exe
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.37.32822/bin/Hostx64/x64/cl.exe - broken
CMake Error at D:/Espressif/tools/cmake/3.24.0/share/cmake-3.24/Modules/CMakeTestCCompiler.cmake:69 (message):  
  The C compiler....
```
instead of 
```
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features       
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info    
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
```

I have no idea why... can you please clarify what is happening?

Since clearly something is changing my configurations, I then decided to remove the project configurations and try to use the  `idf.cmakeCompilerArgs` method.

As suggested here I tried to add `idf.cmakeCompilerArgs` to the workspace `settings.json`  but the result is the same... 

I also tried to use a different approach like `"idf.cmakeCompilerArgs": ["-DP1=On"]` and change the CMakelists in 
```
if(${P1})
    add_compile_definitions(P1)
...
endif()`

or 

`if(${P1})
    add_definitions(-DP1)
...
endif()
```
By removing the `idf.cmakeCompilerArgs` and the profile configurations everything is correctly recognised and the code builds (of course by pressing build I'll get the Unknown PRODUCT: message).

can you please help? what am I missing?
Thanks

_Originally posted by @KestrelLuca in https://github.com/espressif/vscode-esp-idf-extension/issues/1309#issuecomment-2493913798_

KestrelLuca, 2024-11-26:
Hi @brianignacio5, 
when executing the command `idf.py -B builds/build-P1 -DPRODUCT=P1 build` everything works. Alternatively, always within the ESP-IDF terminal (command ESP-IDF: open ESP-IDF terminal) I can also run `idf.py "@P1_Build.txt" build` where the file `P1_Build.txt` contains `-B builds/build-P1 -DPRODUCT=P1`

KestrelLuca, 2024-12-05:
Have you guys experienced the same issue?

brianignacio5, 2024-12-06:
Could you share an example project to reproduce this issue?

I've tried adding in blink example for ESP-IDF v5.4 in the main CMakeLists.txt

```
if(PRODUCT STREQUAL "P1")
    add_compile_definitions(P1)
elseif(PRODUCT STREQUAL "P2")
...
else()
    message(FATAL_ERROR "Unknown PRODUCT: ${PRODUCT}")
endif()
```

And run build with 

```json
"idf.cmakeCompilerArgs": [
    "-G",
    "Ninja",
    "-DPYTHON_DEPS_CHECKED=1",
    "-DESP_PLATFORM=1",
    "-DPRODUCT=P1"
  ]
```

After deleting build directory and build again I can build normally.

KestrelLuca, 2024-12-10:
hi @brianignacio5,
I'm sorry but I don't have any example project to share.
Maybe that the reason why "idf.cmakeCompilerArgs" doesn't work is because I'm using esp-idf 5.3.1?
this is the content of my `settings.json`

```
{
  "C_Cpp.intelliSenseEngine": "default",
  "idf.adapterTargetName": "esp32",
  "idf.portWin": "COM24",
  "idf.flashType": "UART",
  "files.associations": {
    "*.tcc": "cpp",
    "unordered_map": "cpp",
    "initializer_list": "cpp"
  },
 "idf.cmakeCompilerArgs": ["-DPRODUCT=P1"]
}
```
have you tried using the profiles in Project Configuration Editor (in the VS extension)?
regards

brianignacio5, 2024-12-13:
Yes I have tested using both .vscode/settings.json and Project Configuration Editor `idf.cmakeCompileArgs`.

I need to find the point where the error is generated because I can't reproduce the issue myself, that is why I need to some example project that in your setup generates this error.

closed 2025-01-04 by github-actions[bot]