espressif/vscode-esp-idf-extension #1364: deperecated and removed wmic.exe is used internally

AndreasDiet, 2024-12-12:
### OS

Windows

### Operating System version

Win 11 24H2

### Visual Studio Code version

n.a.

### ESP-IDF version

n.a.

### Python version

n.a.

### Doctor command output

n.a.

### Extension

n.a.

### Description

see [https://github.com/espressif/vscode-esp-idf-extension/blob/e61f08a1a242de898f9971b1393f59b34a12fa7c/src/PlatformInformation.ts#L94-L96](url)

wmic.exe is marked as deprecated since release of Windows 10 21H2
see https://learn.microsoft.com/en-us/windows/whats-new/deprecated-features and search for wmic
in Win 11 24H2 it is no longer part of the installation bits

please replace the wmic.exe usage by running
powershell -executionPolicy bypass "(Get-WmiObject Win32_OperatingSystem).OSArchitecture"

### Debug Message

n.a.

### Other Steps to Reproduce

n.a.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-12-13:
Thanks for the issue request. You can test the bug fix in #1365 and will be available in marketplace on our next release.

You can also use the vsix installer from the PR or use the master vsix after it is merged in the meantime.

AndreasDiet, 2024-12-13:
opened a tree of directories - my wmicCatcher is triggered, indicating wmic.exe is used
installed the vsix
opened the very same directory tree- my wmicCatcher is no longer be triggered

thanks for changing to make this extension compatible to Win11 24H2/Server 2025

PR #1365 (2025-01-08): [VSC-1538] rm use of wmic in favor of Get WmiObject
## Description

Remove deprecated `wmic` in favor of calling `Get-WmiObject Win32_OperatingSystem` from powershell.

Fixes #1364

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "ESP-IDF: Configure ESP-IDF Extension"
2. Execute action. Setup should work normally in Windows regardless of current terminal defaults shell. Check ESP-IDF: Doctor Command.
3. Observe results. 

**Test Configuration**:
* ESP [... cut]
closed 2024-12-13 by AndreasDiet