espressif/vscode-esp-idf-extension #1368: JTAG flashing error on 1.9.0

herna, 2024-12-16:
### OS

Windows

### Operating System version

Windows 10

### ESP-IDF version

1.96

### Python version

3.11

### Doctor command output

[doctor.txt](https://github.com/user-attachments/files/18146412/doctor.txt)

### Extension

[esp_idf_vsc_ext.log](https://github.com/user-attachments/files/18146431/esp_idf_vsc_ext.log)

### Description

I have an ESP-PROG connected to ESP32

I can't flash through JTAG unless ESP32 is in "boot" mode and UART0 connected.

Even if you try to flash ESP32 through JTAG, it seems version 1.9.0 of the extension tries to use UART connection somehow no matter what. The extension says at the end "flash through JTAG" but as I said, UART connection must be available and device in boot mode.

I just downgraded to version 1.8.1 and everything works as expected, I am able to use JTAG without UART connection.

There are other users with this problem: https://esp32.com/viewtopic.php?f=40&t=43461

### Debug Message

[debug.txt](https://github.com/user-attachments/files/18146423/debug.txt)

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

closed 2025-01-08 by github-actions[bot]

herna, 2025-01-08:
Hello @radurentea,

Yes, working fine now.

Thank you.

mert-yurt, 2025-01-15:
Hi,
I have the same problem. I can debug and flash by UART but I cant by JTAG. It was properly working last week with JTAG. I dont know what did it happen. This is my entire output;

Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html

[/Flash]
[OpenOCD]
Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection

Info : accepting 'tcl' connection on tcp/6666

Flashing "c:/HW_test1/build/bootloader/bootloader.bin" at 0x1000

** program_esp input args <0x1000 verify> **

Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg:
 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)

Info : [esp32.cpu0] requesting target halt and executing a soft reset

Info : [esp32.cpu0] Debug controller was reset.

Info : [esp32.cpu0] Core was reset.

Info : [esp32.cpu0] Target halted, PC=0x500000CF, debug_reason=00000000

Info : [esp32.cpu0] Reset cause (3) - (Software core reset)

Info : [esp32.cpu1] requesting target halt and executing 
a soft reset

Info : [esp32.cpu0] Core was reset.

Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_
reason=00000000
Info : [esp32.cpu1] De
bug controller was reset.
Info : [esp32.cpu1] Core was reset.

Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000

Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset 
by CPU0)

Info : [esp32.cpu0] Reset cause (3) - (Software core reset)

** Programming Started **

** Flashing Failed **
-1

[/OpenOCD]
[Flash]
Failed to flash the device (JTag), please try again [got response: '-1', expecting: '0']
Flash has finished. You can monitor with ESP-IDF: Monitor your device command

Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html

Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection

Info : accepting 'tcl' connection on tcp/6666

Flashing "c:/HW_test1/build/bootloader/bootloader.bin" at 0x1000
** program_esp input args <0x1000 verify> **

Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/
device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)

Warn : [esp32.cpu1] Unexpected OCD_ID = 00000000

Info : [esp32.cpu0] requesting target halt and executing a soft reset

Info : [esp32.cpu0] Target halted, PC=0x4000940E, debug_reason=00000
000

Info : [esp32.cpu0] Debug controller was reset.

Info : [esp32.cpu0] Core was reset.

Info : [esp32.cpu0] Target halted, PC=0x500000CF, debug_reaso
n=00000000

Info : [esp32.cpu0] Reset cause (3) - (Software core reset)

❌ Info : [esp32.cpu1] requesting target halt and executing a soft reset
Error: Target not examined yet

Info : [esp32.cpu0] Core was reset.

Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000000

Info : [esp32.cpu1] Debug controller was
 reset.

Info : [esp32.cpu1] Core was reset.

Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000

Info : [esp32.cpu1] Reset cause 
(14) - (CPU1 reset by CPU0)

Info : [esp32.cpu0] Reset cause (3) - (Software core reset)

** Programming Started **

** Flashing Failed **
-1

[/OpenOCD]
[Flash]
Failed to flash the device (JTag), please try again [got response: '-1', expecting: '0']
Flash has finished. You can monitor with ESP-IDF: Monitor your device command

Does anyone have any ideas about this problem?

mert-yurt, 2025-01-20:
hi @radurentea ,
Thanks for your answering,

I downloaded the VSIX file and I tried to the flashing. this is my output;

[OpenOCD]
Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

jtag

Info : Listening on port 6666 for tcl connections

Info : Listening on port 4444 for telnet connections

Info : clock speed 1000 kHz

Info : JTAG tap: esp32
.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)

Info : [esp32.cpu0] Examination succeed

Info : [esp32.cpu1] Examination succeed

Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections

Info : [esp32.cpu0] Target halted, PC=0x400093DB, debug_reason=00000000

Info : [esp32.cpu0] Reset cause (3) - (Software core reset)

Info : Set GDB target to 'esp32.cpu0'

Info : [esp32.cpu1] Target halted, PC=0x400076DD, debug_reason=00000000

Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)

Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection

Info : accepting 'tcl' connection on tcp/6666

Flashing "c:/Users/../
esp/project/HW_test1/build/bootloader/bootloader.bin" at 0x1000
** program_esp input args <0x1000 verify> **

Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), 
part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)

Info : [esp32.cpu0] requesting target halt and executing a soft reset

Info : [esp32.cpu0] Debug controller was reset.

Info : [esp32.cpu0] Core was reset.

Info : [esp32.cpu0] Target halted, PC=0x500000CF, debug_reason=00000000

Info : [esp32.cpu0] Reset cause (3) - (Software core reset)

Info : [esp32.cpu1] requesting target halt and executing a soft rese
t

Info : [esp32.cpu0] Core was reset.

Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
[... 145 lines cut ...]

Info : dropped 'gdb' connection

[/OpenOCD]
[Debug Adapter]
Exception in thread Thread-1 (_readerthread):
Traceback (most recent call last):
  File "threading.py", line 1038, in _bootstrap_inner

Exception in thread Thread-2 (_readerthread):
Traceback (most recent call last):
  File "threading.py", line 1038, in _bootstrap_inner

  File "threading.py", line 975, in run
  File "threading.py", line 975, in run

  File "subprocess.py", line 1552, in _readerthread
  File "subprocess.py", line 1552, in _readerthread
OSError: [Errno 22] Invalid argument
OSError: [Errno 22] Invalid argument

DEBUG_ADAPTER_STOPPED
2025-01-20 16:09:36,574 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED

[Stopped] : ESP-IDF Debug Adapter

mert-yurt, 2025-02-17:
hi @radurentea,

I did everything what you said but it still doesn't work. I noticed some things. If I try to debug the device with JTAG after programming it with Serial UART, it fails on the first try but succeeds on the second try. I dont know does it help us for the solve the problem. 

this is the first and second debugging;

[/OpenOCD]
[Debug Adapter]
DEBUG_ADAPTER_STARTED
2025-02-17 16:51:13,295 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
DEBUG_ADAPTER_READY2CONNECT
2025-02-17 16:51:13,300 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT

[/Debug Adapter]
[OpenOCD]
Info : accepting 'gdb' connection on tcp/3333

Info : [esp32.cpu0] Target halted, PC=0x40084D96, debug_reason=00000000

Info : Set GDB target to 'esp32.cpu0'

Info : [esp32.cpu1] Target halted, PC=0x40084D96, debug_reason=00000000

Warn : No symbols for FreeRTOS!

Info : [esp32.cpu0] Target halted, PC=0x400BE65A, debug_reason=00000001

Info : Flash mapping 0: 0x10020 -> 0x3f400020, 44 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 91 KB

Info : [esp32.cpu0] Target halted, PC=0x400BE65A, debug_reason=00000001

Info : Auto-detected flash bank 'esp32.cpu0.flas
h' size 4096 KB
Info : Using flash bank 'esp32.cpu0.flash' size 4096 KB

Info : [esp32.cpu0] Target halted, PC=0x400BE65A, debug_reason=00000001

Info : Flash mapping 0: 0x10020 -> 0x3f400020, 44 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 91 KB
Info :
 Using flash bank 'esp32.cpu0.irom' size 92 KB

Info : [esp32.cpu0] Target halted, PC=0x400BE65A, debug_reason=00000001

Info : Flash mapping 0: 0x10020 -> 0x3f400020, 44 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 91 KB
Info : Using
❌  flash bank 'esp32.cpu0.drom' size 48 KB
Info : New GDB Connection: 1, Target esp32.cpu0, state: halted
Error: GDB missing ack(2) - assumed good

Info : dropped 'gdb' connection

[/OpenOCD]
[Debug Adapter]
DEBUG_ADAPTER_STOPPED
2025-02-17 16:51:26,933 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED

[Stopped] : ESP-IDF Debug Adapter
C:\Users\mert.yurt\esp\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin\openocd.exe

DEBUG_ADAPTER_STARTED
2025-02-17 16:51:34,376 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
DEBUG_ADAPTER_READY2CONNECT
2025-02-17 16:51:34,381 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT

[/Debug Adapter]
[OpenOCD]
Info : accepting 'gdb' connection on tcp/3333

Warn : No symbols for FreeRTOS!
Info : New GDB Connection: 
2, Target esp32.cpu0, state: halted

Warn : Prefer GDB command "target extended-remote :3333" instead of "target remote :3333"

Info : Detected FreeRTOS version: (10.5.1)

Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)

Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)

Info : [esp32.cpu0] requesting target halt and executing a soft reset

Info : [esp32.cpu0] Debug controller was reset.

Info : [esp32.cpu0] Core was reset.

Info : [esp32.cpu0] Target halted, PC=0x500000CF, debug_reason=00000000

Info : [esp32.cpu0] Reset cause (3) - (Software core reset)

Info : [esp32.cpu1] requesting target halt and executing a soft reset

Info : [esp32.cpu0] Core was reset.

Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000000

Info : [esp32.cpu1] Debug controller was reset.

Info : [esp32.cpu1] Core was reset.

Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000

Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)

Info : [esp32.cpu0] Reset cause (3) - (Software core reset)

Info : [esp32.cpu0] Target halted, PC=0x400BE65A, debug_reason=00000001

Info : Flash mapping 0: 0x10020 -> 0x3f400020, 44 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 91 KB

Info : Using flash bank 'esp32.cpu0.irom' size 92 KB

Info : [esp32.cpu0] Target halted, PC=0x400BE65A, debug_reason=00000001

Info : Flash mapping 0: 0x10020 -> 0x3f400020, 44 KB
Info : F
lash mapping 1: 0x20020 -> 0x400d0020, 91 KB
Info : Using flash bank 'esp32.cpu0.drom' size 48 KB

Info : [esp32.cpu0] Target halted, PC=0x
400BE65A, debug_reason=00000001

Info : Flash mapping 0: 0x10020 -> 0x3f400020, 44 KB
Info : Flash map
ping 1: 0x20020 -> 0x400d0020, 91 KB
Info : Using flash bank 'esp32.cpu1.irom' size 92 KB

Info : [esp32.cpu0] Target halted, PC=0x400BE65A, debug_reason=00000001

Info : Flash mapping 0: 0x10020 -> 0x3f
400020, 44 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 91 KB
Info : Using flash bank 'esp32.cpu1.drom' size 48 KB

Info : dropped 'gdb' connection

Info : accepting 'gdb' connection on tcp/3333

Info : Detected FreeRTOS version: (10.5.1)

Info : New GDB Connection: 3, Target esp32.cpu0, state: halted

Info : [esp32.cpu0] Target halted, PC=0x400D5693, debug_reason=00000001

Info : Set GDB target to 'esp32.cpu0'

Info : [esp32.cpu1] Target halted, PC=0x40084D96, debug_reason=00000000

mert-yurt, 2025-02-17:
Btw this is the "Doctor command" output;

[report.txt](https://github.com/user-attachments/files/18826422/report.txt)