espressif/vscode-esp-idf-extension #1371: Unit Testing: Paring of test cases fails for TEST_CASE() formatted with spaces

tmrech, 2024-12-17:
### OS

Linux

### Operating System version

Ubuntu Ubuntu 20.04.6 LTS (WSL)

### Visual Studio Code version

1.96.0

### ESP-IDF version

5.3.1

### Python version

3.8.10

### Doctor command output

[report.txt](https://github.com/user-attachments/files/18164663/report.txt)

### Description

When using the TESTING tab in VSCode, testcases are only parsed correctly if the unity macro `TEST_CASE("testname", "group")` is formatted without space before testname. If a space is inserted like this `TEST_CASE( "testname", "group" )`, the testcase won't show up in the TESTING tab.

This is consistent with the regular expression given in docs_espressif/en/additionalfeatures/unit-testing.rst but causes issues when auto-formating is activated. As the test project will build correctly with this formatting the regex should be changed?

Correct behaviour:
[image]

Faulty behaviour:
[image]

### Debug Message

-

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-12-18:
We can fix the regex to include this case too. I'm just wondering which auto-formatter are you using and why it formats this way ?

tmrech, 2024-12-18:
That is great news, thanks.

> We can fix the regex to include this case too. I'm just wondering which auto-formatter are you using and why it formats this way ?

I'm using clang-format with some custom clang-format style options. I think the relevant option in .clang-format is `SpacesInParentheses: true`

brianignacio5, 2024-12-23:
Could you please try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/2348025072.zip) to see if the issue is fixed ?

tmrech, 2024-12-25:
I'm off my computer until early January. I will Check it out when I'm back.

tmrech, 2025-01-08:
The vsix installer fixes the issue. Thanks!

PR #1377 (2025-01-13): [VSC-1544] update test case regex
## Description

Update TEST_CASE regex to support spaces between ( and " such as:

`TEST_CASE( "New test i just made" , "[mean][fails]" )` and `TEST_CASE("New test d i just made","[mean][fails]")`

this happens when clang auto format on test_*.c files.

Fixes #1371

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Open a ESP-IDF projects with unit tests like the [unity_test]() example.
2. Click on `Testing` tab in the side pane [... cut]