espressif/vscode-esp-idf-extension #1378: Unable to support multiple ESP-IDF versions simultaneously on v1.9.0 extension

go2net, 2024-12-21:
### OS

Windows

### Operating System version

Windows11

### Visual Studio Code version

1.96.2

### ESP-IDF version

4.4.6 and 5.3.1

### Python version

3.13.1

### Doctor command output

This issue is solely an extension-related problem and is not hardware-related.

### Description

Until a week ago, everything was working perfectly. I could use both ESP-IDF v5.3.2 and v4.4.6 for different projects and switch seamlessly between versions. However, now when I try to switch versions, only the last configured version is recognized using the 'Configure ESP-IDF Extension' option. Despite trying various approaches—placing different versions in separate locations, installing the older version first or the newer version first, reinstalling VS Code, and even reinstalling Windows—the issue persists. The extension always defaults to the most recently configured version.

When I downgraded to espressif.esp-idf-extension version 1.8.1, everything returned to normal, confirming that this issue is related to the extension.

### Debug Message

No log, this issue is easy to reproduce

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-12-23:
The doctor command provides extension related configuration and is necessary to understand the issue you are facing.

> when I try to switch versions, only the last configured version is recognized using the 'Configure ESP-IDF Extension' option.

What do you mean ? In Show Existing setups ? Are you sure the setup are valid? Maybe there is a setting value that is failing the validation ? The doctor command and the extension log could help us identify what is the error that is happening in your setup.

brianignacio5, 2024-12-25:
I think I found the issue.

Could you please try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/2361181251.zip) to see if it fixes the issue ?

You can test these changes by installing this VSIX by click menu View -> Command Palette..., type Install from VSIX and then select downloaded esp-idf-extension.vsix file to install the extension.

go2net, 2025-01-09:
[image]
After installing your VSIX file, it shows an error stating "configuration is not valid." Additionally, when I try to select a version, it displays an empty list with no options available.
I tried restarting VS Code, but the problem persists. I rolled back to version 1.8.1, and everything works fine.

brianignacio5, 2025-01-13:
Apologies there were some issue with CI and I didn't get the right vsix installer. 

Could you try this one [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/2420947975.zip)

go2net, 2025-01-16:
With the latest installer, the warning no longer appears. However, when I try to select a version, it still displays an empty list with no available options. Changed back to 1.8.1, no this issue.

brianignacio5, 2025-01-17:
I keep working on fixing this issue and pushed new changes.

You can try with this [new vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/2446125161.zip)

PR #1384 (2025-01-20): [VSC-1546] fix version switch by persist sys python in idfSetup
## Description

Because ESP-IDF version now computes the virtual env python path from a `idf.pythonInstallPath` and current ESP-IDF version, there is an issue for v4.4.8 (uses embed Python 3.8) and v5.3.1 (use embed Python 3.11) calculates the wrong env path because this embed python is not persisted in idfSetup in extension global context. This leads to a wrong python env path calculated which failed the verification when showing the list of available IDF setups in the version switcher or `ESP-IDF: Configure ESP-IDF extension` co [... cut]
closed 2025-01-20 by brianignacio5