espressif/vscode-esp-idf-extension #1380: GDB works on command line. Does not work via extension

gamename, 2024-12-22:
### OS

MacOS

### Operating System version

15.2 (24C101) (Sequoia)

### Visual Studio Code version

Version: 1.96.1 (Universal) Commit: 42b266171e51a016313f47d0c48aca9295b9cbb2 Date: 2024-12-17T17:50:05.206Z Electron: 32.2.6 ElectronBuildId: 10629634 Chromium: 128.0.6613.186 Node.js: 20.18.1 V8: 12.8.374.38-electron.0 OS: Darwin arm64 24.2.0

### ESP-IDF version

5.3.1

### Python version

3.12.27

### Doctor command output

[122224-report.txt.zip](https://github.com/user-attachments/files/18222492/122224-report.txt.zip)

### Extension

[esp_idf_vsc_ext.log.zip](https://github.com/user-attachments/files/18222497/esp_idf_vsc_ext.log.zip)

### Description

I am able to run GDB with no problems from the command line.

Example:
```
idf openocd gdb
Executing action: openocd
Note: OpenOCD cfg not found (via env variable OPENOCD_COMMANDS nor as a --openocd-commands argument)
OpenOCD arguments default to: "-f board/esp32c3-builtin.cfg"
OpenOCD started as a background task 37374
Executing action: gdb
GNU gdb (esp-gdb) 14.2_20240403
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=aarch64-apple-darwin21.1 --target=riscv32-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
add symbol table from file "/Users/tennis/src/C2DS-egg/build/bootloader/bootloader.elf"
app_main () at /Users/tennis/src/C2DS-egg/main/main.c:223
223	    esp_gdbstub_init();
add symbol table from file "/Users/tennis/.espressif/tools/esp-rom-elfs/20240305/esp32c3_rev3_rom.elf"
JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
[esp32c3] Reset cause (3) - (Software core reset)
Hardware assisted breakpoint 1 at 0x42009046: file /Users/tennis/src/C2DS-egg/main/main.c, line 223.
[New Thread 1070244608]
[Remote target exited]
[New Thread 1070245976]
[New Thread 1070239824]
[Switching to Thread 1070244608]

Thread 2 "main" hit Temporary breakpoint 1, app_main () at /Users/tennis/src/C2DS-egg/main/main.c:223
223	    esp_gdbstub_init();
(gdb) ```

### Debug Message

```1: (49) LaunchOptions{"name":"ESP32 GDB Stub Debug","type":"cppdbg","request":"launch","program":"/Users/tennis/src/C2DS-egg/build/CoopCopEgg.elf","cwd":"/Users/tennis/src/C2DS-egg","miDebuggerPath":"/Users/tennis/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin/xtensa-esp32-elf-gdb","miDebuggerServerAddress":"/dev/cu.usbmodem21101","setupCommands":[{"description":"Enable pretty-printing for gdb","text":"-enable-pretty-printing","ignoreFailures":true}],"externalConsole":false,"logging":{"engineLogging":true,"logToFile":true,"logLevel":4},"__configurationTarget":6,"__sessionId":"aada2960-29d6-45f8-b1f4-faf0af203a09","MIMode":"gdb"}
1: (129) codeSign-stderr: Executable=/Users/tennis/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin/xtensa-esp32-elf-gdb
1: (133) Starting: "/Users/tennis/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin/xtensa-esp32-elf-gdb" --interpreter=mi
1: (174) DebuggerPid=40721
1: (380) ->=thread-group-added,id="i1"
1: (381) ->~"GNU gdb (esp-gdb) 14.2_20240403\n"
1: (381) ->~"Copyright (C) 2023 Free Software Foundation, Inc.\n"
1: (381) ->~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
1: (381) ->~"\nType \"show copying\" and \"show warranty\" for details.\n"
1: (382) ->~"This GDB was configured as \"--host=aarch64-apple-darwin21.1 --target=xtensa-esp-elf\".\n"
1: (382) ->~"Type \"show configuration\" for configuration details.\n"
1: (382) ->~"For bug reporting instructions, please see:\n"
1: (382) ->~"<https://www.gnu.org/software/gdb/bugs/>.\n"
1: (382) ->~"Find the GDB manual and other documentation resources online at:\n    <"
1: (382) ->~"http://www.gnu.org/software/gdb/documentation/>.\n\n"
1: (382) ->~"For help, type \"help\".\n"
1: (382) ->~"Type \"apropos word\" to search for commands related to \"word\".\n"
1: (382) ->(gdb)
1: (385) <-1001-gdb-set mi-async on
1: (386) ->1001^done
1: (386) ->(gdb)
1: (387) 1001: elapsed time 2
1: (394) <-1002-enable-pretty-printing
1: (394) ->1002^done
1: (394) ->(gdb)
1: (394) 1002: elapsed time 0
1: (394) <-1003-interpreter-exec console "set pagination off"
1: (395) ->=cmd-param-changed,param="pagination",value="off"
1: (395) ->1003^done
1: (395) ->(gdb)
1: (395) 1003: elapsed time 0
1: (395) <-1004-gdb-set auto-solib-add on
1: (395) ->1004^done
1: (395) ->(gdb)
1: (395) 1004: elapsed time 0
1: (395) <-1005-gdb-set solib-search-path /Users/tennis/src/C2DS-egg/build:
1: (395) ->1005^done
1: (395) ->(gdb)
1: (395) 1005: elapsed time 0
1: (395) <-1006-environment-cd /Users/tennis/src/C2DS-egg
1: (396) ->1006^done
1: (396) ->(gdb)
1: (396) 1006: elapsed time 0
1: (396) <-1007-file-exec-and-symbols /Users/tennis/src/C2DS-egg/build/CoopCopEgg.elf
1: (443) ->1007^done
1: (443) ->(gdb)
1: (443) 1007: elapsed time 47
1: (444) <-1008-interpreter-exec console "show architecture"
1: (444) ->~"The target architecture is set to \"auto\" (currently \"xtensa\").\n"
1: (444) ->1008^done
1: (444) ->(gdb)
1: (444) 1008: elapsed time 0
1: (445) <-1009-break-insert -f main
1: (447) ->&"Function \"main\" not defined.\n"
1: (447) ->1009^done,bkpt={number="1",type="breakpoint",disp="keep",enabled="y",addr="<PENDING>",pending="main",times="0",original-location="main"}
1: (447) ->(gdb)
1: (450) 1009: elapsed time 4
1: (450) <-1010-target-select remote /dev/cu.usbmodem21101
```

### Other Steps to Reproduce

1. Click on the "Run & Debug" icon on the left of the IDE
2. Click on the "Run & Debug" icon above the "Variables" panel
3. The only thing that happens is the floating control panel appears with only the restart and stop icons enabled
[image]

Here is my `launch.json`:
```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "ESP32 GDB Stub Debug",
            "type": "cppdbg",
            "request": "launch",
            "program": "${workspaceFolder}/build/CoopCopEgg.elf",
            "cwd": "${workspaceFolder}",
            "miDebuggerPath": "/Users/tennis/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin/xtensa-esp32-elf-gdb",
            "miDebuggerServerAddress": "/dev/cu.usbmodem21101",
            "setupCommands": [
                {
                    "description": "Enable pretty-printing for gdb",
                    "text": "-enable-pretty-printing",
                    "ignoreFailures": true
                }
            ],
            "externalConsole": false,
            "logging": {
                "engineLogging": true,
                "logToFile": true,
                "logLevel": 4,
            }
        }
    ]
}
```

and here is settings.json:
```json
{
    "editor.formatOnSave": true,
    "clang-format.style": "{ BasedOnStyle: LLVM, IndentWidth: 4, ColumnLimit: 120 }",
    "idf.adapterTargetName": "esp32c3",
    "sonarlint.pathToCompileCommands": "${workspaceFolder}/build/compile_commands.json",
    "terminal.integrated.profiles.osx": {
        "MyCustomProfile": {
            "path": "/bin/zsh",
            "args": [
                "-c",
                "source ~/.zshrc && setidf && export COMPONENT_KCONFIGS_DIR=$(pwd)/components && export EXTRA_COMPONENT_DIRS=$(pwd)/components && exec zsh"
            ]
        }
    },
    "terminal.integrated.defaultProfile.osx": "MyCustomProfile",
    "git.autoRepositoryDetection": true,
    "git.ignoreLimitWarning": true,
    "terminal.integrated.fontSize": 12,
    "files.associations": {
        "esp_netif_types.h": "c",
        "subscription_manager.h": "c"
    },
    "idf.port": "/dev/cu.usbmodem21101",
    "idf.openOcdDebugLevel": 4
}
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-12-23:
Why not use our Eclipse CDT debug adapter?

launch.json
```json
{
  "configurations": [
    {
      "type": "gdbtarget",
      "request": "attach",
      "name": "Eclipse CDT GDB Adapter"
    }
  ]
}
```

If you want to use the Microsoft C/C++ extension for debug, check this template configuration:

```json
{
  "configurations": [
    {
      "name": "GDB",
      "type": "cppdbg",
      "request": "launch",
      "MIMode": "gdb",
      "miDebuggerPath": "${command:espIdf.getToolchainGdb}",
      "program": "${workspaceFolder}/build/${command:espIdf.getProjectName}.elf",
      "windows": {
        "program": "${workspaceFolder}\\build\\${command:espIdf.getProjectName}.elf"
      },
      "cwd": "${workspaceFolder}",
      "setupCommands": [
			  { "text": "set remotetimeout 20" },
		  ],
		  "postRemoteConnectCommands": [
			  { "text": "mon reset halt" },
			  { "text": "maintenance flush register-cache"},
		  ],
      "externalConsole": false,
      "logging": {
        "engineLogging": true
      }
    }
  ]
}
```

gamename, 2024-12-23:
@brianignacio5 

> Eclipse CDT

I'm confused. I use vscode, not Eclipse.  How does Eclipse factor into this at all?

brianignacio5, 2024-12-23:
esp-idf extension implements its own debug adapter, it is a fork of https://github.com/eclipse-cdt-cloud/cdt-gdb-adapter which is a vscode extension from the Eclipse CDT team

gamename, 2024-12-23:
@brianignacio5 

> esp-idf extension implements its own debug adapter, it is a fork of https://github.com/eclipse-cdt-cloud/cdt-gdb-adapter which is a vscode extension from the Eclipse CDT team.
> 
> You just need to add a configuration in launch.json to use it:
> 
> ```json
> {
>   "configurations": [
>     {
>       "type": "gdbtarget",
>       "request": "attach",
>       "name": "Eclipse CDT GDB Adapter"
>     }
>   ]
> }
> ```
I made the change and got this:
[image]

Here is the output:
```log
/Users/tennis/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin/xtensa-esp32s3-elf-gdb

Info : 1113 334257 server.c:90 add_connection(): accepting 'gdb' connection on tcp/3333
Debug: 1114 334257 breakpoints.c:328 breakpoint_remove_all_internal(): [esp32c3] Delete all breakpoints
Debug: 1115 334257 breakpoints.c:655 watchpoint_clear_target(): Delete all watchpoints for target: esp32c3

❌ Debug: 1116 334257 target.c:1781 target_call_event_callbacks(): target event 22 (gdb-attach) for core esp32c3
Debug: 1117 334257 target.c:4679 target_handle_event(): target: esp32c3 (esp32c3) event: 22 (gdb-attach) action: 
		if { $_ESP_SMP_BREAK != 0 } {
			$_TARGETNAME_0 xtensa smpbreak BreakIn BreakOut
		}
		# necessary to auto-probe flash bank when GDB is connected and generate proper memory map
		halt 1000
		if { [$_ESP_MEMPROT_IS_ENABLED] } {
			# 'reset halt' to disable memory protection and allow flasher to work correctly
			echo "Memory protection is enabled. Reset target to disable it..."
			reset halt
		}

		if { $_ESP_ARCH == "riscv" } {
			# by default mask interrupts while stepping
			riscv set_maskisr steponly
		}
	
Debug: 1118 334257 command.c:152 script_debug(): command - halt 1000
Debug: 1119 334257 target.c:3251 handle_halt_command(): -
Error: 1120 334257 target.c:517 target_halt(): Target not examined yet
Debug: 1121 334257 command.c:528 exec_command(): Command 'halt' failed with error code -4
User : 1122 334257 target.c:4704 target_handle_event(): Error executing event gdb-attach on target esp32c3:

Debug: 1123 334257 FreeRTOS.c:1404 freertos_clean(): freertos_clean
Debug: 1124 334257 FreeRTOS.c:899 freertos_update_threads(): freertos_update_threads
Warn : 1125 334257 FreeRTOS.c:907 freertos_update_threads(): No symbols for FreeRTOS!
Error: 1126 334257 esp_flash.c:952 esp_algo_flash_probe(): Target not halted
Error: 1127 334257 core.c:291 get_flash_bank_by_num(): auto_probe failed
Error: 1128 334257 gdb_server.c:1052 gdb_new_connection(): Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: 1129 334257 server.c:94 add_connection(): attempted 'gdb' connection rejected

```

For some reason, it thinks I have an esp32c3.  I don't. I have an esp32s3.

brianignacio5, 2024-12-23:
I can see in your doctor command output that the project is configured to use esp32c3. See the `"idf.openOCDConfigs": ["board/esp32c3-builtin.cfg"]`

Use the `ESP-IDF: Select OpenOCD Board Configuration` command to select your esp32s3 board.

gamename, 2024-12-23:
@brianignacio5 

> ESP-IDF: Select OpenOCD Board Configuration

THanks! That helped alot.  Now I'm stuck on:
```Error: 463 3 server.c:267 add_service(): couldn't bind tcl to socket on port 6666: Address already in use
```

AFAIK, nothing is using that port.  Can it be changed?

brianignacio5, 2024-12-23:
It could be that openOCD didn't close properly before and there is a process still running. See if you can kill the hanging process.

You can also use the `openocd.tcl.port` configuration setting to use a different port.

gamename, 2024-12-23:
@brianignacio5 

> It could be that openOCD didn't close properly before and there is a process still running

That was exactly the problem. Thanks for the help. Everything is working now. :) 

Closing...

closed 2024-12-23 by gamename