espressif/vscode-esp-idf-extension #1386: [Feature Request]: Allow 'idf.svdFilePath' to construct the path to an .svd file based on the current target

GillesZunino, 2024-12-27:
**Is your feature request related to a problem?**

The "ESP-IDF Peripheral Viewer" window requires the setting `idf.svdFilePath` to be set to the full path of an SVD file matching the target being debugged (i.e. esp32s3 or esp32 ...). Before v1.9.0, `idf.adapterTargetName` could be used to construct `idf.svdFilePath` as follows:
```json
"idf.svdFilePath": "${workspaceFolder}/svd/${config:idf.adapterTargetName}.svd"
```
This made configuring the svd file to use automatic. There was no need to remember to adjust `idf.svdFilePath` when changing the target. 

Version v1.9.0, removed `idf.adapterTargetName` (a *great change* in my opinion) making it impossible to automatically construct `idf.svdFilePath`. This feature proposes a few ways to automatically configure `idf.svdFilePath` as the target changes without bringing back the problematic setting `idf.adapterTargetName`.

**Describe the solution you'd like**

Expose a read only configuration setting tentatively called `idf.currentTargetName` which is interpreted as the value returned by `getIdfTargetFromSdkconfig()`. This would allow the value of `idf.svdFilePath` to automatically adapt as the target changes with the following sample configuration in `settings.json`:
```json
"idf.svdFilePath": "${workspaceFolder}/svd/${config:idf.currentTargetName}.svd"
```

**Describe alternatives you've considered**

* Another approach would be to add a new setting tentatively called `idf.svdFilesRoot` pointing to a directory where the extension can locate the svd file, assuming it is called getIdfTargetFromSdkconfig() + '.svd'.

* Another approach would be to add a command tentatively called `idf.getCurrentTargetName` which would return whatever `getIdfTargetFromSdkconfig()` returns at the time `idf.svdFilePath` is being used by the extension. The following sample configuration in `settings.json` would be valid:
    ```json
    "idf.svdFilePath": "${workspaceFolder}/svd/${command:idf.getCurrentTargetName}.svd"
    ```

brianignacio5, 2024-12-27:
Please try [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/2365313106.zip) to see if the issue is fixed.

You can test these changes by installing this VSIX by click menu View -> Command Palette..., type Install from VSIX and then select downloaded esp-idf-extension.vsix file to install the extension.

In this Pull Request, we are saving whatever Espressif target you select with `ESP-IDF: Set Espressif Device Target` in settings.json `idf.customExtraVars` which is a list of environment variables, adding IDF_TARGET variable to selected target. We also add a way to use it in settings: `"idf.svdFilePath": "${workspaceFolder}/${env:IDF_TARGET}.svd"`. So now we resolve env variables also with `idf.customExtraVars`

closed 2025-01-06 by brianignacio5

GillesZunino, 2025-01-10:
Verified the vsix installer above supports `"idf.svdFilePath": "${workspaceFolder}/${env:IDF_TARGET}.svd"`. Works well. Thank you !

PR #1385 (2025-01-06): [VSC-1537] use IDF_TARGET in customExtraVars if no sdkconfig
## Description

To find the current IDF_TARGET the extension reads the sdkconfig in v1.9.0 but before there was a `idf.adapterTargetName` configuration setting which was used to resolve IDF_TARGET. The previous logic would fail if the current project is configure for a target (say esp32) but the configuration setting is set to (say esp32c3).

Reading from sdkconfig seems like a better approach, but what happens when you want to define a IDF_TARGET but no sdkconfig exists ? For this case we still need to persist this somehow. Now we w [... cut]