espressif/vscode-esp-idf-extension #1387: [Feature Request] Make command "Open ESP-IDF Terminal" run user profile and customizations (Powershell specific)

GillesZunino, 2024-12-27:
**Is your feature request related to a problem?**

In v1.9.0, the command "ESP-IDF: Open ESP-IDF Terminal" starts Powershell with argument `-NoProfile` ignoring user profile customizations. I have a fair amount of useful ESP-IDF Powershell customizations and I miss them in terminals opened via "Open ESP-IDF Terminal" !

COMPARE: Prior v1.9.0, "ESP-IDF: Open ESP-IDF Terminal" did not add `-NoProfile` to the Powershell command line, allowing Powershell to apply user's customizations.

I realize that the extension itself needs to spawn Powershell behind the scene to implement some of its capabilities. In this case, it is much better to start Powershell with `-NoProfile` to run commands faster and in a more predictable way since no user customization is applied.

**Describe the solution you'd like**

Enhance VS Code command "ESP-IDF: Open ESP-IDF Terminal" to start Powershell **without** `-NoProfile`. When a user requests a terminal to be created, it is expected the terminal will be interactive and have profile customizations.

Let all ESP-IDF extension internal invocations of Powershell to continue to run with `-NoProfile` to keep the speed improvement and execution predictability.

brianignacio5, 2024-12-27:
Please try [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/2365335124.zip) to see if the issue is fixed.

You can test these changes by installing this VSIX by click menu View -> Command Palette..., type Install from VSIX and then select downloaded esp-idf-extension.vsix file to install the extension.

GillesZunino, 2025-01-06:
It tried the VSIX installed above and it worked very well ! Thank you !

PR #1388 (2025-01-10): [VSC-1551] rm powershell no profile arg idf terminal
## Description

Remove `-NoProfile` argument in `ESP-IDF: Open ESP-IDF Terminal` to avoid user restriction of profiles. Was added due to script need to run in Powershell to ByPass execution policy in order to add command shortcuts.

Fixes #1387

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "ESP-IDF: Open ESP-IDF Terminal`". The Terminal should allow user to apply a Powershell profile now.
2. Exec [... cut]
closed 2025-01-10 by brianignacio5