espressif/vscode-esp-idf-extension #1395: Launch gdbtarget seems to be using wrong program.

i400s, 2025-01-04:
### OS

Linux

### Operating System version

Debian Trixie

### Visual Studio Code version

1.96.2

### ESP-IDF version

v5.5-dev-1050-gb5ac4fbdf9

### Python version

3.12.8

### Doctor command output

[report.txt](https://github.com/user-attachments/files/18306860/report.txt)

### Extension

{"message":"OpenOCD server failed to start because of Error: Unknown target type esp32c6, try one of arm7tdmi, arm9tdmi, arm920t, arm720t, arm966e, arm946e, arm926ejs, fa526, feroceon, dragonite, xscale, xtensa, cortex_m, cortex_a, cortex_r4, arm11, ls1_sap, mips_m4k, avr, dsp563xx, dsp5680xx, testee, avr32_ap7k, hla_target, nds32_v2, nds32_v3, nds32_v3m, esp32, esp32s2, esp32s3, or1k, quark_x10xx, quark_d20xx, stm8, riscv, mem_ap, esirisc, arcv2, aarch64,  or mips_mips64\n❌ /home/wilsonj/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:120: Error: Unknown target type esp32c6, try one of arm7tdmi, arm9tdmi, arm920t, arm720t, arm966e, arm946e, arm926ejs, fa526, feroceon, dragonite, xscale, xtensa, cortex_m, cortex_a, cortex_r4, arm11, ls1_sap, mips_m4k, avr, dsp563xx, dsp5680xx, testee, avr32_ap7k, hla_target, nds32_v2, nds32_v3, nds32_v3m, esp32, esp32s2, esp32s3, or1k, quark_x10xx, quark_d20xx, stm8, riscv, mem_ap, esirisc, arcv2, aarch64,  or mips_mips64\nTraceback (most recent call last):\n  File \"/home/wilsonj/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts/target/esp32c6.cfg\", line 170, in script\n    create_esp_target riscv\n  File \"/home/wilsonj/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts/target/esp_common.cfg\", line 159, in create_esp_target\n    create_openocd_targets\n  File \"/home/wilsonj/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts/target/esp_common.cfg\", line 120, in create_openocd_targets\n    {target create} esp32c6.hp.cpu0 esp32c6 -chain-position esp32c6.tap0 -coreid 0 -rtos hwthread\n","stack":"Error: OpenOCD server failed to start because of Error: Unknown target type esp32c6, try one of arm7tdmi, arm9tdmi, arm920t, arm720t, arm966e, arm946e, arm926ejs, fa526, feroceon, dragonite, xscale, xtensa, cortex_m, cortex_a, cortex_r4, arm11, ls1_sap, mips_m4k, avr, dsp563xx, dsp5680xx, testee, avr32_ap7k, hla_target, nds32_v2, nds32_v3, nds32_v3m, esp32, esp32s2, esp32s3, or1k, quark_x10xx, quark_d20xx, stm8, riscv, mem_ap, esirisc, arcv2, aarch64,  or mips_mips64\n\tat Socket.<anonymous> (/dist/extension.js:2:1565397)\n\tat Socket.emit (node:events:518:28)\n\tat addChunk (node:internal/streams/readable:561:12)\n\tat readableAddChunkPushByteMode (node:internal/streams/readable:512:3)\n\tat Readable.push (node:internal/streams/readable:392:5)\n\tat Pipe.onStreamRead (node:internal/stream_base_commons:191:23)","level":"error","timestamp":"2025-01-04T12:53:38.010Z"}
{"message":"OpenOCD Exit with non-zero error code 1","stack":"Error: Spawn exit with non-zero1\n\tat ChildProcess.<anonymous> (/dist/extension.js:2:1566090)\n\tat ChildProcess.emit (node:events:518:28)\n\tat maybeClose (node:internal/child_process:1104:16)\n\tat Socket.<anonymous> (node:internal/child_process:456:11)\n\tat Socket.emit (node:events:518:28)\n\tat Pipe.<anonymous> (node:net:343:12)","level":"error","timestamp":"2025-01-04T12:53:38.011Z"}

### Description

The debug adapter fails to run which seems to be because its using the /usr/bin/openocd version and not the one stored in the .espressif directory.

From within the terminal of vscode running esp-idf:
```
echo $PATH
/mnt/md8/_development/esp/esp-idf/components/espcoredump:/mnt/md8/_development/esp/esp-idf/components/partition_table:/mnt/md8/_development/esp/esp-idf/components/app_update:/home/wilsonj/.espressif/tools/xtensa-esp-elf-gdb/15.2_20241112/xtensa-esp-elf-gdb/bin:/home/wilsonj/.espressif/tools/riscv32-esp-elf-gdb/15.2_20241112/riscv32-esp-elf-gdb/bin:/home/wilsonj/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin:/home/wilsonj/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin:/home/wilsonj/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:/home/wilsonj/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin:/home/wilsonj/.espressif/python_env/idf5.5_py3.12_env/bin:/mnt/md8/_development/esp/esp-idf/tools:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
which openocd
/home/wilsonj/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin/openocd
```
ESP-IDF Output after running a `gdbtarget` from within a project directory with a `.vscode/launch.json` file
```
/home/wilsonj/.espressif/tools/riscv32-esp-elf-gdb/15.2_20241112/riscv32-esp-elf-gdb/bin/riscv32-esp-elf-gdb

/usr/bin/openocd

Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000

WARNING!!! OpenOCD binary version (Open On-Chip Debugger 0.12.0) doesn't match the .cfg file version (v0.12.0-esp32-20241016)

❌ /home/wilsonj/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:120: Error: Unknown target type esp32c6, try one of arm7tdmi, arm9tdmi, arm920t, arm720t, arm966e, arm946e, arm926ejs, fa526, feroceon, dragonite, xscale, xtensa, cortex_m, cortex_a, cortex_r4, arm11, ls1_sap, mips_m4k, avr, dsp563xx, dsp5680xx, testee, avr32_ap7k, hla_target, nds32_v2, nds32_v3, nds32_v3m, esp32, esp32s2, esp32s3, or1k, quark_x10xx, quark_d20xx, stm8, riscv, mem_ap, esirisc, arcv2, aarch64,  or mips_mips64
Traceback (most recent call last):
  File "/home/wilsonj/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts/target/esp32c6.cfg", line 170, in script
    create_esp_target riscv
  File "/home/wilsonj/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 159, in create_esp_target
    create_openocd_targets
  File "/home/wilsonj/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 120, in create_openocd_targets
    {target create} esp32c6 esp32c6 -chain-position esp32c6.tap0 -coreid 0 -rtos FreeRTOS

For assistance with OpenOCD errors, please refer to our Troubleshooting FAQ: https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
OpenOCD Exit with non-zero error code 1
[Stopped] : OpenOCD Server

```

### Debug Message

none

### Other Steps to Reproduce

Starting openocd on the command line `openocd -f board/esp32c6-builtin.cfg -c "adapter serial 40:4C:CA:43:1C:AC"` starts openocd without issues.

Then when using the `gdbtarget` it fails to start openocd (same issue) but because there is an instance already started via the command line it connects to that one and opens the debug at `app_main()` with the little buttons (continue/jump/etc) being presented.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-01-07:
It seems that is the issue of vscode with environment variables. It doesn't pre append to env variables like PATH so the /usr/bin/openocd is used instead.

closed 2025-01-30 by github-actions[bot]

brianignacio5, 2025-10-29:
This should be fixed with https://github.com/espressif/vscode-esp-idf-extension/pull/1666