espressif/vscode-esp-idf-extension #1399: Visual Studio Disassembly view out of sync

torsts, 2025-01-12:
### OS

Windows

### Operating System version

Microsoft Windows 11 Pro

### Visual Studio Code version

1.96.2

### ESP-IDF version

5.4

### Python version

3.11.2

### Doctor command output

[report.txt](https://github.com/user-attachments/files/18390843/report.txt)

### Extension

[esp_idf_vsc_ext.log](https://github.com/user-attachments/files/18390878/esp_idf_vsc_ext.log)

### Description

I am debugging an esp32-s3 application for M5stack cardputer. When opening the disassembly view, the active line in the disassembly view does not match the current instruction pointer.
[image]
[image]
[image]

I started debugging the problem in Visual Studio Code and ended in vscode-esp-idf-extension.
The problem is in
vscode-esp-idf-extension/src/cdtDebugAdapter/adapter/GDBDebugSession.ts
function:
  protected async disassembleRequest(
    response: DebugProtocol.DisassembleResponse,
    args: CDTDisassembleArguments
  ) {
According to the debug adapter protocol:
https://microsoft.github.io/debug-adapter-protocol//specification.html
interface DisassembleArguments {
has:
  /**
   * Offset (in instructions) to be applied after the byte offset (if any)
   * before disassembling. Can be negative.
   */
  instructionOffset?: number;

This argument is completely ignored by the disassembleRequest function.
In my particular debug session the value of instructionOffset = -50.
This leads to the disassembly view marking 0x420397ef in the view, whereas Registers.pc => 0x420395e1.

The mapping between these addresses happens in
private async loadDisassembledInstructions
found in the following Visual Studio Code source code file:
vscode/src/vs/workbench/contrib/debug/browser/disassemblyView.ts

### Debug Message

[debug_console_output.txt](https://github.com/user-attachments/files/18390888/debug_console_output.txt)

### Other Steps to Reproduce

Repro: I opened one of my projects, set a hardware breakpoint on esp_log_writev in log_write.c from espressif and do start debugging.
When the breakpoint is hit, I open the disassembly view ( and start single step).

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-01-13:
Let me summarize the relevant code and discuss about it:

```ts
const fetchSize = (args.instructionCount - instructions.length) * meanSizeOfInstruction;

const stepStartAddress = `(${args.memoryReference})+${startOffset}`;
let stepEndAddress = `(${args.memoryReference})+${startOffset}+${fetchSize}`;
if (args.endMemoryReference && instructions.length === 0) {
  // On the first call, if we have an end memory address use it instead of
  // the approx size
  stepEndAddress = args.endMemoryReference;
  oneIterationOnly = true;
}
const result = await sendDataDisassemble(
  this.gdb,
  stepStartAddress,
  stepEndAddress
);
```
The main issue with the alignment is that vscode disassemble request doesn't use the actual address you want when you right click and click to open disassemble view and uses some arbitrary memory reference `args.memoryReference`. Here the code tries to calculate a window to fetch from GDB.

In my tests I could not obtain such address from the request. If you in fact are sure that `args.memoryReference` is the right address and the offset too I would revisit this issue.

torsts, 2025-01-13:
Hi, so VSC does an internal remapping based on the instructionOffset. From my understanding it sets the instructionOffset in order to display code prior to the one that the current focus point is. I am attaching a screenshot from the actual function from VSC that requests the disassembly data, The blue box shows where it will go into the esp-idf extension.

The box at line 497 shows on how the instruction reference (that is a string) is mapped to the actual address (based on the instructionOffset). 

When it later goes into the render function it takes the reference and resolves it to the actual address. What lets the disassembly view end up in the wrong location. All this is based on the understanding, that the disassemble function knows on how to deal with the instructionOffset.

The box on the right shows you the mapping entry for reference "0x420395e4" to address 0x4203965e.

[image]

To answer your 2 questions:
a) yes, you spotted the code on esp-idf side that is responsible for handling the disassemby request
b) What I really want is that the disassembly view shows me were in the code I really am. So I thought, hndling the instructionOffset parameter is going to resolve this.

torsts, 2025-01-13:
As a sidenote, VSC in my tests called the loadDisassembledInstructions function twice for the same address (with different instructionOffset)

focusStackFrame -> goToIntructionAndOffset -> loadDisassembledInstructions instructionCount: 50, instructionOffset; 0, instructionReference: "0x420395e4", offset: 0
focusStackFrame -> setFocus -> fire -> goToIntructionAndOffset -> loadDisassembledInstructions instructionCount: 100, instructionOffset; -50, instructionReference: "0x420395e4", offset: 0

osnwt, 2025-02-19:
Same issue.

PR #1518 (2025-05-08): [VSC-1557] add disassembly view updates
## Description

Update disassembly view request code to match latest DAP protocol and Eclipse CDT debug adapter implementation.

Fixes #1399

## Type of change

- New feature (non-breaking change which adds functionality)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Start a debug session on a ESP-IDF project. Right click and select Open Disassembly view
3. Execute action.
4. Observe results.

- Expected behaviour:
The resulting code for assembly should match better with original C code.

 [... cut]