espressif/vscode-esp-idf-extension #1401: vscode cannot connect to Jlink for debugging

zhd2323, 2025-01-16:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.96.3

### ESP-IDF version

5.3.2

### Python version

3.11.2

### Doctor command output

[log (2).txt](https://github.com/user-attachments/files/18432687/log.2.txt)

### Description

vscode cannot connect to Jlink for debugging,openocd cannot open

### Debug Message

Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Executing action: set-target
Set Target to: esp32s3, new sdkconfig will be created.
Running cmake in directory C:\zhd\ESP32_pro\gatt_server_service_table\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\zhd\esp322\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=0 C:\zhd\ESP32_pro\gatt_server_service_table"...
-- Existing sdkconfig 'C:/zhd/ESP32_pro/gatt_server_service_table/sdkconfig' renamed to 'C:/zhd/ESP32_pro/gatt_server_service_table/sdkconfig.old'.

-- Found Git: C:/zhd/esp322/Espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1")

-- The C compiler identification is GNU 13.2.0

-- The CXX compiler identification is GNU 13.2.0

-- The ASM compiler identification is GNU

-- Found assembler: C:/zhd/esp322/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe

-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - done

-- Check for working C compiler: C:/zhd/esp322/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped

-- Detecting C compile features
-- Detecting C compile features - done

-- Detecting CXX compiler ABI info

-- Detecting CXX compiler ABI info - done

-- Check for working CXX compiler: C:/zhd/esp322/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped

-- Detecting CXX compile features

-- Detecting CXX compile features - done

-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
[... 106 lines cut ...]
Open On-Chip Debugger v0.12.0-esp32-20241016 (2024-10-16-14:17)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html

debug_level: 2

adapter speed: 20000 kHz

Info : Listening on port 6666 for tcl connections

Info : Listening on port 4444 for telnet connections

❌ Error: unable to open ftdi device with description '*', serial '*' at bus location '*'

❌ C:\zhd\esp322\Espressif\tools\openocd-esp32\v0.12.0-esp32-20241016\openocd-esp32\share\openocd\scripts/target/esp_common.cfg:9: Error: 
at file "C:\zhd\esp322\Espressif\tools\openocd-esp32\v0.12.0-esp32-20241016\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 9

For assistance with OpenOCD errors, please refer to our Troubleshooting FAQ: https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
OpenOCD Exit with non-zero error code 1
[Stopped] : OpenOCD Server

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

zhd2323, 2025-01-16:
Operating System version ues Windows 11

zhd2323, 2025-01-16:
settings.json:
{
    "C_Cpp.intelliSenseEngine": "default",
    "idf.espIdfPathWin": "c:\\zhd\\esp322\\Espressif\\frameworks\\esp-idf-v5.3.2",
    "idf.toolsPathWin": "C:\\zhd\\esp322\\Espressif",
    "idf.flashType": "JTAG",
    "idf.portWin": "COM24",
    "idf.openOcdConfigs": [
        "interface/ftdi/esp32_devkitj_v1.cfg",
        "target/esp32s3.cfg"
    ],
    "files.associations": {
        "freertos.h": "c",
        "stdio.h": "c",
        "led.h": "c"
    }
}

brianignacio5, 2025-01-16:
Could you update `idf.openOcdConfigs`  in settings.json to :

```json
"idf.openOcdConfigs": [
"interface/jlink.cfg",
"target/esp32s3.cfg"
],
```

karlp, 2025-01-29:
Possibly fixed by: https://github.com/espressif/vscode-esp-idf-extension/pull/1413

closed 2025-02-19 by github-actions[bot]