espressif/vscode-esp-idf-extension #1402: Moving SDKCONFIG stop functioning

ammaree, 2025-01-16:
### OS

Windows

### Operating System version

15.3

### Visual Studio Code version

1.96.3

### ESP-IDF version

5.3.2

### Python version

3.13

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin x64 24.3.0 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin:/Library/Apple/usr/bin:/Applications/VMware Fusion.app/Contents/Public 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.96.3 
Visual Studio Code language en 
Visual Studio Code shell /bin/zsh 
ESP-IDF Extension version 1.9.0 
Workspace folder <HOMEPATH>/DevSpace/z-projects/irmacs 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/DevSpace/z-sdk/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/cmake/3.30.2/CMake.app/Contents/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/ninja/1.12.1:<HOMEPATH>/.espressif/tools/esp-rom-elfs/20240305
ESP-IDF extra vars
    OPENOCD_SCRIPTS: <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/.espressif/tools/esp-rom-elfs/20240305/
    IDF_PYTHON_ENV_PATH: <HOMEPATH>/.espressif/python_env/idf5.3_py3.13_env
    ESP_IDF_VERSION: 5.3
    PATH: <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/qemu-xtensa/esp_develop_9.0.0_20240606/qemu/bin:<HOMEPATH>/.espressif/tools/qemu-riscv32/esp_develop_9.0.0_20240606/qemu/bin:<HOMEPATH>/.espressif/python_env/idf5.3_py3.13_env/bin:<HOMEPATH>/DevSpace/z-sdk/esp-idf/tools:$PATH
    IDF_DEACTIVATE_FILE_PATH: /var/folders/l7/2mnysjf14w11tv45mh1q_lpr0000gn/T/tmpv7hsm5lcidf_76032
User extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/.espressif/tools/esp-rom-elfs/20240530/
System python Path (idf.pythonInstallPath) /usr/local/bin/python3
Virtual environment Python path (computed) <HOMEPATH>/.espressif/python_env/idf5.3_py3.13_env/bin/python
Serial port (idf.port) /dev/tty.usbmodemD4F98D72B48C1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) git
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin: true
[... 129 lines cut ...]
{"message":"Cannot access filePath: ${env:HOMEKIT_PATH}","stack":"Error: ENOENT: no such file or directory, access '${env:HOMEKIT_PATH}'\n\tat Object.accessSync (node:fs:246:11)\n\tat t.accessSync (node:electron/js2c/node_init:2:9332)\n\tat Me (/Users/andremaree/.vscode/extensions/espressif.esp-idf-extension-1.9.0/dist/extension.js:2:1432413)\n\tat /Users/andremaree/.vscode/extensions/espressif.esp-idf-extension-1.9.0/dist/extension.js:2:1667825\n\tat Ca (/Users/andremaree/.vscode/extensions/espressif.esp-idf-extension-1.9.0/dist/extension.js:2:1668628)\n\tat async /Users/andremaree/.vscode/extensions/espressif.esp-idf-extension-1.9.0/dist/extension.js:2:1890460","level":"error","timestamp":"2025-01-16T08:06:50.764Z"}
{"message":"Cannot access filePath: /Users/andremaree/.espressif/tools/cmake/3.30.2/CMake.app/Contents/bin","stack":"Error: ENOENT: no such file or directory, access '/Users/andremaree/.espressif/tools/cmake/3.30.2/CMake.app/Contents/bin'\n\tat Object.accessSync (node:fs:246:11)\n\tat t.accessSync (node:electron/js2c/node_init:2:9332)\n\tat Me (/Users/andremaree/.vscode/extensions/espressif.esp-idf-extension-1.9.0/dist/extension.js:2:1432413)\n\tat /Users/andremaree/.vscode/extensions/espressif.esp-idf-extension-1.9.0/dist/extension.js:2:1668341\n\tat Ca (/Users/andremaree/.vscode/extensions/espressif.esp-idf-extension-1.9.0/dist/extension.js:2:1668628)\n\tat async /Users/andremaree/.vscode/extensions/espressif.esp-idf-extension-1.9.0/dist/extension.js:2:1890460","level":"error","timestamp":"2025-01-16T08:06:50.764Z"}
{"message":"Cannot access filePath: /Users/andremaree/.espressif/tools/ninja/1.12.1","stack":"Error: ENOENT: no such file or directory, access '/Users/andremaree/.espressif/tools/ninja/1.12.1'\n\tat Object.accessSync (node:fs:246:11)\n\tat t.accessSync (node:electron/js2c/node_init:2:9332)\n\tat Me (/Users/andremaree/.vscode/extensions/espressif.esp-idf-extension-1.9.0/dist/extension.js:2:1432413)\n\tat /Users/andremaree/.vscode/extensions/espressif.esp-idf-extension-1.9.0/dist/extension.js:2:1668341\n\tat Ca (/Users/andremaree/.vscode/extensions/espressif.esp-idf-extension-1.9.0/dist/extension.js:2:1668628)\n\tat async /Users/andremaree/.vscode/extensions/espressif.esp-idf-extension-1.9.0/dist/extension.js:2:1890460","level":"error","timestamp":"2025-01-16T08:06:50.764Z"}
{"message":"ENOENT: no such file or directory, open '/Users/andremaree/DevSpace/z-projects/irmacs/.vscode/launch.json'","stack":"Error: ENOENT: no such file or directory, open '/Users/andremaree/DevSpace/z-projects/irmacs/.vscode/launch.json'","level":"error","timestamp":"2025-01-16T08:06:52.902Z"}
{"user":true,"level":"warn","message":"Extension configuration report has been copied to clipboard with errors","timestamp":"2025-01-16T08:06:52.902Z"}

### Description

In order to keep all build artefacts together we use the following CMake command to relocate the SDKCONFIG related output.

set( SDKCONFIG ${CMAKE_BINARY_DIR}/sdkconfig )

This unfortunately has the effect of stopping the VS Code extension from functioning since it is unable to determine the TARGET configuration. In addition debugging and various other things also fail.

### Debug Message

None required since the situation was tested and proven correct

### Other Steps to Reproduce

None

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2025-01-16:
Which `CMakeLists.txt` file are you using to define `set( SDKCONFIG ${CMAKE_BINARY_DIR}/sdkconfig )` ? The extension should pick from root directory CMakeLists.txt and replace `${CMAKE_BINARY_DIR}` with the value of `idf.buildPath` configuration setting which default to `${workspaceFolder}/build`.

I have tested this for the Project Configuration editor feature of the extension documented in here: https://docs.espressif.com/projects/vscode-esp-idf-extension/en/latest/additionalfeatures/project-configuration.html where 2 project profile each have sdkconfig in build directory and each project profile define a build directory path

ammaree, 2025-01-16:
Hi Brian, some more info....

I use a multi-root workspace with about 4~6 active projects.

Our primary build system uses an external file "build-system.cmake" which is located at the same level as "my.code-workspace" file in "~/DevSpace/z-projects" directory

For most projects the change setting happens in this "build-system.cmake" file.

For other simpler once off projects it happens in the project level CMAkeLists.txt level, not at the "main" level.

ammaree, 2025-01-16:
Example of the project CMakeLists.txt file and project root.

```
cmake_minimum_required( VERSION 3.16...3.60 )
set( cmakePROJECT IRMACS )
set( cmakeMODEL ac01 )
include ( ../build-system.cmake )
project( ${cmakeOUTPUT} )
```

brianignacio5, 2025-01-17:
I think the issue is how can the extension knows where sdkconfig is defined. Right now the sdkconfig is resolve as follows:

1) Try to find a SET( SDKCONFIG path) in `<project-dir>/CMakeLists.txt`
2) If exists, Replace `${CMAKE_BINARY_DIR}` with value from `idf.buildPath` configuration setting.
3) If it doesn't exist read `idf.sdkconfigFilePath` configuration setting. (This one is override with currently selected Project Configuration Profile if defined)
4) If it doesn't exist use `<project-dir>/sdkconfig`
5) If it doesn't exist use `<project-dir>/sdkconfig.defaults`

We could try to read an env variable SDKCONFIG but as you see these variable only exists in the build system runtime so the extension can't read it, because build is launch as system process execution and the content of it. We could also use the build directory's project_description.json `config_file` but you would still need to know which build folder to use and fallback when no build directory exists.

Right now You can try to use the `idf.sdkconfigFilePath` configuration setting in `${project_dir}/.vscode/settings.json` if you only have a single build per project. If you have multiple build directories per project, I would suggest using the [Project Configuration editor](https://docs.espressif.com/projects/vscode-esp-idf-extension/en/latest/additionalfeatures/project-configuration.html) feature to define one profile per build directory (and specify sdkconfig file and build directory on each profile as done in the linked documentation).

closed 2025-02-08 by github-actions[bot]